arkq/bluez-alsa #621: build on raspberry pi 4 (Raspbian bullseye)

zoldaten, 2023-03-27:
hi!
following the wiki i built bluez-alsa tools without any errors.
```
sudo apt-get install libsbc1 libsbc-dev
sudo apt-get install libreadline-dev
autoreconf --install
mkdir build && cd build
../configure --enable-cli --enable-rfcomm --enable-systemd
make
sudo make install
```
(printed during installation:
Libraries have been installed in:
   /usr/lib/aarch64-linux-gnu/alsa-lib)

but when i try to start bluealsa-aplay i see:
`bluealsa-aplay: W: Couldn't get BlueALSA PCM list: The name org.bluealsa was not provided by any .service files`

`aplay -D bluealsa yourfile.wav` gives this:
```
ALSA lib ../../../src/asound/bluealsa-pcm.c:1384:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: The name org.bluealsa was not provided by any .service files
aplay: main:830: audio open error: No such device
```

bluealsa-aplay -V:
`v4.0.0-123-g9cb3d75-dirty`

cat ~/.asoundrc:
```
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "DC:A6:32:FE:86:74"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000
```

how to start it working ?

borine, 2023-03-28:
You don't say which OS image you are using, but I'm guessing its either the "Desktop" or "Full" image. Assuming you have read the bluez-alsa project README you will know that to use BlueALSA "The target system must be able to function correctly with all its audio applications interfacing directly with ALSA". Both those images are heavyweight general purpose OS with integrated Bluetooth audio built-in, using pulseaudio. So BlueALSA is not required, and in all but a very few specialized cases it is not appropriate to install BlueALSA on either of those OS images.

I think you need to read the OS documentation for bluetooth audio, because almost certainly you should be using the built-in functionality (or else use a different image, such as the RaspberryPiOS "Lite" image, which does not have a desktop nor integrated pulseaudio).

zoldaten, 2023-03-30:
correct i use "Desktop".
thanks for the answer.
think i could use bluealse-cli.

closed 2023-03-30 by zoldaten