arkq/bluez-alsa #626: airpods3 failed to connect back

luoyuyi111, 2023-04-12:
hi,
I found the airpods3 after successfully connecting, turn off it. Then turn on the airpods3 . At this time, the connection will automatically return, and the return will fail.    

Bluealsa will prompt "bluealsa: Invalid configuration: Invalid object type“.  From hcilog see airpods3  simultaneously set MPEG-2 AAC LC, MPEG-4 AAC LC caused.

borine, 2023-04-12:
> bluealsa: Invalid configuration: Invalid object type

It seems you are using an old release of BlueALSA. That error message was removed in release 3.0.0 (September 2020), so your release must be from before that. I guess you obtained it from your OS distribution repository. Could that be the RaspberryPi repo by any chance?

arkq, 2023-04-12:
> It seems you are using an old release of BlueALSA.

Yes, this issue was fixed some time ago (I do not remember which commit fixes it, though). Please, use the latest release (currently 4.0.0).

luoyuyi111, 2023-04-14:
> It seems you are using an old release of BlueALSA. That error message was removed in release 3.0.0 (September 2020), so your release must be from before that. I guess you obtained it from your OS [...]

I'm currently using V3.1.0, which doesn't seem to be fixed. I'll try it on the latest version later. thanks

closed 2023-04-24 by luoyuyi111