arkq/bluez-alsa #627: Fail to use google home mini as bluetooth speaker

neslinesli93, 2023-04-15:
### Problem

Hi, I'm trying to use my google home mini as a bluetooth speaker to play audio from a raspberry pi 4. The raspberry uses dietpi, a distro that doesn't have pulseaudio nor pipewire, just plain ALSA. After pairing and connecting to the speaker using `bluetoothctl`, running `aplay -D bluealsa beep.wav` works just the first time, then it fails to play further sounds until a reboot is performed, followed by a reconnect. Using the home mini as a bluetooth speaker from my android phone works fine with no issues.

Here are the debug logs of the bluealsa service, which contain the successful play attempt and a couple failed attempts: 
[logs.txt](https://github.com/arkq/bluez-alsa/files/11240344/logs.txt)

And here are the logs of aplay:
```
root@romeo:~# aplay -D bluealsa beep.wav 
[1491] D: ../../../src/asound/bluealsa-pcm.c:1380: Getting BlueALSA PCM: PLAYBACK 00:00:00:00:00:00 a2dp
[1491] D: ../../../src/asound/bluealsa-pcm.c:1133: /org/bluealsa/hci0/dev_F0_EF_86_6A_C6_3D/a2dpsrc/sink: Setting constraints
Playing WAVE 'beep.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Mono
[1491] D: ../../../src/asound/bluealsa-pcm.c:546: /org/bluealsa/hci0/dev_F0_EF_86_6A_C6_3D/a2dpsrc/sink: Initializing HW
[1491] D: ../../../src/asound/bluealsa-pcm.c:566: /org/bluealsa/hci0/dev_F0_EF_86_6A_C6_3D/a2dpsrc/sink: Couldn't open PCM: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
aplay: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 44100
PERIOD_TIME: 125000
PERIOD_SIZE: (5512 5513)
PERIOD_BYTES: (11024 11026)
PERIODS: (3 5)
BUFFER_TIME: 500000
BUFFER_SIZE: 22050
BUFFER_BYTES: 44100
TICK_TIME: 0
[1491] D: ../../../src/asound/bluealsa-pcm.c:457: /org/bluealsa/hci0/dev_F0_EF_86_6A_C6_3D/a2dpsrc/sink: Closing
```

### Setup

**OS**: Linux 5.15.84-v8+, DietPi v8.14.2, RPi 4 Model B (aarch64)
**Bluealsa**: I'm using the master branch of `bluez-alsa` (541523ae1a3c4c2fb7264543777a48a93d66cd47), built using the following flags: `../configure --enable-mp3lame  --enable-manpages --enable-systemd --enable-debug --disable-payloadcheck`
**BlueZ**: 5.55
**ALSA**: aplay: version 1.2.4

arkq, 2023-04-15:
It's quite possible that you've hit the bug introduced some time ago. Unfortunately, I've got no time right now to fix it properly. Try to replace these lines:
https://github.com/arkq/bluez-alsa/blob/541523ae1a3c4c2fb7264543777a48a93d66cd47/src/ba-transport.c#L1625-L1626
with:
```c
	if ((fd = t->acquire(t)) != -1) {
		ba_transport_thread_state_set_idle(&t->thread_enc);
		ba_transport_thread_state_set_idle(&t->thread_dec);
		acquired = true;
	}
```

neslinesli93, 2023-04-15:
Thank you for the quick reply. I tried your fix and the result was virtually the same as before: `aplay` worked two times and then it got stuck. Here are the logs:  [logs_aplay2.txt](https://github.com/arkq/bluez-alsa/files/11240582/logs_aplay2.txt)

Disconnecting and reconnecting to the speaker does not change anything, I always have to reboot. Not even restarting `bluealsa.service` fixes it :(

arkq, 2023-04-15:
OK, I'll look deeper into that issue...
Restarting `bluealsa.service` should have fixed the "stuck" issue. However, it's quite possible that bluealsa service is not restarted. Simply kill it with -9: `pkill -9 -x bluealsa`

neslinesli93, 2023-04-15:
You were right, killing the daemon fixed the issue! It still got stuck after a couple `aplay` runs, but at least it didn't require a reboot. It was kind of strange as the `bluetoothctl connect` command took almost 30 seconds to run, and it even logged an error before connecting. I'll add the logs again, as I don't know if they are useful or not... [logs_pkill_connect.txt](https://github.com/arkq/bluez-alsa/files/11240646/logs_pkill_connect.txt)

arkq, 2023-04-24:
Please try the current master branch. If the issue is not fixed, please reopen it.

PR #632 (2023-04-24): fix A2DP source acquistion on slow machines
My fix for the thread synchronization problem mentioned here https://github.com/arkq/bluez-alsa/pull/618#issuecomment-1483938339

Hopefully it also fixes #627 

I had to special-case the A2DP source profile because I was unable to find a more general solution.

Tested by inserting a call to usleep(3) to simulate a slow host:

```diff
diff --git a/src/ba-transport.c b/src/ba-transport.c
index d26ffef..cc82b45 100644
--- a/src/ba-transport.c
+++ b/src/ba-transport.c
@@ -1633,6 +1633,8 @@ int ba_transport_acquire(struct ba_transport *t) {
 final:
 	pthr [... cut]