arkq/bluez-alsa #628: Illegal Instruction on PiZero

gearhead, 2023-04-15:
### Problem
I cannot run bluealsa compiled on 32 bit RPiOS (armhf) on armv6 hardware. I built this binary on a Pi4 running the armv7l kernel. This binary runs correctly on a Pib+ running the armv7 kernel but fails when running on the armv6 hardware.  

### Reproduction steps

Build and install bluealsa and run 'bluealsa --version'
Armv6:
```
# uname -a
Linux rpizero 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
#bluealsa -V
Illegal instruction
```
Armv7:
```
# uname -a
Linux rpi3b 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
# bluealsa -V
6837b54
```

### Setup
This binary was built on a Pi4 running the kernel7l.img
`Linux raspberrypi 6.1.19-v7l+ #1637 SMP Tue Mar 14 11:07:55 GMT 2023 armv7l GNU/Linux `

I use makedeb and distccd to build the binaries and have had success with many binaries I have built but this one is a problem.

The flags I use to build the binary are:
```
 flags=(--with-libopenaptx
         --enable-ldac
         --enable-mp3lame
         --enable-cli
         --enable-aac
         --enable-manpages
         --enable-systemd
         --enable-aptx-hd
         --enable-faststream)
  ./configure --prefix=/usr \
      --sysconfdir=/etc \
      "${flags[@]}"
  make
```

The compiler flags for distccd are:
```
CFLAGS="-march=armv6 -mfloat-abi=hard -mfpu=vfp -O2 -pipe -fstack-protector-strong -fno-plt -fexceptions \
        -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security \
        -fstack-clash-protection"
CXXFLAGS="$CFLAGS -Wp,-D_GLIBCXX_ASSERTIONS"
LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"
#RUSTFLAGS="-C opt-level=2"
#-- Make Flags: change this for DistCC/SMP systems
MAKEFLAGS="-j5"
#-- Debugging flags
DEBUG_CFLAGS="-g -fvar-tracking-assignments"
DEBUG_CXXFLAGS="-g -fvar-tracking-assignments"
```
Associated packages
```
bluetoothd --version
5.55
aplay --version
aplay: version 1.2.4 by Jaroslav Kysela <perex@perex.cz>
```
The same versions of bluetoothd and aplay are on the image I created and this image runs fine on a pi3b.

### Additional context

borine, 2023-04-16:
Could you try running
```
gdb -ex run --args bluealsa -V
```
That should tell us in which library the illegal instruction is occurring.

Also you do not say which bluealsa sources you are using.

gearhead, 2023-04-16:
Borine,

Sorry, I thought the version gave the info. I am building front got, the
latest commit from when I pulled it last week. That is what shows on the
version.

When I run gdb, I get this
'''
# gdb -ex run --args bluealsa -V
GNU gdb (Raspbian 10.1-1.7) to 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later
<http://gnu.org/licenses/gpl.html>This
is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bluealsa...
(No debugging symbols found in bluealsa)
Starting program: /usr/bin/bluealsa -V
[Thread debugging using libthread_db enabled]
Using host libthread_db library
"/lib/arm-linux-gnueabihf/libthread_db.so.1".
Program received signal SIGILL, Illegal instruction.
0xb6ec6490 in ?? () from /lib/arm-linux-gnueabihf/libfdk-aac.so.2
'''
They installed version is:
'''
libfdk-aac-dev/now 2.0.1-1 armhf [installed,local]
libfdk-aac2/now 2.0.1-1 armhf [installed,local]
'''

borine, 2023-04-16:
... and what is the output of (on the pi zero)
```
readelf -A /lib/arm-linux-gnueabihf/libfdk-aac.so.2
```

gearhead, 2023-04-16:
@borine 
```
# readelf -A /lib/arm-linux-gnueabihf/libfdk-aac.so.2
Attribute Section: aeabi
File Attributes
  Tag_CPU_name: "7-A"
  Tag_CPU_arch: v7
  Tag_CPU_arch_profile: Application
  Tag_ARM_ISA_use: Yes
  Tag_THUMB_ISA_use: Thumb-2
  Tag_FP_arch: VFPv3-D16
  Tag_ABI_PCS_wchar_t: 4
  Tag_ABI_FP_denormal: Needed
  Tag_ABI_FP_exceptions: Needed
  Tag_ABI_FP_number_model: IEEE 754
  Tag_ABI_align_needed: 8-byte
  Tag_ABI_align_preserved: 8-byte, except leaf SP
  Tag_ABI_enum_size: int
  Tag_ABI_VFP_args: VFP registers
  Tag_CPU_unaligned_access: v6
```
Just to note, I did not build this library, but installed it via apt and have no other 2nd party sources except to add the upmpdcli binary. This is not added on my build machine, I do not think...

Keith

borine, 2023-04-16:
> Tag_CPU_arch: v7

That answers your question. libfdk-aac is built for armv7 and cannot be loaded on an armv6 architecture machine

gearhead, 2023-04-16:
so, how do we move forward? A special version of libfdk-aac for armv6? How would we go about that? I guess I could beuild a verison on armv6 then put it especially on when the image is loaded on v6 hardware? Or can libfdk-aac be built in a different manner so that it will run on all Pis (v6 and v7)? Since the image we are using is the 'ALL' version which should run on all Pis, I would think that this question may be for the RPi OS group...

borine, 2023-04-16:
> I would think that this question may be for the RPi OS group...

I would think that I agree :laughing:

gearhead, 2023-04-18:
Posted:
[RPiOS Forum Post](https://forums.raspberrypi.com/viewtopic.php?t=350372&sid=777b7762ffab40df0cc59669e72a87f8)

borine, 2023-05-01:
This issue is caused by a problem outside of the control of this project. Not a BlueALSA issue, closing.

closed 2023-05-01 by borine