arkq/bluez-alsa #629: No such device (Bluetooth, IPhone)

daniil-loban, 2023-04-16:
### Problem

I can't  record file from audio on my IPhone by this [instrution](https://github.com/arkq/bluez-alsa)  because: `No such device` 

### Reproduction steps

```
$ arecord -D bluealsa capture.wav
D: bluealsa-pcm.c:895: Getting BlueALSA PCM: CAPTURE 4C:AB:4F:E4:8B:B9 a2dp
ALSA lib bluealsa-pcm.c:897:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: Rejected send message, 1 matched rules; type="method_call", sender=":1.778" (uid=1000 pid=44527 comm="arecord -D bluealsa capture.wav " label="unconfined") interface="org.bluealsa.Manager1" member="GetPCMs" error name="(unset)" requested_reply="0" destination="org.bluealsa" (uid=0 pid=38735 comm="/usr/bin/bluealsa " label="unconfined")
arecord: main:831: audio open error: No such device
```
```
$ cat ~/.asoundrc 
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "4C:AB:4F:E4:8B:B9"
defaults.bluealsa.profile "a2dp"
```

```
$ bluetoothctl
Agent registered
[CHG] Device 4C:AB:4F:E4:8B:B9 ServicesResolved: no
[CHG] Device 4C:AB:4F:E4:8B:B9 Connected: no
[CHG] Device 4C:AB:4F:E4:8B:B9 Connected: yes
[CHG] Device 4C:AB:4F:E4:8B:B9 ServicesResolved: yes
```

### Setup

OS distribution and version: `Kubuntu 22.04`
bluealsa: `v3.0.0`
bluetoothd: `5.64`
aplay: `aplay: version 1.2.6 by Jaroslav Kysela <perex@perex.cz>`

arkq, 2023-04-16:
> Couldn't get BlueALSA PCM: Rejected send message

It's quire possible that that part is missing from the documentation... The problem is with D-Bus configuration. Make sure that you've got a file like this:

```
> cat /etc/dbus-1/system.d/bluealsa.conf
  <policy user="root">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>
  <policy group="audio">   <!-- allow user in "audio" group to connect with bluealsa service -->
    <allow send_destination="org.bluealsa"/>
  </policy>

  <!-- this is optional -->
  <policy user="myuser">   <!-- allow user "myuser" to connect with bluealsa service -->
    <allow send_destination="org.bluealsa"/>
  </policy>
  <!-- this is optional -->

</busconfig>
```

It might happen that the rules will not be applied automatically, so in that case reboot.

arkq, 2023-05-06:
@daniil-loban is this issue resolved or you need more help?

closed 2023-06-19 by arkq