arkq/bluez-alsa #635: connected airpods pro2, the music is played without sound

luoyuyi111, 2023-05-15:
Hello, I have a problem. I try to ask if you have the same problem. (bluez5.54 + bluealsa3.1.0)

The device is connected to airpods pro2 and plays music, but there is no sound. Strangely, only this airpods pro2 appeared.
I got the air log and hci log, the audio is normal.

arkq, 2023-05-15:
Try, latest release or better yet current master. Due to lack of resources, old releases are not maintained on per-release bases.

Also, which codec you are using with your airpods2?

```sh
bluealsa-cli list-pcms
bluealsa-cli info <PCM-path>
```

luoyuyi111, 2023-05-15:
> Try, latest release or better yet current master. Due to lack of resources, old releases are not maintained on per-release bases.
> 
> Also, which codec you are using with your airpods2?
> 
> ```shell
> bluealsa-cli list-pcms
> bluealsa-cli info <PCM-path>
> ```

I'm used SBC codec. I tried to update to current master code  and got a sound. So 
  ,this problem is the reason why the bluealsa version is too low.

I found a discrepancy to be more likely, this commit: https://github.com/arkq/bluez-alsa/commit/f105eb58dcc2cab56a6416b1bad115d3f45d08fd

arkq, 2023-05-15:
> I found a discrepancy to be more likely, this commit: https://github.com/arkq/bluez-alsa/commit/f105eb58dcc2cab56a6416b1bad115d3f45d08fd

I'm not sure I get you correctly. Does it mean that everything works as expected AFTER that commit or BEFORE?
Because, as far as SBC codec is concerned, the https://github.com/arkq/bluez-alsa/commit/f105eb58dcc2cab56a6416b1bad115d3f45d08fd commit does not change anything (I think)...

luoyuyi111, 2023-05-16:
> 

this commit mainly changes the algorithm for the timestamp variable。
The old calculation（V3.1.0） : 
```
timestamp += pcm_frames * 10000 / samplerate;
```
This violates the Bluetooth spec, Now change it to:
```
timestamp += pcm_frames;
```
Validate pass
Thanks

arkq, 2023-05-16:
> this commit mainly changes the algorithm for the timestamp variable

Yes, yes. I've made that fix some time ago and I thought (I couldn't recollect exactly which codecs were affected) that the SBC was OK before that change. But apparently it was not :D

So, I'm closing this issue as resolved.

closed 2023-05-16 by arkq