arkq/bluez-alsa #649: changing a2dp codec broken

borine, 2023-07-04:
### Problem

commit fc940df9d4b9638d79aa64eb4c3c51afd243dc08 appears to have broken the ability to change A2DP codec.

### Reproduction steps

build:
```
./configure --enable-debug --enable-test --enable-rfcomm --enable-cli --enable-msbc --enable-aptx --with-libopenaptx --enable-faststream --enable-manpages --enable-systemd --enable-ofono --enable-aac
```

service command line:
```
sudo ./src/bluealsa -p a2dp-source -p hfp-ag -p hsp-ag -c aptx
```

sequence to test codec change:
```
$ bluealsa-cli codec /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink
Available codecs: SBC AAC aptX
Selected codec: AAC
$ bluealsa-cli codec /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink aptx
$ bluealsa-cli codec /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink
Available codecs: SBC AAC aptX
Selected codec: aptX
$ bluealsa-cli codec /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink aac
$ bluealsa-cli codec /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink
Available codecs: SBC AAC aptX
Selected codec: AAC
$ bluealsa-cli codec /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink aptx
bluealsa-cli: [83686] E: ../../../utils/cli/cmd-codec.c:109: CMD "codec": Couldn't select BlueALSA PCM Codec: Input/output error
```

bluealsa service log:
```
bluealsa: [83640] D: ../../src/ofono.c:1090: Checking oFono service presence
bluealsa: [83640] D: ../../src/a2dp-aac.c:141: FDK-AAC lib capabilities: dec:0x1a4ffff enc:0x410b1
bluealsa: [83640] D: ../../src/storage.c:84: Initializing persistent storage: /var/lib/bluealsa
bluealsa: [83640] D: ../../src/main.c:604: Starting main dispatching loop
bluealsa: [83640] D: ../../src/main.c:114: Acquired D-Bus service name: org.bluealsa
bluealsa: [83640] D: ../../src/bluealsa-dbus.c:391: Registering BlueALSA D-Bus manager: /org/bluealsa
bluealsa: [83640] D: ../../src/bluez.c:185: Registering battery provider: /org/bluez/hci0/battery
bluealsa: [83640] D: ../../src/bluez.c:130: Registering media application: /org/bluez/hci0
bluealsa: [83640] D: ../../src/bluez.c:705: Exporting media endpoint object: /org/bluez/hci0/A2DP/aptX/source/1
bluealsa: [83640] D: ../../src/bluez.c:705: Exporting media endpoint object: /org/bluez/hci0/A2DP/aptX/source/2
bluealsa: [83640] D: ../../src/bluez.c:705: Exporting media endpoint object: /org/bluez/hci0/A2DP/AAC/source/1
bluealsa: [83640] D: ../../src/bluez.c:705: Exporting media endpoint object: /org/bluez/hci0/A2DP/AAC/source/2
bluealsa: [83640] D: ../../src/bluez.c:705: Exporting media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: [83640] D: ../../src/bluez.c:705: Exporting media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: [83640] D: ../../src/bluez.c:1087: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: [83640] D: ../../src/bluez.c:1015: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: [83640] D: ../../src/bluez.c:1087: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: [83640] D: ../../src/bluez.c:1015: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: [83640] D: ../../src/bluez.c:152: BlueZ battery provider support not available
bluealsa: [83640] D: ../../src/dbus.c:47: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/AudioGateway
bluealsa: [83640] D: ../../src/storage.c:117: Loading storage: /var/lib/bluealsa/5C:F3:70:9B:FB:1A
bluealsa: [83660] D: ../../src/ba-rfcomm.c:1192: Starting RFCOMM loop: HFP Audio Gateway (...)
bluealsa: [83640] D: ../../src/ba-rfcomm.c:1649: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (...)
bluealsa: [83640] D: ../../src/sco.c:219: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: [83640] D: ../../src/bluez.c:930: HFP Audio Gateway (...) configured for device 5C:F3:70:9B:FB:1A
bluealsa: [83661] D: ../../src/sco.c:90: Starting SCO dispatcher loop: hci0
bluealsa: [83660] D: ../../src/at.c:179: AT message: SET: command:+BRSF, value:756
bluealsa: [83660] D: ../../src/ba-rfcomm.c:319: AG features [2784]: reject-call, enhanced-call-status, enhanced-call-control, codec-negotiation, esco-s4-settings
bluealsa: [83660] D: ../../src/ba-rfcomm.c:331: HF features [756]: cli-presentation, volume-control, enhanced-call-status, enhanced-call-control, codec-negotiation, esco-s4-settings
bluealsa: [83660] D: ../../src/ba-rfcomm.c:109: Sending AT message: RESP: command:+BRSF, value:2784
bluealsa: [83660] D: ../../src/ba-rfcomm.c:109: Sending AT message: RESP: command:(null), value:OK
bluealsa: [83660] D: ../../src/ba-rfcomm.c:128: RFCOMM: HFP Audio Gateway (...) state transition: 0 -> 2
bluealsa: [83660] D: ../../src/at.c:179: AT message: SET: command:+BAC, value:1,2
bluealsa: [83660] D: ../../src/ba-rfcomm.c:109: Sending AT message: RESP: command:(null), value:OK
bluealsa: [83660] D: ../../src/ba-rfcomm.c:128: RFCOMM: HFP Audio Gateway (...) state transition: 2 -> 3
[... 69 lines cut ...]
bluealsa: [83668] D: ../../src/dbus.c:47: Called: org.bluealsa.PCM1.SelectCodec() on /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink
bluealsa: [83640] D: ../../src/dbus.c:47: Called: org.bluez.MediaEndpoint1.ClearConfiguration() on /org/bluez/hci0/A2DP/aptX/source/1
bluealsa: [83640] D: ../../src/bluez.c:581: Disconnecting media endpoint: /org/bluez/hci0/A2DP/aptX/source/1
bluealsa: [83640] D: ../../src/bluez.c:1382: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved(/org/bluez/hci0/dev_5C_F3_70_9B_FB_1A/sep1/fd32, ...)
bluealsa: [83640] D: ../../src/bluez.c:1382: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved(/org/bluez/hci0/dev_5C_F3_70_9B_FB_1A/player0, ...)
bluealsa: [83640] D: ../../src/bluez.c:1280: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_5C_F3_70_9B_FB_1A/sep3/fd33, ...)
bluealsa: [83640] D: ../../src/dbus.c:47: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/AAC/source/1
bluealsa: [83640] D: ../../src/bluez.c:540: A2DP Source (AAC) configured for device 5C:F3:70:9B:FB:1A
bluealsa: [83640] D: ../../src/bluez.c:543: A2DP selected configuration blob [len=6]: 400014035b60
bluealsa: [83640] D: ../../src/bluez.c:545: PCM configuration: channels: 2, sampling: 96000
bluealsa: [83640] D: ../../src/ba-transport.c:1195: Freeing transport: A2DP Source (aptX)
bluealsa: [83668] D: ../../src/dbus.c:47: Called: org.bluealsa.PCM1.GetCodecs() on /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink
bluealsa: [83640] D: ../../src/bluez.c:1280: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_5C_F3_70_9B_FB_1A/player0, ...)
bluealsa: [83668] D: ../../src/dbus.c:47: Called: org.bluealsa.PCM1.SelectCodec() on /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink
bluealsa: [83668] E: ../../src/ba-transport.c:1290: Couldn't set A2DP configuration: GDBus.Error:org.bluez.Error.InvalidArguments: Invalid arguments in method call
bluealsa: [83668] E: ../../src/bluealsa-dbus.c:756: Couldn't select codec: aptX: Input/output error
```
### Setup

OS: Ubuntu 22.04 LTS
`bluealsa --version` v4.1.1-3-gfc940df
`bluetoothd --version` 5.64
`git log -1 --oneline` fc940df (HEAD) Export media endpoints by registering application

##

arkq, 2023-07-04:
Yes, indeed... I've not tested changing codec back and forth few times.... I'll try to fix it right away.

arkq, 2023-07-04:
From logs, I can see that upgrade to BlueZ 5.68 should fix the issue. Anyway, I'll try to fix it for BlueZ < 5.67.

arkq, 2023-07-04:
@borine, please check branch "sepindex" and let me know whether now it works for you.

borine, 2023-07-04:
Yes, the sepindex branch allows me to repeatedly change the codec as many times as I choose. Thanks!

commit 07f5aad2 (2023-07-04): When changing SEP select one with the lowest index

This commit fixes issue with BlueZ < 5.67 where it was not possible to
register new A2DP endpoint with already registered application. In that
case newly exported endpoint is in fact not registered within BlueZ, so
it returns "Invalid arguments" in case someone wants to use it. Now, we
will use pre-registered endpoints before using new ones, so in the most
sample case everything should work fine even with BlueZ < 5.67.

Fixes #649
closed 2023-07-04 by arkq