arkq/bluez-alsa #651: Build error undefined references

skidoo23, 2023-07-10:
> Please read the [troubleshooting guide](../blob/master/TROUBLESHOOTING.md)
> before raising a new issue.

### Problem

Build error

```
/usr/bin/ld: bluealsa-dbus.o: in function `bluealsa_dbus_register':
/home/skidoo23/src/bluez-alsa/build/src/../../src/bluealsa-dbus.c:394: undefined reference to `org_bluealsa_manager1_skeleton_new'
/usr/bin/ld: bluealsa-dbus.o: in function `bluealsa_dbus_pcm_register':
/home/skidoo23/src/bluez-alsa/build/src/../../src/bluealsa-dbus.c:924: undefined reference to `org_bluealsa_pcm1_skeleton_new'
/usr/bin/ld: bluealsa-dbus.o: in function `bluealsa_dbus_rfcomm_register':
/home/skidoo23/src/bluez-alsa/build/src/../../src/bluealsa-dbus.c:1044: undefined reference to `org_bluealsa_rfcomm1_skeleton_new'
/usr/bin/ld: ofono.o: in function `ofono_register':
/home/skidoo23/src/bluez-alsa/build/src/../../src/ofono.c:892: undefined reference to `org_ofono_handsfree_audio_agent_skeleton_new'
```

> A clear and concise description of what the bug is.
> If possible, please check if the bug still exists in the master branch (or
> the latest release if you are not using it already).

### Reproduction steps

> Provide a minimal example of how to reproduce the problem. State `bluealsa`
> command line arguments and the content of .asoundrc file (if PCM alias with
> "type bluealsa" was added to that file).

```
$ make distclean
$ autoreconf --install
$ cd build
$ ../configure --enable-aac --enable-ofono --enable-rfcomm
$ make
```

### Setup

> - the OS distribution and version
Linux test 6.3.0-1-686-pae #1 SMP PREEMPT_DYNAMIC Debian 6.3.7-1 (2023-06-12) i686 GNU/Linux

> - the version of BlueALSA (`bluealsa --version`)
Recent version from github.

> - the version of BlueZ (`bluetoothd --version`)
5.66

> - the version of ALSA (`aplay --version`)
aplay: version 1.2.9 by Jaroslav Kysela <perex@perex.cz>

> - if self-built from source, please state the branch and commit
> (`git log -1 --oneline`), and the used configure options.
07f5aad (HEAD -> master, origin/master, origin/HEAD) When changing SEP select one with the lowest index

### Additional context

N/A

borine, 2023-07-11:
I think this is probably a side-effect of commit 23f249e. To confirm, please could you try (in the top-level bluez-alsa directory)
```
find . -name bluealsa-iface.c
```
The file should be found at `./build/src/bluealsa-iface.c`

If you have it at `./src/bluealsa-iface.c` then this is because this work area was previously used to build an earlier bluez-alsa version. My work-around was to delete the iface.c files from ./src, then make distclean and try again:
```
cd bluez-alsa
rm src/*-iface.c
cd build
make distclean
cd ..
autoreconf -fi
cd build
../configure --enable-aac --enable-ofono --enable-rfcomm
make
```

If the `find` command did not find any instance of bluealsa-iface.c, or if it was found (only) in `./src/build/` then there is some other problem which will require more investigation.

skidoo23, 2023-07-11:
`rm src/*-iface.c` did the trick :) Thank you.

skidoo23, 2023-07-11:
Case closed.

closed 2023-07-11 by skidoo23