arkq/bluez-alsa #659: LDAC problem after update from 4.0 to 4.1.1

KMartus, 2023-08-06:
After update from 4.0 to 4.1.1 LDAC doesn't work. SBC and AAC are ok.

System: Manjaro Linux, RPI 4.0, aarch64, bluetooth dongle ID 2550:8761 Realtek Bluetooth Radio

info about error from log:

_ba-transport.c:1290: Couldn't set A2DP configuration: Extra A2DP endpoint not available
bluealsa-dbus.c:757: Couldn't select codec: LDAC: Input/output error_

Tested with:
- headphones Sony WX-1000XM5
- speaker Huawei Sound-01941

arkq, 2023-08-06:
Can you provide full debug logs (project configured with `--enable-debug`) from bluealsa service?

KMartus, 2023-08-07:
bluealsa: [45420] D: bluez.c:1407: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: [46079] D: ba-transport.c:1983: Exiting IO thread [ba-a2dp-sbc]: A2DP Source (SBC)
bluealsa: [46084] D: dbus.c:47: Called: org.bluealsa.PCM1.SelectCodec() on /org/bluealsa/hci0/dev_10_5D_DC_7B_2C_D4/a2dpsrc/sink
bluealsa: [46084] E: ba-transport.c:1290: Couldn't set A2DP configuration: Extra A2DP endpoint not available
bluealsa: [46084] E: bluealsa-dbus.c:757: Couldn't select codec: LDAC: Input/output error
bluealsa: [46084] D: dbus.c:47: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_10_5D_DC_7B_2C_D4/a2dpsrc/sink
bluealsa: [46084] D: ba-transport.c:741: New A2DP transport: 12
bluealsa: [46084] D: ba-transport.c:742: A2DP socket MTU: 12: R:672 W:1008
bluealsa: [45420] D: bluez.c:1407: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: [45420] D: ba-transport.c:1528: Starting transport: A2DP Source (SBC)
bluealsa: [45420] D: ba-transport.c:448: Created BT socket duplicate: [12]: 13
bluealsa: [46085] D: codec-sbc.c:262: SBC setup: 44100 Hz DualChannel allocation=SNR blocks=16 sub-bands=8 bit-pool=47 => 551250 bps
bluealsa: [46085] D: a2dp-sbc.c:222: IO loop: START: a2dp_sbc_enc_thread: A2DP Source (SBC)
bluealsa: [45420] D: ba-transport.c:1949: Created new IO thread [ba-a2dp-sbc]: A2DP Source (SBC)

arkq, 2023-08-07:
Sorry, I did not say it clearly. I'd like to see log from the very beginning of bluealsa service start/restart. I'd like to see the moment when the headset connects, because currently I've got no headset with LDAC support... Also, you can provide the output from the `bluealsa-cli info /org/bluealsa/hci0/dev_10_5D_DC_7B_2C_D4/a2dpsrc/sink`.

KMartus, 2023-08-07:
with -c ldac everything is ok, previously it worked without this option
now my command line looks like this:

/usr/bin/bluealsa -S --profile=a2dp-source --a2dp-volume --a2dp-force-audio-cd --sbc-quality=xq+ -c ldac --ldac-abr --aac-afterburner

so I can use all codecs SBC (all profiles), AAC, LDAC (all profiles)

arkq, 2023-08-07:
> After update from 4.0 to 4.1.1 LDAC doesn't work. SBC and AAC are ok.
> with -c ldac everything is ok, previously it worked without this option

Are you sure that you've upgraded from 4.0.0 ? Because the requirement to enable all vendor codecs on the command line was introduced in 4.0.0.

KMartus, 2023-08-07:
100% sure
now I see that default codecs are SBC and AAC

KMartus, 2023-08-07:
Sure. THX

closed 2023-08-07 by arkq