arkq/bluez-alsa #663: v4.1.1 fails to build on on Debian buster i386

D1eter, 2023-08-26:
**Thanks for bluez-alsa, much appreciated!**

### Problem
FYI: v4.1.1 fails to build on on Debian buster i386:

make[2]: Entering directory '/home/ds/git/bluez-alsa.v4.1.1/build/src'
/usr/bin/gdbus-codegen --interface-info-body --output bluez-iface.c ../../src/bluez-iface.xml
Traceback (most recent call last):
  File "/usr/bin/gdbus-codegen", line 55, in <module>
    sys.exit(codegen_main.codegen_main())
  File "/usr/share/glib-2.0/codegen/codegen_main.py", line 296, in codegen_main
    gen.generate()
  File "/usr/share/glib-2.0/codegen/codegen.py", line 901, in generate
    self.define_infos()
  File "/usr/share/glib-2.0/codegen/codegen.py", line 864, in define_infos
    key = (m.since, '%s_interface__%s_property' % (i.name_lower, p.name_lower))
UnboundLocalError: local variable 'm' referenced before assignment
make[2]: *** [Makefile:1177: bluez-iface.c] Error 1
make[2]: Leaving directory '/home/ds/git/bluez-alsa.v4.1.1/build/src'
make[1]: *** [Makefile:461: all-recursive] Error 1
make[1]: Leaving directory '/home/ds/git/bluez-alsa.v4.1.1/build'
make: *** [Makefile:393: all] Error 2

I've attached bluez-iface.xml from the failing v4.1.1 build.

Versions v3.1.0, v4.0.0 and v4.1.0 build and work fine.

Note: To build on debian buster (i386) it is necessary to install bluez >= 5.50 from buster/backports and sbc >= 1.5 from source

Let me know if if I can provide anything else to help fix this.

[bluez-iface.xml.txt](https://github.com/arkq/bluez-alsa/files/12446729/bluez-iface.xml.txt)

borine, 2023-08-27:
Yes, BlueALSA v4.1.1 requires libglib2.0 version 2.64 or later in order to generate the D-Bus interface code. This should be documented, I will update the wiki page.

See the option `--glib-min-required=` in the gdbus-codegen manual page for more detail on what is wrong (Bluez and BlueALSA both use file descriptor parameters)

D1eter, 2023-08-27:
Okay, thanks. v4.1.0 will do just fine.

D1eter, 2023-08-27:
Closing as there is actually no issue.

borine, 2023-08-29:
On further investigation, it turns out this is an old bug in gdbus-codegen; The fix appeared in glib2.0 release 2.61.0 - see
https://github.com/GNOME/glib/commit/56492553515ec60eea847edf45c686fef255d6fe

@arkq - One possible workaround for BueALSA to work with older glib2.0 versions would be to move the `BatteryProvider1` interface declaration lower down in the bluez-iface.xml file, so that the first interface is `MediaEndpoint1` which does have methods and therefore this bug would not be triggered. The same applies to the `Manager1` interface in `bluealsa-iface.xml`

@D1eter - could you possibly try that workaround with BlueALSA 4.1.1 on your buster installation to check that it really does fix this issue?

D1eter, 2023-08-29:
Sure, just tell me what to do. Note that Debian buster has glib2.0 release 2.58.3. Will that do?

borine, 2023-08-29:
> glib2.0 release 2.58.3. Will that do?
Yes, any release before 2.61 has the bug linked above.

To test this workaround, apply this patch before running configure:
```diff
diff --git a/src/bluealsa-iface.xml b/src/bluealsa-iface.xml
index ef14291..d81f173 100644
--- a/src/bluealsa-iface.xml
+++ b/src/bluealsa-iface.xml
@@ -7,13 +7,6 @@
 -->
 <node>
 
-	<interface name="org.bluealsa.Manager1">
-		<property name="Version" type="s" access="read"/>
-		<property name="Adapters" type="as" access="read"/>
-		<property name="Profiles" type="as" access="read"/>
-		<property name="Codecs" type="as" access="read"/>
-	</interface>
-
 	<interface name="org.bluealsa.PCM1">
 		<method name="Open">
 			<arg direction="out" type="h" name="fd_pcm"/>
@@ -41,6 +34,13 @@
 		<property name="Volume" type="q" access="readwrite"/>
 	</interface>
 
+	<interface name="org.bluealsa.Manager1">
+		<property name="Version" type="s" access="read"/>
+		<property name="Adapters" type="as" access="read"/>
+		<property name="Profiles" type="as" access="read"/>
+		<property name="Codecs" type="as" access="read"/>
+	</interface>
+
 	<interface name="org.bluealsa.RFCOMM1">
 		<method name="Open">
 			<arg direction="out" type="h" name="fd"/>
diff --git a/src/bluez-iface.xml b/src/bluez-iface.xml
index 83568ce..7f46eba 100644
--- a/src/bluez-iface.xml
+++ b/src/bluez-iface.xml
@@ -7,12 +7,6 @@
 -->
 <node>
 
-	<interface name="org.bluez.BatteryProvider1">
-		<property name="Device" type="o" access="read"/>
-		<property name="Percentage" type="y" access="read"/>
-		<property name="Source" type="s" access="read"/>
-	</interface>
-
 	<interface name="org.bluez.MediaEndpoint1">
 		<method name="SelectConfiguration">
 			<arg direction="in" type="ay" name="capabilities"/>
@@ -29,6 +23,12 @@
 		</method>
 	</interface>
 
+	<interface name="org.bluez.BatteryProvider1">
+		<property name="Device" type="o" access="read"/>
+		<property name="Percentage" type="y" access="read"/>
+		<property name="Source" type="s" access="read"/>
+	</interface>
+
 	<interface name="org.bluez.Profile1">
 		<method name="NewConnection">
 			<arg direction="in" type="o" name="device"/>
```

If you are not sure how to apply a patch, then instead you can edit the files `src/bluealsa-iface.xml` and `src/bluez-iface.xml` instead. In each file just move the first
```
<interface name=...>
...
</interface>
```
block so that it becomes the second interface block.

D1eter, 2023-08-29:
With the patch v4.1.1 builds without problems. Listening to v4.1.1 just now. Thanks!

arkq, 2023-08-29:
@borine many thanks for finding a cause of that, and many thanks to @D1eter for testing.
Regarding the glib version, it seems that anyway I will have to bump minimal requirement to at least 2.57.2, because it's the first version when `gdbus-codegen` has got `--interface-info-body` option... I will check whether everything builds correctly with that version.

commit ed24c9c0 (2023-08-29): Bump glib library minimal version to 2.58.2

Fixes #663
closed 2023-08-29 by arkq