arkq/bluez-alsa #666: No sound from Onkyo receiver

spamtree, 2023-09-16:
### Problem

No sound from Onkyo receiver with bluealsa. Onkyo receiver
works fine with blueman and gnome bluetooth settings. With blueman had to
change sound from AAC to SBC to enable sound. No changes required for
gnome settings. Attempts to play sound through the bluealsa device
and bluealsa plugin  always crashes the playback program and the playback 
program requires kill -KILL to stop the program other methods to stop the program 
do not work. Bluealsa works fine with several different bluetooth speakers. 

### Reproduction steps

    1) Started bluealsa service
    
    2) Used bluetootctl to connect onkyo receiver. 

    3) Attempted play audio with speaker-test. Also tried other programs 
    no success. 

    There is no .asoundrc file installed. 

### Setup

        the OS distribution and version
	    Arch Linux (update rolling release) 
        the version of BlueALSA (bluealsa --version -- v4.1.1)
        the version of BlueZ (bluetoothd --version -- 5.69)
        the version of ALSA (aplay --version -- 1.2.9)
	the version of speaker-test 1.2.9
        if self-built from source, please state the branch and commit
        (git log -1 --oneline), and the used configure options.
	Note: Built for the AUR for arch linux with debugging support.
	git log -1 --oneline -- 78a74d6 Update to 4.1.1
	From the PKGBUILD file from the AUR for arch linux.
	optdepends=( 'lame: build with mp3 support'
             'libbsd: build with hcitop tool'
             'libopenaptx-git: build with libopenaptx for apt-X'
             'mpg123: build with mpg123 decoding support'
             'ncurses: build with hcitop tool'
             'readline: build with bluealsa-rfcomm tool'
             'spandsp: build mSBC codec support' 
             'enable-debug: build debugging support' )

### Add any other context about the problem here, e.g. log messages printed by
bluealsa and/or client application.

Used two different bluetooth devices on two different computers:

```
Bus 001 Device 003: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 001 Device 005: ID 0bda:2550 Realtek Semiconductor Corp. Bluetooth Radio

```

### bluealsa -S --keep-alive=5 -p a2dp-sink

```
bluealsa: [26110] D: a2dp-aac.c:141: FDK-AAC lib capabilities: dec:0x1a4ffff enc:0x410b1
buealsa: [26110] D: storage.c:84: Initializing persistent storage: /usr/var/lib/bluealsa
bluealsa: [26110] W: storage.c:87: Couldn't create storage directory: No such file or directory
bluealsa: [26110] D: main.c:604: Starting main dispatching loop
bluealsa: [26110] D: main.c:114: Acquired D-Bus service name: org.bluealsa
bluealsa: [26110] D: bluealsa-dbus.c:392: Registering D-Bus manager: /org/bluealsa
bluealsa: [26110] D: bluez.c:783: Registering battery provider: /org/bluez/hci0/battery
bluealsa: [26110] D: bluez.c:803: BlueZ battery provider support not available
bluealsa: [26110] D: bluez.c:598: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: [26110] D: bluez.c:509: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: [26110] D: bluez.c:598: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: [26110] D: bluez.c:509: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: [26110] D: bluez.c:598: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/1
[... 258 lines cut ...]
### Onkyo TX-NR676 EE6F60 receiver
```
BLUETOOTH Specs
Communication system
BLUETOOTH Specification version 4.1+LE
Frequency band
2.4 GHz band
Modulation method
FHSS (Freq Hopping Spread Spectrum)
Compatible BLUETOOTH profiles
A2DP 1.2
AVRCP 1.3
HOGP-Host (Client)
HOGP-HID Device (Server)
HID Service (HIDS)
Supported Codecs
SBC
AAC

20 Hz - 20 kHz (Sampling frequency 44.1 kHz)
Maximum communication range
Line of sight approx. 15 m 
```

`

borine, 2023-09-17:
Looking at the server logs, it appears that the Onkyo is dropping the A2DP transport soon after it has been acquired. Perhaps it has a timeout waiting for the first codec frames to arrive after acquisition, or perhaps it just does not like Bluez. Perhaps the bluetooth logs will give more clues as to why that is happening.

The `speaker-test` log is slightly worrying. I have no idea why it hangs. The final "Stopping" message shows that the BlueALSA plugin has sent the thread cancel signal to its IO thread, but the lack of "IO thread cleanup"  after that may indicate that the signal is not received. I cannot explain how that can happen; but, whatever, I think that this is a consequence of the above transport failure and not a cause of it.

spamtree, 2023-09-17:
Thanks for the response. I thought I would report this in case anyone else runs across this problem. If there is any more information needed or you think that is helpful please let me now.

borine, 2023-09-17:
Have you had success with this receiver when sending music from a phone or other OS (mac/windows)? If so, it would be useful to test it with pulseaudio or pipewire to see if they have more success than BlueALSA.

spamtree, 2023-09-18:
I reported that it worked perfectly with other devices in the initial bug report. It works perfectly with ios, android 12/13, gnome Bluetooth setting on PC, and mate (blueman) on PC. On the PC with blueman that worked with the receiver I compiled bluealsa with debugging turned on and had the same result as the pi2. Blueman worked with sound setup for SBC or AAC. Did a quick search - found how to turn on debugging information for blueman and attached is the blueman and bluetooth debug log. Also repeated the bluealsa logs. Had the same results as on the pi2 - could not play any sound thru the bluealsa device to the Onkyo  receiver. System was arch linux setup with pipewire. Pipewire was NOT running while running bluealsa (killed the x-server and used the console).  
[blueman_bluetooth.log](https://github.com/arkq/bluez-alsa/files/12644726/blueman_bluetooth.log)
[blueman.log](https://github.com/arkq/bluez-alsa/files/12644727/blueman.log)
[bluealsa_bluetooth.log](https://github.com/arkq/bluez-alsa/files/12644728/bluealsa_bluetooth.log)
[bluealsa_service.log](https://github.com/arkq/bluez-alsa/files/12644729/bluealsa_service.log)
[bluealsa_speaker-test.log](https://github.com/arkq/bluez-alsa/files/12644730/bluealsa_speaker-test.log)

Files uploaded
blueman.log -- blueman-applet --loglevel debug
blueman_bluetooth.log -- bluetoothd -d (while blueman debug was running)
speaker-test worked fine with blueman running.

bluealsa_speaker-test.log - bluetoothctl, bluealsa-aplay -L, and speaker-test 
    speaker-test locked hard only could stop with pkill -KILL speaker-test
bluealsa_service.log - bluealsa running in debug mode - 
     `/usr/bin/bluealsa -S -p a2dp-source -p a2dp-sink`
bluealsa_bluetooth.log - bluetoothd -d while running bluealsa.

arkq, 2023-09-19:
>  It works perfectly with ios, android 12/13, gnome Bluetooth setting on PC, and mate (blueman) on PC

`blueman` is just a graphical `bluetootctl`, but the underlying Bluetooth audio application might be either PulseAudio or PipeWire. I don't know how to debug them (in order to get meaningful information about differences between them and BlueALSA), but you might try to dump BlueZ communication. Please attach logs collected with such command (from the setup which works with Onkyo):

```
sudo dbus-monitor --system "sender=org.bluez" "destination=org.bluez"
```

Also, you might do the same thing on system with BlueALSA (it will be easier to search for potential differences). If potential issues with A2DP setup will be rulled out, then we will have to look into some differences with RTP audio packets.

borine, 2023-09-19:
I think I can explain the application lockups.

When Bluez unexpectedly changes the transport state to "idle", the BlueALSA daemon releases the Bluetooth socket and stops the encoder I/O thread, but it does not release the client socket and FIFO.

So the plugin I/O thread continues to write samples to the FIFO until the FIFO becomes full, at which point the I/O thread becomes blocked. Meanwhile, the application calls `snd_pcm_drain()`, which waits for the I/O thread to empty the buffer - but that never happens because the I/O thread is blocked. So the application is now blocked in the call to `snd_pcm_drain()`.

What happens now depends on the application. In the case of `speaker-test` a SIGINT from pressing `Ctrl-C` is caught, and the handler sets a flag which terminates the main loop on its next iteration. But the mainloop is blocked, so never gets as far as checking the termination flag.

@arkq Perhaps the `poll()` within `bluealsa_drain()` needs to include a timeout so that the plugin has an opportunity to abort the drain if requested by the application (eg if an application thread or signal handler calls `snd_pcm_abort()` during the call to `snd_pcm_drain()`). I'm not sure what condition to test for here yet, I need to do some more reading.

spamtree, 2023-09-20:
Interesting. I will give this a try in 2-3 days and get back with you the results. Have to deal with life now. 

Thanks for looking into this problem.

spamtree, 2023-09-20:
@arkq - Will also give that a try in 2-3 days. Thanks for the information!

spamtree, 2023-09-26:
No improvement with patch. Same behavior. Here are the logs added aplay and bluez logs. Had to kill aplay and speaker-test and no sound.  Thanks again for looking into this matter. 

[aplay.log](https://github.com/arkq/bluez-alsa/files/12732306/aplay.log)
[speaker-test.log](https://github.com/arkq/bluez-alsa/files/12732307/speaker-test.log)
[bluez-dbus-monitor.log](https://github.com/arkq/bluez-alsa/files/12732308/bluez-dbus-monitor.log)
[bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/12732309/bluetooth.service.log)
[bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/12732310/bluealsa.service.log)

spamtree, 2023-09-26:
@[borine](https://github.com/borine) borine mentioned this issue Sep 25, 2023
Improve PCM plugin drain #667

Should I try this patch? If I try this patch should I remove the other patch?

Thanks.

borine, 2023-09-27:
> Had to kill aplay and speaker-test and no sound.

The "no sound" problem is most likely somewhere in the BlueALSA server, so my patch was not expected to fix that. The patch is supposed to enable you to kill aplay/speaker-test using ctrl-C instead of needing `kill -KILL`. It applies to the plugin shared library, so it is necessary to `make install` so that the patched shared library is placed into the correct directory for aplay or speaker-test to find it. Perhaps I should have placed an extra debug message in there so that we could see if the revised signal handling code is being called when ctrl-C is used. I will try to set up some different tests here to look at it again in case I have made a mistake in the code.

> Improve PCM plugin drain #667
> Should I try this patch? If I try this patch should I remove the other patch?

Sure, you can try it if you wish, it would be good to have feedback from a real test case. You would need to remove the other patch first, and make sure that the new plugin shared library is installed. Again, that PR does not attempt to fix the "no sound" problem. The idea is that the client application can not be stuck in a drain, so that after a certain amount of time the drain will stop and the application will "wake up" without needing to be killed. For speaker-test that may be a couple of seconds, because it uses a very large period size, but for aplay it should be less than half a second. The PR also includes the attempted ctrl-C fix from the first patch, so it should be easy to interrupt speaker-test instead of just waiting for it to finish.

I'm not very experienced at reading the Bluez logs, so that will need to wait until @arkq has time to go through them.

arkq, 2023-09-28:
> bluez-dbus-monitor.log

@spamtree, it seems that you've recorded D-Bus communication when using BlueALSA. Can you please record the same thing in a setup where audio works, e.g. on gnome?

spamtree, 2023-10-02:
Okay I applied the patches from the Improve PCM plugin drain [667](https://github.com/arkq/bluez-alsa/pull/667) and ran the tests again. 

The most interesting development is that sound played for about 5 seconds and stopped. Also ctrl-c worked perfectly. 

Attached are logs - if the filename starts with bluealsa then those are the bluealsa logs and filename starts with gnome are the gnome logs. Sound worked perfectly with gnome. Thanks for looking into the problem. 

[gnome-bluez.log](https://github.com/arkq/bluez-alsa/files/12778625/gnome-bluez.log)
[gnome-bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/12778626/gnome-bluetooth.service.log)

[bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/12778628/bluealsa.service.log)
[bluealsa-speaker-test.log](https://github.com/arkq/bluez-alsa/files/12778629/bluealsa-speaker-test.log)
[bluealsa-bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/12778630/bluealsa-bluetooth.service.log)
[bluealsa-bluez-dbus-monitor.log](https://github.com/arkq/bluez-alsa/files/12778631/bluealsa-bluez-dbus-monitor.log)

arkq, 2023-10-02:
@spamtree I will have to review all the dump files you've provided (thanks for that). I've looked at the D-Bus monitor logs, and there is one difference. Can you please try to run bluealsa system service with `--a2dp-force-audio-cd` option enabled? This will force 44100 Hz sampling rate. Also, check with and without `--codec=-aac`.

spamtree, 2023-10-03:
Here are the logs with --a2dp-force-audio-cd. No sound with speaker-test. Had to hit ctrl-c three times to stop speaker-test. Had to reboot to restart bluetooth.service. 

[bluealsa-bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/12787624/bluealsa-bluetooth.service.log)
[bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/12787625/bluealsa.service.log)
[speaker-test.log](https://github.com/arkq/bluez-alsa/files/12787626/speaker-test.log)
[bluealsa-bluez.log](https://github.com/arkq/bluez-alsa/files/12787627/bluealsa-bluez.log)

Below are the log files with the --codec=-aac and --a2dp-force-audio-cd options. Unable to connect to onkyo receiver. Bluetoothctl keep disconnecting. 

[bluetoothctl-no-acc.log](https://github.com/arkq/bluez-alsa/files/12787647/bluetoothctl-no-acc.log)
[btmon-no-acc.log](https://github.com/arkq/bluez-alsa/files/12787648/btmon-no-acc.log)
[bluealsa-bluez-no-acc.log](https://github.com/arkq/bluez-alsa/files/12787649/bluealsa-bluez-no-acc.log)
[bluealsa-bluetooth.service.no-aac.log](https://github.com/arkq/bluez-alsa/files/12787650/bluealsa-bluetooth.service.no-aac.log)
[bluealsa.service.no-acc.log](https://github.com/arkq/bluez-alsa/files/12787651/bluealsa.service.no-acc.log)

spamtree, 2023-10-03:
I have noted that when bluealsa does connect to the onkyo it always connects with AAC codec and in gnome I must select the SBC codec to enable sound. Removing the AAC codec from bluealsa causes a disconnection with the onkyo receiver. Actually if I select AAC codec in gnome the sound stops and the connection is broken to the onkyo. I have to clear the bluetooth pairing in onkyo receiver and repair again in order to connect the bluetooth after it crashes with AAC codec. Looks like to me the real problem is the onkyo receiver bluetooth is buggy. That is my opinion. 

```
bluealsa-aplay -L
bluealsa:DEV=00:09:B0:1C:CD:D4,PROFILE=a2dp,SRV=org.bluealsa
    Onkyo TX-NR676 EE6F60, trusted audio-card, playback
    A2DP (AAC): S16_LE 2 channels 44100 Hz
```

arkq, 2023-10-03:
If something is wrong with AAC then you can switch to SBC with CODEC option for PCM device. See bluealsa-plugins manual for reference. Start bluealsa service with AAC enabled.

spamtree, 2023-10-03:
Thanks for the suggestion. Got it working see bluealsa-aplay -L

Did not work but it did connect. bluealsa locked completely and had to kill -KILL. Same results. 

```
# bluealsa-aplay -L
bluealsa:DEV=00:09:B0:1C:CD:D4,PROFILE=a2dp,SRV=org.bluealsa
    Onkyo TX-NR676 EE6F60, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 44100 Hz
```

```
speaker-test -D bluealsa -f 44100 -c 2 

speaker-test 1.2.10

Playback device is bluealsa
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
[2347] D: bluealsa-pcm.c:1423: Getting BlueALSA PCM: PLAYBACK 00:00:00:00:00:00 a2dp
[2347] D: bluealsa-pcm.c:1176: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Setting constraints
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 960 to 570653
Period size range from 480 to 285327
Using max buffer size 570652
Periods = 4
[2347] D: bluealsa-pcm.c:546: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing HW
[2347] D: bluealsa-pcm.c:495: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Attempting to fix hw params buffer size
[2347] D: bluealsa-pcm.c:584: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: FIFO buffer size: 1024 frames
[2347] D: bluealsa-pcm.c:589: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Selected HW buffer: 4 periods x 524284 bytes == 2097136 bytes
[2347] D: bluealsa-pcm.c:620: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing SW
[2347] D: bluealsa-pcm.c:620: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing SW
[2347] D: bluealsa-pcm.c:620: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing SW
[2347] D: bluealsa-pcm.c:661: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Prepared
was set period_size = 142663
was set buffer_size = 570652
[2347] D: bluealsa-pcm.c:620: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing SW
 0 - Front Left
[2347] D: bluealsa-pcm.c:620: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing SW
[2347] D: bluealsa-pcm.c:629: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Changing SW avail min: 131071 -> 1
[2347] D: bluealsa-pcm.c:620: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing SW
[2347] D: bluealsa-pcm.c:629: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Changing SW avail min: 1 -> 131071
[2347] D: bluealsa-pcm.c:667: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Draining
[2347] D: bluealsa-pcm.c:374: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Starting
[2350] D: bluealsa-pcm.c:229: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Starting IO loop: 7
ALSA lib bluealsa-pcm.c:730:(bluealsa_drain) Drain timed out - possible Bluetooth transport failure
^C[2347] D: bluealsa-pcm.c:410: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Stopping
[2350] D: bluealsa-pcm.c:165: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: IO thread cleanup
^C[2347] D: bluealsa-pcm.c:661: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Prepared
 1 - Front Right
Time per period = 118.025413
[2347] D: bluealsa-pcm.c:620: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing SW
[2347] D: bluealsa-pcm.c:629: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Changing SW avail min: 131071 -> 1
[2347] D: bluealsa-pcm.c:620: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Initializing SW
[2347] D: bluealsa-pcm.c:629: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Changing SW avail min: 1 -> 131071
[2347] D: bluealsa-pcm.c:667: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Draining
[2347] D: bluealsa-pcm.c:374: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Starting
^C[2347] D: bluealsa-pcm.c:385: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Couldn't start PCM: Resource temporarily unavailable
[2347] D: bluealsa-pcm.c:410: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Stopping
^C[2347] D: bluealsa-pcm.c:410: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Stopping
^C[2347] D: bluealsa-pcm.c:599: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Freeing HW
[2347] D: bluealsa-pcm.c:457: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Closing

```

arkq, 2023-10-03:
> bluealsa locked completely and had to kill -KILL.

By bluealsa you mean the system service (bluealsa application)? If yes, if such lockup happens, can you please try to get backtrace of all threads in that process? Simply try to connects to it with `gdb -p <PID-of-the-process>` and type `thread apply all bt`.

arkq, 2023-10-03:
Also, I would suggest to keep the `--a2dp-force-audio-cd` option all the time. In you first comment you've posted Onkyo receiver spec which says "20 Hz - 20 kHz (Sampling frequency 44.1 kHz)". Why would they mention sampling frequency here? I guess it might be better to stick with 44.1 kHz.

spamtree, 2023-10-03:
Thanks will do. Have to get some work done but will get back to you. Yes the bluealsa.service and can't restart with systemctl restart bluealsa.service. Will get a backtrace. Thank you very much.

borine, 2023-10-04:
Just quickly scanning through https://repairalmostanything.com/thread/767/never-wanted-onkyo-firmware I see that Onkyo receivers use Linux on the chip that performs audio decoding, networking and USB. So it is possible that Bluetooth functionality is provided by Bluez (although other Linux Bluetooth solutions exist). Now Bluez historically has a number of issues related to cacheing of codec infromation. AFAIK they are fixed in Bluez release 5.65, but if your Onkyo is using Bluez then it is very likely to be an older version than that.

The Onyo documentation should mention what open-source software it is using (if any) to comply with the licensing requirements. If it does use Bluez then I think some of the behaviour you see may be because the Onkyo has cached codec info from the first time your computer was paired with it (using Gnome or Blueman), so when you later change the audio agent on your computer you get different available codecs and different codec configurations, but the broken Blluez cache does not pick up on those changes so the transport fails. There may be a workaround, but it would require cllearing the Bluez cache. Unpairing and/or re-pairing a device does not clear the cache on broken releases of Bluez.

borine, 2023-10-04:
BTW
> speaker-test -D bluealsa -f 44100 -c 2 

You need to use 
```speaker-test -D bluealsa -r 44100 -c 2 ```

to force the output sample rate to 44100

borine, 2023-10-05:
Here's my interpretation of what is happening in the last speaker-test log above.

```
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
```

speaker-test is generating pink noise sampled at 48000Hz

```
Using max buffer size 570652
Periods = 4
```

speaker-test has set a period size of 570652 / 4 = 142663 frames.
This translates to 2972 milliseconds at 48000Hz, so bluealsa will use a drain
timeout of just over 3 seconds.

`ALSA lib bluealsa-pcm.c:730:(bluealsa_drain) Drain timed out - possible Bluetooth transport failure`

The bluealsa plugin is unable to drain the buffer because the bluealsa server is not responding. It blocks for the timeout period (just over 3 seconds in this case) then reports that error.

`^C`

I think if you wait 3 seconds here then the `^C` is not necessary; but as a result of this speaker-test sets a flag so that its mainloop is stopped on the next iteration, and future ALSA blocking calls (write, drain) to this PCM return immediately with the error EINTR.

`[2347] D: bluealsa-pcm.c:410: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Stopping`

As a result of the drain timeout, the bluealsa plugin now stops the PCM

`^C`

not necessary - the first ctrl-c told speaker-test to set its stop flag, this one and subsequent presses of ctrl-c have no effect.

`1 - Front Right`

speaker-test ignores the error code returned by the drain function and proceeds to send noise to the second channel. This time it does not call drain() because its stopping flag is set. Had you not pressed ctrl-c earlier, then it is possible speaker-test may become blocked at this point because the attempt to write will block if the buffer is full. If so ctrl-c here will unblock it (but I don't think the buffer would have filled yet). 

`Time per period = 118.025413`

The speaker-test prints timing statistics before leaving its mainloop.

`^C`

same as above

`[2347] D: bluealsa-pcm.c:385: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Couldn't start PCM: Resource temporarily unavailable`

speaker-test tries to drain the PCM one last time before exiting. The bluealsa plugin tries to send a control message to the server to resume the PCM; but the server is no longer reading control messages, and that error results.

`[2347] D: bluealsa-pcm.c:410: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Stopping`

 So now the plugin makes sure that the PCM is stopped, which is a no-op because it is not currently running

`^C`

same as above

`[2347] D: bluealsa-pcm.c:410: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Stopping`

Finally speaker-test closes the PCM. So again the plugin makes sure the PCM is stopped (still a no-op)

`^C`

Same as above

```
[2347] D: bluealsa-pcm.c:599: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Freeing HW
[2347] D: bluealsa-pcm.c:457: /org/bluealsa/hci0/dev_00_09_B0_1C_CD_D4/a2dpsrc/sink: Closing
```

the bluealsa plugin cleans up its resources and then speaker-test exits with exit code 0

spamtree, 2023-10-11:
I will try to address the questions and I followed the recommendations.  Changed settings on speaker-test and bluealsa as recommended. 

It has been about 30 minutes now and speaker-test is hung with the last message being ->

`ALSA lib bluealsa-pcm.c:730:(bluealsa_drain) Drain timed out - possible Bluetooth transport failure`

Had to hit Ctrl-C four times to stop speaker test and was able to restart bluealsa.service after killing the bluealsa process (kill -KILL bluealsa_process). 

dmesg log showed that bluetooth killed the connection to the onkyo. 

`Bluetooth: hci0: killing stalled connection 00:09:b0:1c:cd:d4`

attached are logs. 

[dmesg_hci0.log](https://github.com/arkq/bluez-alsa/files/12864552/dmesg_hci0.log)
[bluez.log](https://github.com/arkq/bluez-alsa/files/12864553/bluez.log)
[bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/12864554/bluetooth.service.log)
[bluealsa.log](https://github.com/arkq/bluez-alsa/files/12864555/bluealsa.log)
[speaker-test.log](https://github.com/arkq/bluez-alsa/files/12864556/speaker-test.log)
[gdb.log](https://github.com/arkq/bluez-alsa/files/12864557/gdb.log)

arkq, 2023-10-11:
Thanks for the gdb.log. Unfortunately, there are no debug symbols there.... Please check if binary was built with debug symbols (with debug_info, not stripped), e.g.:
 ```
$ file build/src/bluealsa
build/src/bluealsa: ELF 64-bit [...] for GNU/Linux 3.2.0, with debug_info, not stripped
```

If it's not, please try to add `-g` to CFLAGS before compiling. If executable has got symbols, please go to the directory from which you've run `make` and from this directory run `gdb -p ....`. This should allow gdb to load symbols and source files to generate verbose backtrace.

borine, 2023-10-11:
> Had to hit Ctrl-C four times to stop speaker test 

@arkq - it looks like the server glib client event loop may be locked up, the speaker-test log above appears to indicate that the plugin is stuck waiting for a reply from the server for the `Drop` command, then again for the `Resume` command,etc. Do you think there should be a timeout in the client code when waiting for a command reply? If so I could push another commit to PR #669

arkq, 2023-10-11:
> Do you think there should be a timeout in the client code when waiting for a command reply?

To be honest I'm not sure. I don't know how long we should wait, 1 second or maybe less? I'd like to try to fix the server first, so such lockup will not happen. But anyway, maybe timeout on the client side is a good idea as well... dunno :D

borine, 2023-10-11:
OK, I'll leave that for further consideration, perhaps I can come up with some rationale for a specific timeout value for each command. I'll also look into the possible usefulness of skipping other actions, in addition to drain, when the ioplug nonbock value indicates an abort request from the application.

spamtree, 2023-10-11:
> Thanks for the gdb.log. Unfortunately, there are no debug symbols there.... Please check if binary was built with debug symbols (with debug_info, not stripped), e.g.:
>
> ```
> $ file build/src/bluealsa [...]

Pretty sure I didn't do the debug symbols and thanks for the tip on how to turn it on. Will do. This is interesting. Learning a lot.

borine, 2023-10-14:
> Bluetooth: hci0: killing stalled connection 00:09:b0:1c:cd:d4

The only relevant search results I get for this error suggest the underlying cause is possibly a kernel bug related to power-save mode, see for example https://bugzilla.kernel.org/show_bug.cgi?id=203535

That does not explain why it is specific to the Onkyo, nor how pipewire can work correctly with the same USB dongle, so it may not be the problem here.

borine, 2023-10-15:
Here's a review of my interpretation so far of the logs, before I leave this alone for a few days.

The encoder io thread ("ba-a2dp-sbc") is blocked in poll() waiting for the BT socket to become writeable. I can find no clue in the logs as to why the socket is blocked.

The client event handler thread ("pool-spawner") is blocked in pthread_cond_wait() waiting for the drain to finish. But the drain never finishes because the encoder thread is blocked.

The main thread ("bluealsa") is blocked in pthread_cond_wait() but there is no clue as to why or where. Debug symbols so that we can see the function call stack are required before any progress can be made on debugging this. Originally I thought this thread might be waiting for the "stopping" flag, but now I see that there are no events in the logs that may have triggered the transport to be stopped. So this is a mystery. Perhaps memory corruption, perhaps the condition variable is actually inside the glib main loop and we are looking at a glib bug, or any other theory you like.

spamtree, 2023-10-15:
More info. Hopefully I did this correctly. I tried to follow the directions given but if I did not collect the right logs or did something incorrect please let me know. Thanks 
[gdb.log](https://github.com/arkq/bluez-alsa/files/12909793/gdb.log)
[speaker-test.log](https://github.com/arkq/bluez-alsa/files/12909794/speaker-test.log)
[bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/12909795/bluetooth.service.log)
[bluealsa.log](https://github.com/arkq/bluez-alsa/files/12909796/bluealsa.log)

I noted that I forgot the the bluez output. I will repeat and send that log also.

spamtree, 2023-10-15:
Okay. I leave the last post up but did not follow the directions correctly. I was supposed to run "gdb" once bluealsa crashed. Well I started gdb session while bluealsa for running (bluealsa did not crash). For the previous post and this post I installed the latest patch from  "Improve PCM plugin drain #667". The good news is bluealsa no longer crashes when attempting to play sound, but still no sound. So I don't have gdb log since bluealsa did not crash. speaker-test did not crash either. 

New logs below. 

[speaker-test.log](https://github.com/arkq/bluez-alsa/files/12909854/speaker-test.log)
[bluealsa.log](https://github.com/arkq/bluez-alsa/files/12909855/bluealsa.log)
[bluez.log](https://github.com/arkq/bluez-alsa/files/12909856/bluez.log)
[bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/12909857/bluetooth.service.log)

borine, 2023-10-15:
> bluealsa did not crash

Actually, it is not locked up, but it has still failed. This isn't so obvious now using speaker-test because speaker-test ignores the error returns from the ALSA PCM and just continues to loop. However, you still get this in the speaker-test log:
```
ALSA lib bluealsa-pcm.c:744:(bluealsa_drain) Drain timed out: Possible Bluetooth transport failure
```
These logs are using AAC, whereas previously we had SBC logs. I don't know how significant that is, but the bluez dbus messaging is now different. I'll read through these in detail next week. You said that pipewire/gnome also fails with AAC, so perhaps we should stick with SBC for testing now so that we can compare logs between pipewire and bluealsa.

For me, it would be easier to follow if aplay was used for testing rather than speaker-test. That way we would have a log of a simple open-write-drain-drop-close sequence from the client to help spot any anomalies. Something like
```
aplay -D bluealsa /usr/share/sounds/alsa/Front_Center.wav
```
Also, if at all possible, building the current master branch sources and using that for testing would make things a little easier, although I appreciate that is not always possible depending on your build and test environment, distribution, etc.

Thanks for sticking with this, I really hate to think that pipewire can succeed where bluealsa fails,

spamtree, 2023-10-15:
That was my mistake I was trying to disable AAC by using .asoundrc.conf but that didn't work. I just noticed it and was about to post the new log. The log is the same as with AAC. Will get some more logs today. 

Thanks for the looking into this matter.

spamtree, 2023-10-15:
In most of the logs when aplay is started device is active and then switches immediately to idle. 

More logs with SBC and aplay. The same effect if the "wav" file is 48000 Hz or 44100 Hz. Expect when playing the 44100 Hz file aplay requires Ctrl-C to return to command prompt. Note the 44100 Hz file was longer than the 48000 Hz file. Not sure if that has any effect, but the bluealsa logs look similar.  

Logs: 

[bluealsa.service-aplay-48000Hz-file.log](https://github.com/arkq/bluez-alsa/files/12910354/bluealsa.service-aplay-48000Hz-file.log)
[bluealsa.service-aplay-44100Hz-file.log](https://github.com/arkq/bluez-alsa/files/12910355/bluealsa.service-aplay-44100Hz-file.log)
[bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/12910356/bluetooth.service.log)
[aplay-48000.log](https://github.com/arkq/bluez-alsa/files/12910357/aplay-48000.log)
[aplay-41000.log](https://github.com/arkq/bluez-alsa/files/12910358/aplay-41000.log)
[bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/12910359/bluealsa.service.log)
[bluez.log](https://github.com/arkq/bluez-alsa/files/12910360/bluez.log)

borine, 2023-10-15:
OK, so these logs are much easier to read - thankyou!

first the `aplay-48000` run.

The aplay log `aplay-48000.log` is exactly as I would expect for a working system. The BlueALSA plugin has successfully played out the whole 1 second file and cleaned up correctly.

The bluealsa service log `bluealsa.service-aplay-48000Hz-file.log` again shows the server correctly setting up the A2DP transport, play out the audio, then cleaning up.

So as far as BlueALSA is concerned this file was played correctly. The only thing that looks slightly odd is that the Resume command occurs at 12:54:58 but the drain does not complete until 12:55:00, so it looks like 2 seconds of audio played. However, that may be rounding error in the timestamps, so its not really a big concern. So I cannot explain why no sound was heard from the Onkyo.

Now the `aplay-441001 run.

The aplay log `aplay-41000.log` actually shows 2 separate runs of aplay, process id 3864 then process id 3949. In both runs aplay appears to stall while playing the audio - I guessing you used Ctl-C to terminate them both. This tells me that the bluealsa server is not reading from the PCM audio pipe.

The bluealsa service log `bluealsa.service-aplay-44100Hz-file.log` shows that the server corretly setp up the A2DP transport, but 5 seconds after the encoder started to play out the audio to the bluetooth socket, Bluez changed the transport state to "Idle" causing the bluealsa server to close the transport before the audio was finished. 25 seconds after that the client connection was closed - I'm guessing that's when you used Ctl-C on aplay.

The log bluez.log appears tp confirm that Bluez changes the transport state to idle 5 seconds after activating it. The consistency with which this happens suggests to me that somewhere in the Bluez/Bluetooth layers there is a timeout of 5 seconds waiting for something that apparently BlueALSA is not doing but Pipewire does.

 I'm guessing the test file Track06.wav is longer that 5 seconds, and that is the real difference between these test runs, the different sample rates are not relevant. If you have a test sample at 48000Hz longer than 5 seconds, that would prove it.

I can't spend any more time on this now, but will look at it again later next week.

spamtree, 2023-10-15:
You are correct with a test sample wave file at 48000Hz longer than 5 seconds the behavior is the exactly the same.

borine, 2023-10-18:
> When compiling from the master branch did I need to apply the drain patches

No, please initially use the source without any changes, to give a baseline. I may ask you to apply the patch in #670 later, but only if we can trigger the lockup again first. You will probably need to enable aac, and also please make sure you enable debug. If you are able to successfully build, and install, and then connect to the Onkyo, then I have some more tests I would like to you to run. 

I now know that the lack of sound from the Onkyo is caused by it sending a AVDTP SUSPEND command, which causes bluez to force the A2DP transport to the idle state. I need to know if that same command is also sent when using pipewire/gnome, and then try to work either why not, or else how pipewire, or some other component of the gnome desktop, deals with it.

But first some logs using bluealsa master branch; using aplay, then using speaker-test. While aplay is blocked, and before pressing control-C, please  try to produce the gdb backtrace. Hopefully your manual build of the master branch will have gdb debug symbols as that is the default. If speaker-test also gets blocked, then please also produce the gdb backtrace.

Be prepared though, that depending on why the Onkyo is sending that suspend command, it may turn out to be impossible to use bluealsa with it. I'm looking through the pipewire source code and haven't found any hints yet, so at this stage I really don't know whether we can fix this.

borine, 2023-10-21:
One more idea - possibly more convincing than the last one. The consistent interval of 5 seconds between acquiring the A2DP transport and receiving the AVDTP SUSPEND command has been bugging me. There is no mention of such a time delay, or specific use of the suspend command in that way, in the Bluetooth specs. So I looked through the Bluez A2DP source code.

When used on a sink (eg the Onkyo), on receipt of a AVDTP START command Bluez sends back a confirmation response, changes the corresponding transport state from IDLE to PENDING, **starts a 5 second timer**, then waits for some local application to acquire the transport. If acquisition does not happen within the 5 seconds it changes the transport state back to IDLE and sends a AVDTP SUSPEND command to the source.

Now look at the bluetooth.service logs on our bluealsa A2DP source host. When aplay starts playback, bluealsa asks Bluez to acquire the transport, causing Bluez to send a AVDTP START command to the Onkyo. Bluez immediately recieves a confirmation response and so changes the transport state to ACTIVE, causing bluealsa to start sending out encoded audio frames. **5 seconds later** Bluez receives a AVDTP SUSPEND command from the Onkyo, causing it to change the transport state to IDLE. Bluealsa then stops sending audio and so aplay becomes blocked waiting for the ALSA ring buffer which has become full.

Can that just be coincidence, or is it possible that the Onkyo is using Bluez, and is waiting for some event to start the playback of the incoming stream? Perhaps some button press, perhaps some other Bluetooth event such an AVRCP command or other? I don't see any other Bluetooth profile active in the gnome-bluez log so I don't know how gnome is managing to persuade the Onkyo to play the stream, but this theory does at least explain the troubling 5 second interval.

borine, 2023-10-22:
@spamtree I've created a test A2DP sink using BlueALSA, modified so that the bluealsa daemon does not acquire the transport when a new incoming stream arrives. Tested against a source using BlueALSA, and the bluetooth.service log on the source looks exactly like yours. So I am now convinced the problem is that the Onkyo is not starting a bluetooth playback application when the incoming stream arrives. To work out how Gnome is different I would need to see the bluetooth.service log for a working Gnome session from the time just before aplay is started until the file is played out (a short file, but more than 10 seconds, would be best). You could use the original 4.1.1 BlueALSA build for this. as we know that is working, or a build from the latest master branch if you have it, whichever is the easiest for you to set up.

spamtree, 2023-10-22:
Wanted to let know that I'm still here but have a lot of work to do. I have installed the bluealsa git master and was running some tests yesterday. The results were the same timeout after 5 seconds and bluealsa did not crash (no gdb info). After reading your posts what I have been working on is capturing some logs for a working gnome session and trying to crash the onkyo bluetooth stack. What is interesting is  the bluetooth.service logs are almost empty even with debug turned on. The bluetooth connection appears to be managed by wiremanager. So I have been playing around with getting debugging  logs from wiremanager. I think I understand how to do this after reading [Debugging Pipewire](https://gitlab.freedesktop.org/pipewire/pipewire/-/wikis/Troubleshooting?version_id=8e027b3eaf9e0628471d5cacc31a18b5dff0390f)

Will work on getting some logs. If I am going in the wrong direction let me know. Thanks.

borine, 2023-10-22:
> The bluetooth connection appears to be managed by wiremanager. 

I've never heard of wiremanager. There is a section https://gitlab.freedesktop.org/pipewire/pipewire/-/wikis/Troubleshooting?version_id=8e027b3eaf9e0628471d5cacc31a18b5dff0390f#bluetooth-traffic-dump on how to use `btmon`, and that info would be useful, but the file will be huge, so I was hoping that just looking at the bluetooth debug log would be enough to give a clue.

> bluetooth.service logs are almost empty even with debug turned on

That's weird, I wonder if gnome is somehow turning off the debug option. What method do you use to enable debug in bluetooth?

 While bluetooth is started, and producing these near-empty logs, please can you run:
```
ps -ef | grep bluetoothd
```
and check that the output shows the `-d` option, something like
```
root      127755  1 1 14:54 ?    00:00:00 /usr/lib/bluetoothd -d
```

spamtree, 2023-10-23:
Whoops! Didn't write the correct systemd override unit file for bluetooth.service. Thanks for the tip that is fixed. Attached is gnome.bluetooth.service.log before aplay and after aplay was completed. If I run the aplay command again the gnome.bluetooth.service.log was the same. Will add a new bluealsa.bluetoothd.service.log
[bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/13065269/bluealsa.service.log)
[bluealsa.bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/13065270/bluealsa.bluetooth.service.log)
[bluealsa.aplay.log](https://github.com/arkq/bluez-alsa/files/13065271/bluealsa.aplay.log)
[gnome.aplay.log](https://github.com/arkq/bluez-alsa/files/13065272/gnome.aplay.log)
[gnome.bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/13065273/gnome.bluetooth.service.log)
 before and after aplay. If something does not look write let me know. Thanks!

borine, 2023-10-23:
Thanks for the new logs. They confirm that pipewire/gnome are not sending any "special" bluetooth control commands to the Onkyo to start the stream, the command sequence is the same as used by BlueLASA. However, comparing with earlier logs I noticed one interesting difference: when using pipewire the stream configuration "blob" is different than when using BlueALSA. The Pipewire stream uses "21150228" which decodes to:
```text
SBC <hex:21150228> {
  sampling-frequency:4 = 44100
  channel-mode:4 = JointStereo
  block-length:4 = 16
  subbands:2 = 8
  allocation-method:2 = Loudness
  min-bitpool-value:8 = 2
  max-bitpool-value:8 = 40
}
```
Note the allocation method: "Loudness"

Whereas the BlueALSA stream configuration is reported in the logs as:
```
Oct 22 20:17:40 travel bluealsa[1397]: codec-sbc.c:264: SBC setup: 44100 Hz JointStereo allocation=SNR blocks=16 sub-bands=8 bit-pool=40 => 256331 bps
```
Note the allocation method is "SNR"

So perhaps we can persuade BlueALSA to use the "Loudness" allocation method. I have no idea if this will work, but please can you try, while running bluealsa:
```text
aplay -D bluealsa:CODEC=sbc:21150228  Track06.wav
```

That command requests that bluealsa use the same SBC configuration as is used by pipewire on the Onkyo. So perhaps that will persuade the Onkyo to play the stream.

spamtree, 2023-10-23:
Wow! That did it! You are great! 

Thanks!!!

borine, 2023-10-24:
Thats good news. You could make it easier to use by creating a ALSA configuration entry in either `~/.asoundrc` or `/etc/asound.conf`. I suggest:
```
pcm.onkyo {
	type empty
	slave.pcm "bluealsa:DEV=00:09:B0:1C:CD:D4,CODEC=sbc:21150228"
	hint.description "Onkyo TX-NR676 EE6F60"
}
```
Then you can use it as ` aplay -D onkyo Track06.wav`

It may be possible to also get AAC working using the same trick if we can find the correct configuration blob in the bluez-dbus-monitor.log from a working AAC stream using Pipewire/Pulseaudio or whatever.

I will continue to look into this to see if there is anything at all thar BlueALSA can do to identify when a device is ignoring ta requested  codec configuration change.

arkq, 2023-10-24:
> but `... allocation=SNR ...`

It seems that it is only due to a bug in the sbc_print_internals() function. So I wonder why `bluealsa:CODEC=sbc:21150228` worked....

borine, 2023-10-24:
> So I wonder why bluealsa:CODEC=sbc:21150228 worked....

I think it is because that request causes bluealsa to invoke org.bluez.MediaTransport1.SetConfiguration(). Without it the local endpoint uses the configuration set by the Onkyo and I suspect that, possibly because of cache management issues, the Onkyo is not correctly applying that configuration to its own endpoint, leading to failure when the first codec frames arrive. By calling SetConfiguration() itself, bluealsa forces the Onkyo to change the configuration of its own endpoint so that both ends have the same configuration.

To test that idea we would need some way of knowing what the actual configuration set on the Onkyo endpoint is; perhaps a org.freedesktop.DBus.Properties.Get() call would do that? I don't know enough about the workings of A2DP or Bluez to know how this situation can happen, or how to test for it.

borine, 2023-10-24:
@spamtree now that bluealsa has successfully configured the A2DP stream, it is possible that the Onkyo has updated its cached configuration. One way to test that is to try bluealsa without specifying a specific config:
```
aplay -D bluealsa Track06.wav
```
Could you try that to see if it now works? And if it does, then try with Gnome again to see whether that still works.

spamtree, 2023-10-25:
@borine Thanks for the .asoundrc, I was working on one but yours is much better. 

AAC does not work with gnome either and when I looked at the configuration for the onkyo in gnome the selection for AAC was missing completely. It was there at one time. Don't have a clue why it is missing now. 

Before I saw the above messages - I started changing the sound configuration in gnome for the onkyo receiver playing with higher quality SBC codecs. Don't have a clue why I did this but messing around. Anyway sound played fine with gnome, but when I went back to bluealsa on a different pc, no sound in bluealsa and then no sound in gnome! Well cleared all the bluetooth caches on the computers and then cleared the bluetooth caches on the onkyo and reset the onkyo receiver. I first setup bluealsa but NO sound. Then I setup gnome and sound worked. Then tried bluealsa again and sound returned! No more messing with codecs will stick with plain SBC. Not sure but once the cache is setup on the onkyo I guess one can't change the codec (only a thought but no way I can prove it). Onkyo is probably buggy. 

@borine 
After running 
`aplay -D bluealsa:CODEC=sbc:21150228  Track06.wav`

Then running 
`aplay -D bluealsa  Track06.wav`

Both command played sound! 

Hope this helps. Had logs but did not send them since sound is working again.

spamtree, 2023-10-25:
Here are the logs that I collected from the above session. 

[sound-gnome-bluez.log](https://github.com/arkq/bluez-alsa/files/13168656/sound-gnome-bluez.log)
[sound-gnome-bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/13168657/sound-gnome-bluetooth.service.log)
[sound_CODEC-sbc-21150228-bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/13168658/sound_CODEC-sbc-21150228-bluetooth.service.log)
[sound_CODEC-sbc-21150228-bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/13168660/sound_CODEC-sbc-21150228-bluealsa.service.log)
[nosound-CODEC-sbc-21150228-bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/13168662/nosound-CODEC-sbc-21150228-bluetooth.service.log)
[nosound-CODEC-sbc-21150228-bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/13168663/nosound-CODEC-sbc-21150228-bluealsa.service.log)

spamtree, 2023-10-26:
You good folks have taught me a lot about debugging bluetooth sound problems. I have learned how to capture bluetooth logs and slowly learning how to read these logs but still way over my head. I think my initial bug request was probably inaccurate. I will explain. 

I returned from work today and wanted to check and see if sound was still working. Started the two computers and No sound on bluealsa or gnome. I noted the same 5 sec delay - state changing to active to idle on both bluealsa and gnome. I got sound working on gnome after restarting bluetooth.service and reconnecting to the onkyo.  Decided to pair an iPhone to the onkyo and audio played nicely. 

I decided to NOT reset all the bluetooth caches, NOT unpair and repair bluetooth, and NOT reset the onkyo. I reviewed the previous logs and compared them with new logs today and noted something interesting with the file descriptor. The file descriptor had to be fd0 for sound to play with bluealsa to the onkyo. Example:  

`Transport /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1/fd0`

Whenever the transport was on any other file descriptor fd[1-9] sound will not play. Example

`Transport /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1/fd1`

So all I did today was (without restarting the bluealsa service)

```
# systemctl restart bluetooth.service
# bluetoothctl connect 00:09:B0:1C:CD:D4
Attempting to connect to 00:09:B0:1C:CD:D4
hci0 00:09:B0:1C:CD:D4 type BR/EDR connected eir_len 23
[CHG] Device 00:09:B0:1C:CD:D4 Connected: yes
[NEW] Endpoint /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep2 
[NEW] Endpoint /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1 
[NEW] Transport /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1/fd0 
```
The file descriptor changed to fd0 and sound worked fine with 

`aplay -D bluealsa Track06.wav`

Then I ran the commands
```
# bluetoothctl disconnect
# bluetoothctl connect 00:09:B0:1C:CD:D4
Attempting to connect to 00:09:B0:1C:CD:D4
hci0 00:09:B0:1C:CD:D4 type BR/EDR connected eir_len 23
[CHG] Device 00:09:B0:1C:CD:D4 Connected: yes
[NEW] Endpoint /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep2 
[NEW] Endpoint /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1 
[NEW] Transport /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1/fd1 
Connection successful
# aplay -D bluealsa Track06.wav
```
Sure enough no sound with bluealsa. 

I repeated this test four times and had the same result. Sound will only play when fd0 is the file descriptor with bluealsa. Sound played fine with any file descriptor with gnome (pipewire). 

I did an internet search didn't find anything related to this problem. 

I apologize for not understanding the problem well enough to report the problem correctly. I tested bluealsa with a bluetooth speaker and sound played fine with any file descriptor. 

Now to make this more complicated. I have noticed that  gnome rarely goes from active to idle with no sound but it recovers and sound is played in gnome through pipewire. When this occurs in gnome a bluetooth sink is not created. I restarted bluetooth.service and the bluetooth sink was recreated once I reconnected to the onkyo. However, sound played on any file descriptor (fd[0-9). I have not been able to reproduce this problem in gnome like I have done in bluealsa. I will see if I can force this condition in gnome and capture logs. I hope my explanation is clear if not please let me know. 

Logs showing the findings discussed above. 

[sound.log](https://github.com/arkq/bluez-alsa/files/13172551/sound.log)
[nosound.log](https://github.com/arkq/bluez-alsa/files/13172552/nosound.log)

spamtree, 2023-10-26:
Since I can now duplicate the problem easily. I went back and obtained some better logs. Timestamps are clear. 

Sound and no sound logs. The no sound logs were obtained after disconnecting and then reconnecting bluetooth and sounds logs were obtained after restart bluealsa.service and bluetooth.service. 

The only major differences I could see are the following sections in the log files:

# no_sound_bluealsa.service.log.

```
Oct 26 11:12:41 travel bluealsa[1731]: codec-sbc.c:264: SBC setup: 44100 Hz JointStereo allocation=SNR blocks=16 sub-bands=8 bit-pool=40 => 256331 bps
Oct 26 11:12:41 travel bluealsa[1731]: a2dp-sbc.c:225: PCM IO loop: START: a2dp_sbc_enc_thread: A2DP Source (SBC)
Oct 26 11:12:41 travel bluealsa[1731]: ba-transport-pcm.c:259: PCM resume: 15
Oct 26 11:12:46 travel bluealsa[1731]: bluez.c:1450: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
Oct 26 11:12:46 travel bluealsa[1731]: ba-transport.c:323: Closing BT socket duplicate [19]: 20
Oct 26 11:12:46 travel bluealsa[1731]: ba-transport.c:656: Closing A2DP transport: 19
Oct 26 11:12:46 travel bluealsa[1731]: ba-transport-pcm.c:153: Exiting IO thread [ba-a2dp-sbc]: A2DP Source (SBC)
```

# sound_bluealsa.service.log

```
Oct 26 11:05:46 travel bluealsa[1731]: codec-sbc.c:264: SBC setup: 44100 Hz JointStereo allocation=SNR blocks=16 sub-bands=8 bit-pool=40 => 256331 bps
Oct 26 11:05:46 travel bluealsa[1731]: a2dp-sbc.c:225: PCM IO loop: START: a2dp_sbc_enc_thread: A2DP Source (SBC)
Oct 26 11:05:46 travel bluealsa[1731]: ba-transport-pcm.c:259: PCM resume: 15
Oct 26 11:06:08 travel bluealsa[1731]: ba-transport-pcm.c:275: PCM drain: 15
Oct 26 11:06:08 travel bluealsa[1731]: ba-transport.c:471: PCM clients check keep-alive: 10000 ms
Oct 26 11:06:09 travel bluealsa[1731]: ba-transport-pcm.c:295: PCM drained
Oct 26 11:06:09 travel bluealsa[1731]: ba-transport-pcm.c:303: PCM drop: 15
Oct 26 11:06:09 travel bluealsa[1731]: ba-transport-pcm.c:303: PCM drop: 15
Oct 26 11:06:09 travel bluealsa[1731]: io.c:201: PCM client closed connection: 15
Oct 26 11:06:09 travel bluealsa[1731]: ba-transport-pcm.c:238: Closing PCM: 15
Oct 26 11:06:09 travel bluealsa[1731]: ba-transport.c:471: PCM clients check keep-alive: 10000 ms
Oct 26 11:06:09 travel bluealsa[1731]: ba-transport.c:471: PCM clients check keep-alive: 10000 ms
Oct 26 11:06:09 travel bluealsa[1731]: ba-transport.c:471: PCM clients check keep-alive: 10000 ms                                                               
Oct 26 11:06:19 travel bluealsa[1731]: ba-transport.c:422: Stopping transport: No PCM clients                                                                   
Oct 26 11:06:19 travel bluealsa[1731]: ba-transport.c:323: Closing BT socket duplicate [19]: 20                                                                 
Oct 26 11:06:19 travel bluealsa[1731]: ba-transport.c:629: Releasing A2DP transport: 19                                                                         
Oct 26 11:06:19 travel bluealsa[1731]: bluez.c:1450: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
Oct 26 11:06:19 travel bluealsa[1731]: ba-transport.c:656: Closing A2DP transport: 19
Oct 26 11:06:19 travel bluealsa[1731]: ba-transport-pcm.c:153: Exiting IO thread [ba-a2dp-sbc]: A2DP Source (SBC)
```

# no_sound_bluetooth.service.log

```
Oct 26 11:12:41 travel bluetoothd[1730]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1/fd1: TRANSPORT_STATE_REQUESTING ->TRANSPORT_STATE_ACTIVE
Oct 26 11:12:41 travel bluetoothd[1730]: profiles/audio/a2dp.c:setup_unref() 0x5559a0c21ca0: ref=0
Oct 26 11:12:41 travel bluetoothd[1730]: profiles/audio/a2dp.c:setup_free() 0x5559a0c21ca0
Oct 26 11:12:41 travel bluetoothd[1730]: profiles/audio/avdtp.c:avdtp_unref() 0x5559a0c45050: ref=2
Oct 26 11:12:41 travel bluetoothd[1730]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: OPEN -> STREAMING
Oct 26 11:12:41 travel bluetoothd[1730]: profiles/audio/sink.c:sink_set_state() State changed /org/blue/hci0/dev_00_09_B0_1C_CD_D4: SINK_STATE_CONNECTED -> SINK_STATE_PLAYING
Oct 26 11:12:41 travel bluetoothd[1730]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1/fd1 State=TRANSPORT_STATE_ACTIVE Playing=1
Oct 26 11:12:46 travel bluetoothd[1730]: profiles/audio/avdtp.c:session_cb()
Oct 26 11:12:46 travel bluetoothd[1730]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received SUSPEND_CMD
Oct 26 11:12:46 travel bluetoothd[1730]: profiles/audio/a2dp.c:suspend_ind() Source 0x5559a0bf6bf0: Suspend_Ind
Oct 26 11:12:46 travel bluetoothd[1730]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: STREAMING -> OPEN
Oct 26 11:12:46 travel bluetoothd[1730]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_00_09_B0_1C_CD_D4: SINK_STATE_PLAYING -> SINK_STATE_CONNECTED
Oct 26 11:12:46 travel bluetoothd[1730]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1/fd1 State=TRANSPORT_STATE_ACTIVE Playing=0
Oct 26 11:12:46 travel bluetoothd[1730]: profiles/audio/transport.c:media_transport_remove_owner() Transport /org/bluez/hci0/dev_00_09_B0_1C_CD_D4/sep1/fd1 Owner :1.46
Oct 26 11:12:46 travel bluetoothd[1730]: profiles/audio/transport.c:media_owner_free() Owner :1.46
[... cut ...] uetooth.service.logs

Full logs with timestamps below. 

[no_sound_bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/13180565/no_sound_bluealsa.service.log)
[sound_bluealsa.service.log](https://github.com/arkq/bluez-alsa/files/13180566/sound_bluealsa.service.log)
[no_sound_bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/13180567/no_sound_bluetooth.service.log)
[sound_bluetooth.service.log](https://github.com/arkq/bluez-alsa/files/13180568/sound_bluetooth.service.log)

arkq, 2023-10-26:
> The file descriptor had to be fd0 for sound to play with bluealsa to the onkyo. 

Hmm... interesting notice. There is something in BlueALSA that is not in PulseAudio nor in PipeWire. BlueALSA registers more than one SEP for a single A2DP codec (in order to allow more then one device to be connected via the same codec). However, that might trip onkyo.... Here is a patch which will "turns off" that feature. Please, try to use is. I'm not sure whether it will work or not, but applying it will bring BlueALSA closer to PipeWire implementation (which works).

```diff
diff --git a/src/bluez.c b/src/bluez.c
index 94f107c..0782abf 100644
--- a/src/bluez.c
+++ b/src/bluez.c
@@ -740,7 +740,7 @@ static void bluez_export_a2dp(
                if (dbus_obj->connected)
                        connected++;

-               continue;
+               break;

 fail:
                if (err != NULL) {
```

spamtree, 2023-10-27:
Will do. Thanks

spamtree, 2023-10-27:
That patch did not work. Audio output only when file descriptor is fd0. 

Followup: I know this has nothing to due with the bug report but wanted to clear up this issue in case anyone else reads this thread. The Gnome problem with no audio after boot is a bug somewhere. Pipewire audio sink is not created unless bluetooth.service is started after the user logon. Restarting bluetooth.service after logon, bluetooth audio works perfectly. This happened with a bluetooth speaker and the onkyo receiver. This is an easy work around. This does not happen on my other laptop. Probably has something to do with this error on boot. 

`systemd[1]: Bluetooth service was skipped because of an unmet condition check (ConditionPathIsDirectory=/sys/class/bluetooth).`

Thanks again.

borine, 2023-10-27:
The bluez transport object path uses a simple global counter to number the `fdN` component. So the first A2DP transport created after the bluetooth daemon has started is `fd0`. When you disconnect the onkyo that transport is removed, and when you reconnect a new one is created, `fd1`, etc. That is normal. So I think a better way of describing the symptoms here is that only the first connection attempt to the onkyo after starting bluetooth produces a working transport, all subsequent attempts result in the stream being suspended 5 seconds after starting. I would be interested to know what happens if the first connection after starting bluetooth is to some other device, then disconnect that and now connect to the Onkyo. The Onkyo transport will be `fd1`, but does it produce a working stream? 

On the same theme, what happens if the first connection (fd0) is to the Onkyo, then the second (fd1) to some other device, then the third (fd2) to the Onkyo again; does the final Onkyo connection produce sound?

borine, 2023-10-27:
> systemd[1]: Bluetooth service was skipped because of an unmet condition check (ConditionPathIsDirectory=/sys/class/bluetooth).

That indicates that systemd is trying to start the bluetooth service when then are no bluetooth adapters available. Either the USB adapter was not plugged in at boot, or else the kernel has not finished probing the USB bus before systemd tries to start bluetooth.

[30 more replies]

closed 2024-08-11 by spamtree