arkq/bluez-alsa #673: Sink producing no audio with aptX codec

rudism, 2023-11-19:
### Problem

I've managed to build with AAC, aptX, and aptX-HD (using libopenaptx) support on a Pi Zero W with a DAC hat, which I'm running as an A2DP sink for some other devices to connect to and play audio to connected speakers. When source devices connect and select the SBC or AAC codecs the audio plays just fine, but switching to aptX always results in no audio. I've tried both with the 4.1.1 snapshot and off `master` with the same results.

### Reproduction steps

On devices with the ability to switch codecs, I can go from AAC or SBC with audio playing to aptX, get silence, then go back to AAC or SBC and the audio resumes.

### Setup

- OS: Raspbian GNU/Linux 12 (bookworm), kernel `6.1.0-rpi6-rpi-v6`
- the version of BlueALSA: v4.1.1 (also tried building from `master`)
- the version of BlueZ: 5.66
- the version of ALSA: 1.2.8

### Additional context

I installed `libopenaptx` via package manager (also tried installing from source with no difference), installed `fdk-aac` manually from source, configured and built bluealsa with no errors:
```
../configure --enable-aac --enable-aptx --enable-aptx-hd --with-libopenaptx --enable-systemd --enable-debug --enable-cli
```
```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for wchar.h... yes
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking whether _XOPEN_SOURCE should be defined... no
checking for gcc... (cached) gcc
checking whether the compiler supports GNU C... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to enable C11 features... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking whether ln -s works... yes
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... armv6l-unknown-linux-gnueabihf
checking host system type... armv6l-unknown-linux-gnueabihf
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
[... 273 lines cut ...]
Nov 18 17:40:48 piaudio bluealsa[770]: ../../src/bluez.c:1308: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
Nov 18 17:40:48 piaudio bluealsa[770]: ../../src/bluez.c:1308: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/bluez.c:1199: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/dbus.c:47: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/aptX/sink/1
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/storage.c:117: Loading storage: /var/lib/bluealsa/00:02:5B:00:FF:0F
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/bluez.c:420: A2DP Sink (aptX) configured for device 00:02:5B:00:FF:0F
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/bluez.c:423: A2DP selected configuration blob [len=7]: 4f000000010012
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/bluez.c:425: PCM configuration: channels: 2, sampling: 48000
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/dbus.c:47: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_00_02_5B_00_FF_0F/a2dpsnk/source
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/bluez.c:1407: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/ba-transport.c:741: New A2DP transport: 19
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/ba-transport.c:742: A2DP socket MTU: 19: R:672 W:895
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/bluez.c:1407: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/ba-transport.c:1528: Starting transport: A2DP Sink (aptX)
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/ba-transport.c:448: Created BT socket duplicate: [19]: 15
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/ba-transport.c:1949: Created new IO thread [ba-a2dp-aptx]: A2DP Sink (aptX)
Nov 18 17:40:49 piaudio bluealsa[770]: ../../src/a2dp-aptx.c:253: IO loop: START: a2dp_aptx_dec_thread: A2DP Sink (aptX)
Nov 18 17:40:51 piaudio bluealsa[770]: ../../src/bluez.c:1199: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Nov 18 17:40:51 piaudio bluealsa[770]: ../../src/bluez.c:1407: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: Volume
Nov 18 17:40:51 piaudio bluealsa[770]: ../../src/bluez.c:1429: Updating A2DP volume: 127 [0.00 dB]
```

Got no audio again now.

I don't see any obvious red flags in any of the logs, and I'm not sure how to debug this further. I've tried building using openaptx instead of libopenaptx but it doesn't seem to provide decoder support so it's not compatible with my use case as a sink.

arkq, 2023-11-19:
I'm not sure how feasible it will be for you to record the Bluetooth traffic on your RPi Zero with e.g. `tcpdump` or something similar. Then I can try to "decode" the aptX audio on my side and see whether there are any issues in the stream. In the meantime I will check aptX on my RPi4 (I have not been using aptX for a long time, so it's possible that there is a regression in the code).

rudism, 2023-11-19:
@arkq I since took the time to compile a new kernel with [this patch](https://github.com/anonymix007/libldacdec#building-for-raspberry-pi) from anonymix007's libldacdec repo and then recompile bluealsa following the directions there to add ldac support (along with aac, aptx, and aptx-hd same as before).

In addition to ldac working ok with the new kernel, it has also fixed aptx for me, so perhaps my problem was related to the same MTU issue that prevents the LDAC library from working against the standard Raspberry Pi kernel.

At any rate, I'll close this issue since it's working now.

closed 2023-11-19 by rudism