arkq/bluez-alsa #676: allow build with different dbus name

peteasa, 2023-12-12:
### Feature description
hard coded values are usually bad and this code has "org.bluealsa" hard coded.  I would suggest changing this to be a configurable item so that different values such as org.bluez can be used.

## Additional context
This is a suggestion rather than a requirement!.. I am attempting to build the latest bluez-alsa for Ubuntu 22.04 to see if that fixes some problems that I have found with the distribution version from the repo.  Ubuntu seem to use org.bluez for the dbus name so I would like to use the same so that my own built version will just replace the distribution version.

Without this change I get 
bluealsa: W: UUID already registered in BlueZ [hci0]: 0000110B-0000-1000-8000-00805F9B34FB
bluealsa: W: UUID already registered in BlueZ: 00001108-0000-1000-8000-00805F9B34FB

I would expect to be able to configure the UUID and the dbus name at configure time so that I can build an slot in equivalent binary easily.

arkq, 2023-12-12:
> Ubuntu seem to use org.bluez

The D-Bus service name is a well known name of a service, like a domain name on the Internet. Using different name for org.bluealsa by Linux distribution would be like using a different name for google.com by ISP.

The org.bluez is "reserved" name for BlueZ service, so it's rather impossible that anyone will use that for BlueALSA. Also, it's rather impossible that some distribution will use any other name because some other 3rd party software might search for org.bluealsa to communicate with it.

> Without this change I get
> bluealsa: W: UUID already registered in BlueZ [hci0]: 0000110B-0000-1000-8000-00805F9B34FB
> bluealsa: W: UUID already registered in BlueZ: 00001108-0000-1000-8000-00805F9B34FB

This error means that there is some other application which communicates with BlueZ and has registered these profiles. The warning message is printed by BlueALSA to help solve issues with not-working BlueALSA instance. Please see this: https://github.com/arkq/bluez-alsa/blob/master/TROUBLESHOOTING.md#4-using-bluealsa-alongside-pulseaudio-or-pipewire

peteasa, 2023-12-13:
thanks for the comments

closed 2023-12-13 by peteasa