arkq/bluez-alsa #683: Buttons not working during callback recording

jeffersonchua, 2024-02-18:
### Problem

I’ve written a python program pyaudio callback with both input and output enabled, and I can hear and speak at the same time, but the bottons on the headsets are not detected while the callback is running. Is this a bug?

Buttons work fine after recording stopped. I’m listening to evdev.Inputdevice() read_loop() for that in a separate process.

Using latest version of bluealsa git.

borine, 2024-04-29:
You give very little information on which anyone can begin to investigate, but as a random guess perhaps the "buttons" (you do not even say what function you expect each button to perform) send AVRCP profile messages, you have connected the HFP or HSP profile, and the headset disables AVRCP when HFP or HSP are active. Perhaps the headset instead uses the buttons to send HFP or HSP rfcomm messages when using HFP or HSP, in which case Bluez does not generate evdev events, but BlueALSA will handle the rfcomm volume change messages but ignore the call handling ones.

jeffersonchua, 2024-05-02:
> You give very little information on which anyone can begin to
investigate, but as a random guess perhaps the "buttons" (you do not even
say what function you expect each button to perform) send AVRCP profile
messages, you have connected the HFP or HSP profile, and the headset
disables AVRCP when HFP or HSP are active. Perhaps the headset instead uses
the buttons to send HFP or HSP rfcomm messages when using HFP or HSP, in
which case Bluez does not generate evdev events, but BlueALSA will handle
the rfcomm volume change messages but ignore the call handling ones.

My bad. Buttons refers the Play/Stop which as you mentioned "disables AVRCP
when HFP or HSP are active".  So, that's why I'm not seeing the events.

My best.,
Jeff

closed 2024-05-04 by borine