arkq/bluez-alsa #685: HSP HS profile:  "arecord -d"  duration is ignored

sergey-suloev, 2024-02-27:
The "arecord" command ignores -d parameter - recording never ends.

root@nanopi-a64-bullseye:~/music# arecord -D bluealsa:DEV=B4:F1:DA:2C:13:2B,PROFILE=sco -d 5 -f s16_le -c 1 -r 8000  capture.wav
Recording WAVE 'capture.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono

borine, 2024-02-29:
> recording never ends

Does the file "capture.wav" contain any audio after you have stopped `arecord` with ctrl-C ?
My guess is that it most likely contains only the WAV header (typically 44 bytes).

If that is true, then the recording is never actually *starting*. On hsp-hs (and hfp-hf and a2dp-sink) nodes the BlueALSA PCM plugin simply blocks until the remote (phone) device enables the audio stream. This is described in the manual page here: https://github.com/arkq/bluez-alsa/blob/master/doc/bluealsa-plugins.7.rst#transport-acquisition). Because `arecord` is blocked it cannot start its timer.

I am currently looking at the possibility of offering alternative behaviours, selected by a new plugin parameter (which I am calling `hwcompat` for "hw plugin compatibility, but that may change before I raise a PR).

For example, with `defaults.bluealsa.hwcompat "busy"` the plugin would fail immediately with error `-EBUSY` ("Device or resource busy") if opened on a hsp-hs/hfp-hf/a2dp-sink before the transport has been acquired.

Other possible values for `hwcompat` that I am considering include:
- `"none"` - the default, existing behaviour
- `"silence"` - insert silence samples for capture, or discard samples for playback, until the audio stream is started by the remote device
- any others that may be useful in various (yet to be investigated) circumstances.

borine, 2024-08-13:
closing due to lack of activity

closed 2024-08-13 by borine