arkq/bluez-alsa #687: Test all deps with pkgconfig

sergey-suloev, 2024-03-02:
Could you test for all deps using pkgconfig, please ? 
For example, mp3lame is tested with gcc and this is a big headache while cross-compiling.

borine, 2024-03-03:
Does libmp3lame have a pkgconfig file? My Ubuntu and RaspberrypiOS installations do not have a libmp3lame.pc, nor a lame.pc.

sergey-suloev, 2024-03-03:
Of course it does !  I am using stable debian , package libmp3lame-dev

borine, 2024-03-03:
- Ah, OK I've just checked. There is a lame.pc in debian bookworm (libmp3lame-dev 3.100-6) but not in debian buster (libmp3lame-dev 3.100-3). So it is a relatively recent addition. I don't see it in the upstream mp3lame sources though.

arkq, 2024-03-03:
Unfortunately, the only reliable way of checking for the libmp3lame is testing for the library presence... Upstream has never provided support for pkg-config. There are some distributions which patch their devel package with pkg-config (like Gentoo or the latest Debian), but most distributions do not...

sergey-suloev, 2024-03-03:
I think the best solution would be using both methods

arkq, 2024-03-03:
Put apparently things are moving in the right direction:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=829327
https://sourceforge.net/p/lame/feature-requests/42/

sergey-suloev, 2024-03-03:
> Put apparently things are moving in the right direction: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=829327 https://sourceforge.net/p/lame/feature-requests/42/

That's really ridiculous how long the discussion had been going while the problem itself needs 3 minutes of work

arkq, 2024-03-03:
> That's really ridiculous how long the discussion has been lasting in community while the problem itself needs 3 minutes of work

That's the case where there are no incentives involved. In open source community no one wants extra things which needs to be maintained (for free) later :D

> I think the best solution would be using both methods , pkgconfig first and lib testing as a fallback

Since the `lame.pc` will land in the next release of mp3lame, I think that we can add support for that.

arkq, 2024-03-03:
I've added pkg-config support for LAME. The only library which still does not have pkg-config support is LC3plus (it's not available in the upstream). In the future, LC3plus will be supported via Google's LC3 library too (which does have pkg-config support), so stay tuned.

commit 4282f3a1 (2024-03-03): Support for detecting LAME library via pkg-config

Fixes #687
closed 2024-03-03 by arkq