arkq/bluez-alsa #688: arecord Unable to install hw params, record from bluetooth failed

idreamerhx, 2024-03-26:
> Please read the [troubleshooting guide](../blob/master/TROUBLESHOOTING.md)
> before raising a new issue.

### Problem

Thanks for the project!

I can not record from bluetooth device using arecord. (And I tested connect to a headset using aplay has no sound output.)

### Reproduction steps

arecord --verbose -D bluealsa:B4:6B:FC:95:D7:5A -f S16_LE -c 2 -r 44100   aa.raw
Recording WAVE 'aa.raw' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: 125000
PERIOD_SIZE: (5512 5513)
PERIOD_BYTES: 22050
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 22050
BUFFER_BYTES: 88200
TICK_TIME: 0

journalctl -n 1000 |grep blue
Mar 26 03:50:03 rockpis-2 bluetoothd[419]: Stopping SDP server
Mar 26 03:50:03 rockpis-2 bluetoothd[419]: Exit
Mar 26 03:50:03 rockpis-2 systemd[1]: bluetooth.service: Deactivated successfully.
Mar 26 03:50:03 rockpis-2 systemd[1]: Stopped bluetooth.service - Bluetooth service.
Mar 26 03:50:07 rockpis-2 systemd[1]: Started bluealsa-aplay.service - BlueALSA player service.
Mar 26 03:50:07 rockpis-2 systemd[1]: Starting bluetooth.service - Bluetooth service...
Mar 26 03:50:07 rockpis-2 (uetoothd)[425]: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)
Mar 26 03:50:08 rockpis-2 bluealsa-aplay[423]: Couldn't get BlueALSA PCM list: The name org.bluealsa was not provided by any .service files
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: Bluetooth daemon 5.66
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: src/main.c:check_options() Unknown key Discoverable for group General in /etc/bluetooth/main.conf
Mar 26 03:50:08 rockpis-2 systemd[1]: Started bluetooth.service - Bluetooth service.
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: Starting SDP server
Mar 26 03:50:08 rockpis-2 systemd[1]: Starting bluealsa.service - BlueALSA service...
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: src/plugin.c:plugin_init() Failed to init vcp plugin
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: src/plugin.c:plugin_init() Failed to init mcp plugin
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: src/plugin.c:plugin_init() Failed to init bap plugin
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: Bluetooth management interface 1.22 initialized
Mar 26 03:50:08 rockpis-2 dbus-daemon[351]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.7' (uid=0 pid=425 comm="/usr/libexec/bluetooth/bluetoothd")
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Mar 26 03:50:08 rockpis-2 bluetoothd[425]: sap-server: Operation not permitted (1)
Mar 26 03:50:10 rockpis-2 systemd[1]: Started bluealsa.service - BlueALSA service.
Mar 26 03:50:10 rockpis-2 systemd[1]: Reached target bluetooth.target - Bluetooth Support.
Mar 26 03:50:10 rockpis-2 bluetoothd[425]: Endpoint registered: sender=:1.9 path=/org/bluez/hci0/A2DP/SBC/source/2
Mar 26 03:50:10 rockpis-2 bluetoothd[425]: Endpoint registered: sender=:1.9 path=/org/bluez/hci0/A2DP/SBC/sink/1
Mar 26 03:50:10 rockpis-2 bluetoothd[425]: Endpoint registered: sender=:1.9 path=/org/bluez/hci0/A2DP/SBC/sink/2
Mar 26 03:50:10 rockpis-2 bluetoothd[425]: Endpoint registered: sender=:1.9 path=/org/bluez/hci0/A2DP/AAC/sink/2
Mar 26 03:50:10 rockpis-2 bluetoothd[425]: Endpoint registered: sender=:1.9 path=/org/bluez/hci0/A2DP/SBC/source/1
Mar 26 03:50:10 rockpis-2 bluetoothd[425]: Endpoint registered: sender=:1.9 path=/org/bluez/hci0/A2DP/AAC/sink/1
Mar 26 03:50:10 rockpis-2 bluetoothd[425]: Endpoint registered: sender=:1.9 path=/org/bluez/hci0/A2DP/AAC/source/2
Mar 26 03:50:10 rockpis-2 bluetoothd[425]: Endpoint registered: sender=:1.9 path=/org/bluez/hci0/A2DP/AAC/source/1
Mar 26 03:50:11 rockpis-2 NetworkManager[421]: <info>  [1711425011.0178] Loaded device plugin: NMBluezManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
Mar 26 03:51:54 rockpis-2 systemd[667]: Reached target bluetooth.target - Bluetooth.
Mar 26 03:51:54 rockpis-2 bluetoothd[425]: profiles/audio/avctp.c:uinput_create() Can't open input device: No such file or directory (2)
Mar 26 03:51:54 rockpis-2 bluetoothd[425]: profiles/audio/avctp.c:init_uinput() AVRCP: failed to init uinput for HOME-8500
Mar 26 03:54:36 rockpis-2 bluetoothd[425]: /org/bluez/hci0/dev_B4_6B_FC_95_D7_5A/fd0: fd(28) ready
Mar 26 03:54:36 rockpis-2 bluealsa-aplay[423]: Couldn't open ALSA playback PCM: Open PCM: Host is down
Mar 26 03:54:37 rockpis-2 bluealsa-aplay[423]: Couldn't open ALSA playback PCM: Open PCM: Host is down
Mar 26 03:54:38 rockpis-2 bluealsa-aplay[423]: Couldn't open ALSA playback PCM: Open PCM: Host is down

 bluealsa-aplay -L
bluealsa:DEV=B4:6B:FC:95:D7:5A,PROFILE=a2dp,SRV=org.bluealsa
    HOME-8500, trusted computer, capture
    A2DP (SBC): S16_LE 2 channels 44100 Hz

hcitop
HCI    BUS ADDR              FLAGS           RX       TX     RX/s     TX/s
hci0  UART 00:E0:4C:23:99:87 U RPI        20 MB   234 KB  43132 B    414 B

### Setup

root@rockpis-2:/tmp# uname -a
Linux rockpis-2 6.1.68-1-stable #ba6f5fb46 SMP PREEMPT Thu Dec 14 04:36:49 UTC 2023 aarch64 GNU/Linux
root@rockpis-2:/tmp# bluealsa --version
v4.1.1-84-g1c1e9c0
root@rockpis-2:/tmp# bluetoothd --version
5.66
root@rockpis-2:/tmp# aplay --version
aplay: version 1.2.8 by Jaroslav Kysela <perex@perex.cz>

### Additional context

> Add any other context about the problem here, e.g. log messages printed by
> `bluealsa` and/or client application.
>
> Please delete instructions prefixed with '>' to prove you have read them.

idreamerhx, 2024-03-26:
ffmpeg -f alsa  -i 'bluealsa:B4:6B:FC:95:D7:5A' aa.wav

[alsa @ 0xaaaad821d8c0] cannot set parameters (Device or resource busy)
bluealsa:B4:6B:FC:95:D7:5A: Input/output error

Please help!

borine, 2024-03-26:
A BlueALSA PCM can be opened by only one process at a time. So you cannot use `arecord -D bluealsa` or `ffmpeg -i bluealsa` when also running `bluealsa-aplay` because `bluealsa-aplay` has already opened the PCM, so `arecord` or `ffmpeg` fails with the error "Device or resource busy" (EBUSY)

borine, 2024-05-06:
The [troubleshooting guide](https://github.com/arkq/bluez-alsa/blob/master/TROUBLESHOOTING.md#4-couldnt-open-pcm-device-or-resource-busy) has now been updated to clarify this limitation

closed 2024-05-06 by borine