arkq/bluez-alsa #691: Audio cut off in the beginning of the file

NGkaltzidis, 2024-04-10:
### Problem

While attempting to playback a file through my audio jack output, I'm experiencing the cut off in the beginning of the audio file.
Unfortunately this occurs only with my audio jack output, while also I've realised that this problem occurs on transition from idle state to playing.

### Reproduction steps

Tried to play a file where sound goes like this -> "One, two, three, four" 
Using `aplay -D plughw:0,0 <sound.wav> ` the playback sounds like this -> "n, two, three, four"

### Setup

Have been testing on a mini pc with [Ubuntu 22.04.4 LTS (Jammy Jellyfish)

Any thoughts or similar behaviour ?

NGkaltzidis, 2024-04-11:
The only workaround that I found here, is to start playing the audio file, immediately stop it after 300ms for example and initiate a second one. In this case there are no cut offs of the audio file playback.

borine, 2024-04-29:
This issue doe not appear to be relevant to BlueALSA. You do not mention Bluetooth anywhere in your problem description. Perhaps you have posted to the wrong project?

NGkaltzidis, 2024-04-29:
In fact yes, you are absolutely right. The issue here is related to ALSA.

closed 2024-04-29 by borine