arkq/bluez-alsa #699: how can an SCO link be reestablished after it has been disconnected by the HF device?

Sleepalltheday, 2024-04-28:
### Problem

* I have observed that during interactions with certain headset devices using the HFP (Hands-Free Profile), the headset devices tend to disconnect the SCO link upon receiving the call termination indication ("CIEV:2,0"). 
* After the SCO link is disconnected, bluez-alsa does not automatically initiate a reconnection, and I have not discovered a relevant D-Bus interface to prompt bluez-alsa to establish an SCO connection again.
* I tried using the "SelectCodec" feature to renegotiate the audio encoding, but when bluez-alsa sees the codec hasn't changed, it doesn't do anything.
* In this case, how can I prompt bluez-alsa to re-establish the SCO connection?

### Reproduction steps

* bluealsa -p hfp-ag -p a2dp-source
* use bluetoothctl connect to HF device
* run bluealsa-rfcomm, make the call setup "CIEV:2,1"
* send the call end "CIEV:2,0"

### Setup

* linux 5.1
* bluealsa --version
    * v4.1.1
* bluetoothd --version
    * 5.56

### Additional context
* After a call finishes, the SCO connection gets closed down
```
bluealsa: D: ba-rfcomm.c:110: Sending AT message: RAW: command:
+BSIR:0
, value:(null)
bluealsa: D: ba-rfcomm.c:110: Sending AT message: RAW: command:
+CIEV:3,3
, value:(null)
bluealsa: D: ba-rfcomm.c:110: Sending AT message: RAW: command:
+CIEV:2,1
, value:(null)
bluealsa: D: ba-rfcomm.c:110: Sending AT message: RAW: command:
+CIEV:3,0
, value:(null)
bluealsa: D: ba-rfcomm.c:110: Sending AT message: RAW: command:
+CIEV:2,0
, value:(null)
bluealsa: D: at.c:179: AT message: CMD: command:+CLCC, value:(null)
bluealsa: D: ba-rfcomm.c:110: Sending AT message: RAW: command:
OK
, value:(null)

bluealsa: D: io.c:46: BT socket disconnected: Connection reset by peer
bluealsa: D: ba-transport.c:462: Closing BT socket duplicate [29]: 31
bluealsa: D: sco.c:504: IO loop: EXIT: sco_msbc_dec_thread: HFP Audio Gateway (mSBC)
bluealsa: D: ba-transport.c:1983: Exiting IO thread [ba-sco-dec]: HFP Audio Gateway (mSBC)
bluealsa: D: ba-transport.c:462: Closing BT socket duplicate [29]: 30
bluealsa: D: ba-transport.c:903: Releasing SCO link: 29
bluealsa: D: ba-transport.c:1983: Exiting IO thread [ba-sco-enc]: HFP Audio Gateway (mSBC)

Sleepalltheday, 2024-04-28:
I found that opening the PCM channel can establish the SCO connection, so should I proceed with this approach?

borine, 2024-04-29:
> I found that opening the PCM channel can establish the SCO connection

When using BlueALSA, that is in fact the **only** way to establish the SCO connection. There is an example in the wiki of using `bluealsa-rfcomm` to simulate a phone transferring call audio to a hands-free device; see [RFCOMM with HFP](https://github.com/arkq/bluez-alsa/wiki/Using-BlueALSA-with-HFP-and-HSP-Devices#rfcomm-with-hfp), or if you are using an ALSA application you can use a PCM plugin to take care of the rfcomm messages for you; see [HFP AG Helper Plugin for BlueALSA](https://github.com/borine/bluealsa-hfp-ag-plugin)

borine, 2024-08-13:
closing due to lack of activity

closed 2024-08-13 by borine