arkq/bluez-alsa #710: LDAC shows only 96KHz as sample rate | support for other sampling rates

dstagger, 2024-05-23:
### Problem

```
$ bluealsa-cli info /org/bluealsa/hci0/dev_XXX/a2dpsrc/sink
Device: /org/bluez/hci0/dev_XXX
Sequence: 0
Transport: A2DP-source
Mode: sink
Running: true
Format: S32_LE
Channels: 2
Sampling: 96000 Hz <<< support for other rates?
Available codecs: SBC LDAC
Selected codec: LDAC
Delay: 200.1 ms
DelayAdjustment: 0.0 ms
SoftVolume: true
Volume: L: 127 R: 127
Muted: L: false R: false
```
### Additional context
On Windows, all sample rate families (44.1, 48, 88.2 and 96 KHz) are supported using the same earbud (Sony WF-1000XM5) and bluetooth adapter. Is this possible with bluealsa?

borine, 2024-05-23:
Assuming you are using ALSA applications, then you can avoid the need to use `bluealsa-cli` after every connection by defining a PCM with the desired codec configuration in ~/.asoundrc, for example:
```
pcm.earbuds {
	type plug
	slave.pcm {
		type bluealsa
		device XX:XX:XX:XX:XX:XX
		profile a2dp
		codec ldac:2D010000AA001001
	}
}
```
Then use that PCM instead of `bluealsa:XX:XX:XX:XX:XX:XX` as in:
aplay -D earbuds music.wav

dstagger, 2024-05-24:
Sincerely appreciate your prompt and super useful response!

I just need to let the application which is using the ALSA backend, select from supported sampling rates. So in the example above, if I choose the codec `ldac:2D010000AA003F07` the application should be able to pick any one of those sampling rates?

arkq, 2024-05-24:
> if I choose the codec ldac:2D010000AA003F07 the application should be able to pick any one of those sampling rates?

Not exactly. You have to choose specific configuration (when decoding configuration blob with `a2dpconf` you can not see more than one value next to each property). Otherwise, BlueALSA will not reconfigure codec (you will see error message).

dstagger, 2024-05-28:
Thanks for the explanation. Closing the issue.

dstagger, 2024-05-29:
How can I see the other codecs e.g. `LDAC:2d010000aa000401`? I can only see the one which shows all sampling frequencies `LDAC:2d010000aa003c07`:
```
$ bluealsa-cli -v info /org/bluealsa/hci0/dev_XXX/a2dpsrc/sink
Device: /org/bluez/hci0/dev_XXX
Sequence: 0
Transport: A2DP-source
Mode: sink
Running: false
Format: S32_LE
Channels: 2
Sampling: 96000 Hz
Available codecs: SBC:3fff0235 LDAC:2d010000aa003c07
Selected codec: LDAC:2d010000aa000401
Delay: 200.0 ms
DelayAdjustment: 0.0 ms
SoftVolume: true
Volume: L: 127 R: 127
```
```
$ ./a2dpconf LDAC:2d010000aa003c07
LDAC <hex:2d010000aa003c07> {
  vendor-id:32 = 0x0000012d [Sony Corporation]
  vendor-codec-id:16 = 0x00aa
  <reserved>:2
  sampling-frequency:6 = 96000 88200 48000 44100
  <reserved>:5
  channel-mode:3 = Stereo DualChannel Mono
}
```

borine, 2024-05-29:
> How can I see the other codecs

I guess you mean "how can I create the configuration blob that chooses a specific sample frequency?"

Unfortunately the answer is "by understanding the semantics of the bit sequence represented by the hex blob, and creating the appropriate bit sequence by hand".

It would be great if there was a tool to do this, but I do not know of any (perhaps someone else can advise if they know of any Linux tools for this?.)

As an example, starting with the full capabilities blob given by the "Available codecs:" above:
```
$ ./a2dpconf LDAC:2d010000aa003c07
LDAC <hex:2d010000aa003c07> {
  vendor-id:32 = 0x0000012d [Sony Corporation]
  vendor-codec-id:16 = 0x00aa
  <reserved>:2
  sampling-frequency:6 = 96000 88200 48000 44100
  <reserved>:5
  channel-mode:3 = Stereo DualChannel Mono
}
```
We see that the first 32 bits identify the capabilities as being for a Sony codec. That is "2d010000", so we just copy that.
```
LDAC:2d010000
```

The next 16 bits ("aa00") identifies the codec ac LDAC, so again we copy that
```
LDAC:2d010000aa0
```

The next 8 bits("3c") have the first 2 unused and the last 6 indicating the frequency. Each bit represents 1 supported frequency, and in this example 4 of the possible 6 are available. So we will need to guess which bit represents which frequency (or we could just look at the bluez-alsa source code here https://github.com/arkq/bluez-alsa/blob/adf9334e30553c9ce7f4f868b75ff6da92d4d857/src/shared/a2dp-codecs.h#L558-L567. Suppose we want to use a sampling rate of 48000; that is bit 4 or `10` hex for the full 8 bits:
```
LDAC:2d010000aa0010
```

The final 8 bits have 5 unused and the the last 3 each indicating a channel mode. Again by guesswork or looking at the bluez-alsa code we select stereo which is bit 0, so `01` hex for the full 8 bits:
```
LDAC:2d010000aa001001
```

arkq, 2024-05-29:
> It would be great if there was a tool to do this, but I do not know of any (perhaps someone else can advise if they know of any Linux tools for this?.)

I think that the `a2dpconf` could be extended/improved a bit. When creating the LDAC config for 48000 Stereo I had to use calculator to convert hex to binary and binary to hex, and that was annoying to be honest :D

One improvement might be in terms of display format. I think that I will try to change current format to be more Wireshark-like (where each bit is shown in case of bitmap fields). The other improvement might be a support for command like options to set particular bitmaps to desired values (in order to create desired configurations or capabilities). There is a set of common fields and some codecs have unique fields, but I think that the options "space" is countable :)

dstagger, 2024-05-30:
Amazing explanation @borine!! But phew...bitwise operations for selecting codec sampling? :D
@arkq a practical enhancement would be allowing all supported sampling rates at the same time, as displayed by the full capabilities blob. The downstream ALSA application can then simply use the requested frequency. A common use case is e.g. streaming from Tidal where sampling rates can vary by track.
Not sure if that's doable though...

arkq, 2024-05-30:
> @arkq a practical enhancement would be allowing all supported sampling rates at the same time, as displayed by the full capabilities blob. The downstream ALSA application can then simply use the requested frequency.

Hmmm... Maybe the "simplest" ideas are the best :D There are some drawbacks there, though. The reconfiguration step is not as fast as for HW PCMs, so there might be some initial lag when starting to play audio with different sampling frequency (or number of channels) than currently selected. So, maybe such feature indeed would be nice, but not as a default option, but as an optional one. It should be doable (I think) :)

dstagger, 2024-06-01:
> but as an optional one. It should be doable (I think) :)

Tell me more, I'm keen to try it! `--enable-sample-rates` :) 
Lag isn't a problem, it doesn't need to be as fast as HW PCMs, (no one is thinking about gapless etc in this context I guess). It just needs to support all supported sampling rates without having to reconfigure everytime using `a2dpconf`, that's pretty much it!

dstagger, 2024-06-12:
@arkq should we change this to an enhancement? Wondering if you see this happening in the short or long term?

arkq, 2024-06-12:
I'm trying to make a PoC for that. Maybe I will squeeze it in an upcoming 4.3.0 release.

dstagger, 2024-06-16:
> I'm trying to make a PoC for that. Maybe I will squeeze it in an upcoming 4.3.0 release (month or two from now).

Awesome! Looking forward to it.

arkq, 2024-07-12:
I'm still working on the PoC (some internal refactoring is required), but in the meantime I've [pushed to master](https://github.com/arkq/bluez-alsa/commit/435bfd363961abb798324451c50e20659bb91185) a change which by default selects sampling rate up to 48 kHz.

arkq, 2024-07-13:
@dstagger I've pushed a PoC here https://github.com/arkq/bluez-alsa/tree/auto-codec I've added support for SBC, AAC and LDAC codecs. I've tested it only with SBC and AAC, though. Now, when opening PCM, the ALSA plugin will check for supported channels and sampling rates and if currently selected one does not match with requested one, the reconfiguration is triggered. This might cause a delay in playback up to 1 second (it depends on headset response time). Please, test it on your side a share any thoughts or encountered issues :D In the final version, this feature will be behind a configuration flag for `bluealsa` PCM. Also, I will have to rework other parts, so this feature will integrate well with custom A2DP configurations, and with HFP profiles.

borine, 2024-07-26:
@arkq This looks good as a default setup, and feels more "alsa-like" in that the application gets to choose the best config for its own purpose without needing any external effort from the user. It will present difficulty for my "multi-client" branch which I will need to investigate, but that must not prevent or delay merging here.

dstagger, 2024-08-08:
@arkq do I need to checkout the [auto-codec](https://github.com/arkq/bluez-alsa/tree/auto-codec) branch and recompile everything to test this?
I also see that you've an issue with the reconfiguration [here](https://github.com/arkq/bluez-alsa/issues/710#issuecomment-2248753846)
Just to sure that it's ready for testing as I've only fully working environment and avoid messing with it as much as possible :D

arkq, 2024-08-08:
> do I need to checkout the [auto-codec](https://github.com/arkq/bluez-alsa/tree/auto-codec)

This feature has been merged into the master, so the auto-codec branch was removed.

> I also see that you've an issue with the reconfiguration https://github.com/arkq/bluez-alsa/issues/710#issuecomment-2248753846
> Just to sure that it's ready for testing as I've only fully working environment and avoid messing with it as much as possible :D

Yes, I think that it's ready for testing. Pull the latest code from master branch and test it please. I've checked it on my side and it works quite well. However, this feature introduces a "drastic" change for the whole ecosystem. Previous behavior was that once the Bluetooth configuration was established, all audio streams ware transmitted with the same setup (unless modified by user). So, if it worked for the first time, it was like 99.999% chance that it will work for future playbacks. Now, every playback can reconfigure A2DP connection. So, it is possible that due to some bug with "not very tested" configuration (e.g. mono with 11025 Hz), all of the sudden playback will not work either due to BlueALSA issue (I've found some bugs with AAC due to this :D), or remote Bluetooth device. So, it would be very much appreciated for users to test this feature as much as possible :D

dstagger, 2024-10-31:
@arkq how do I enable the auto rate change in my existing setup? I've currently `v4.2.0-5-g9d73103` installed

arkq, 2024-11-01:
You need at least 4.3.0 or better yet 4.3.1 (bug fix release). Then auto rate/channels selection should work out of the box (no special configuration required) . However, I'd recommend to use latest master, because in master there is another fix which is kind of related to auto-selection (deadlock fix).

dstagger, 2024-11-01:
Do I need to uninstall 4.2? Or just the following on the existing installation:
```
git clone https://github.com/arkq/bluez-alsa.git
cd bluez-alsa
autoreconf --install --force
mkdir build
cd build
../configure --enable-ldac --enable-systemd --enable-cli
make
sudo make install
```

arkq, 2024-11-01:
Uninstall and install new one. See the migration guide: https://github.com/arkq/bluez-alsa/wiki/Migrating-from-release-4.3.1-or-earlier

dstagger, 2024-11-01:
Yes had figured that out :) 
How do I select the codec that support multiple rates?
```
$ bluealsactl -vv info /org/bluealsa/hci0/dev_AC_80_0A_05_F0_E7/a2dpsrc/sink
Device: /org/bluez/hci0/dev_AC_80_0A_05_F0_E7
Sequence: 1
Transport: A2DP-source
Mode: sink
Running: true
Format: S32_LE
Channels: 2
ChannelMap: FL FR
Rate: 96000 Hz
Available codecs:
	SBC:3fff0235 [channels: 1 2] [rate: 44100 48000]
	LDAC:2d010000aa003c07 [channels: 1 2] [rate: 44100 48000 88200 96000]
Selected codec:
	LDAC:2d010000aa000401 [channels: 2] [rate: 96000]
Delay: 201.4 ms
ClientDelay: 0.0 ms
SoftVolume: true
Volume: 127 127
Mute: off off
```

arkq, 2024-11-01:
LDAC is already selected and transport supports these options: channels: 1 2 rate: 44100 48000 88200 96000; simply play something (with rate different than 96k) using bluealsa ALSA PCM and you will see that the configuration will automatically change.

dstagger, 2024-11-01:
Hmm seems like it has an issue with `alsaloop`. I'm piping the source player output to a loopback device which I created like this:
```
$ cat /etc/modprobe.d/alsa-aloop.conf 
options snd-aloop index=1 pcm_substreams=1 id=Loopback
```
which is used by the alsaloop daemon as
```
$ cat alsahook.sh 
#!/bin/bash
nohup alsaloop -C hw:Loopback,1 -P bluealsa -t 500000 -S 0 > /dev/null 2>&1 &
```
I've configured the player to use Loopback as the playback device. When the audio sample rate changes, it seems I need to stop the alsaloop daemon and restart it *after* the playeback starts for bluez-alsa to pickup the changes.

arkq, 2024-11-01:
Yes, the configuration is changed only when the PCM is reconfigured. It can be opened all the time but hw params need to be changed. I have never been using alsaloop with BlueALSA, so I'm not sure whether there is a possibility to do that... But I guess that it could be done properly with some patches to alsaloop...

dstagger, 2024-11-01:
So just to conclude, rates changes are working!! I just tried between 88.2 and 96 as I'm not worried about lower sampling rates with LDAC :)
Amazing job!

commit 496908ee (2024-07-24): Multi channels and sampling rates mode in ALSA PCM

Fixes #710
commit 044927c4 (2024-08-06): Multi channels and sampling rates mode in ALSA PCM

Fixes #710
commit 13b0bf85 (2024-08-07): Multi channels and sampling rates mode in ALSA PCM

Fixes #710
closed 2024-11-03 by borine