arkq/bluez-alsa #711: Configuring asoundrc for default bluetooth ignored for Internet Browsers?

rogerxxxx, 2024-05-24:
Trying to get some of these only Internet browser audio sources (eg. retail store podcasts) played on a remote bluetooth audio device.

I've used two separate, popularly found on the Internet forums, $HOME/.asoundrc configurations for using ALSA for both, using a variable and for a default system bluetooth audio device.  (eg. JBL bluetooth speaker)

Using the JBL bluetooth speaker with mpv relatively works here almost problem free, once I connect the JBL bluetooth device using a manual script with expect/btmon incantation. (example: $HOME/bin/bt-connectmy-jbl.sh)

However, trying to get audio from Internet browsers (eg. chromium, firefox, seamonkey, ...) playing through the bluetooth audio device is not working at all.  After some fiddling with asoundrc and chromium, I finally just set the asoundrc with a default bluetooth device rather than using VARS and prefixing the browser command with a VAR for ALSA routing to bluetooth, finally achieved at least getting a bluetooth audio "PCM device busy" error from chromium command line.  Didn't help much, as nothing is using the device at all, with subsequent mpv bluetooth audio playing just fine.

Without posting several additional lengthy asoundrc files, I'm already aware of Internet browsers being problematic with routing to bluetooth audio devices, due to power suspend/disconnects, however routing through to an ALSA audio to bluetooth audio devices should be possible using just Linux/ALSA from what other Internet forum/posts have state.  If anything, setting asoundrc to default bluetooth audio device should just work.

Although pulseaudio is installed here (not by choice) alongside apulse (pulseaudio emulation/routing for ALSA), pulseaudio is explicitly disabled on this computer, requiring manually starting pulseaudio; and without posting large amounts of multiple asoundrc files (yet); is there something else I should know or the status changed for piping Internet browser audio through to an ALSA bluetooth connected device?

Another error initially printed by one of the Internet browsers (firefox?), during one configuration, "libasound_module_pcm_bluetooth.so" not found.

DISTRIBUTION: Void Linux

WINDOW MANAGER: dwm, a stripped-down window manager, without using all of the other commonly GNOME/KDE helper applications/programs.

VERSIONS: alsa-firmware-1.2.4_1 alsa-lib-1.2.11_1 alsa-lib-devel-1.2.11_1 alsa-oss-1.1.8_1 alsa-plugins-1.2.7.1_2 alsa-plugins-ffmpeg-1.2.7.1_2 alsa-plugins-pulseaudio-1.2.7.1_2 alsa-plugins-samplerate-1.2.7.1_2 alsa-ucm-conf-1.2.11_1 alsa-utils-1.2.11_1 apulse-0.1.13_2 bluez-alsa-3.1.0_1

Linux 6.6.25_1 #1 SMP PREEMPT_DYNAMIC Fri Apr  5 01:38:23 UTC 2024 x86_64 GNU/Linux

$ bluealsa --version
v3.1.0

> (`git log -1 --oneline`), and the used configure options.

One hiccup I've seen in the past, sometimes a program (eg. mutt, bluealsa, ...) might not have been configured with proper compile options; but cannot foresee any thing affecting my scenario.

Any other ideas before posting a ton of logs and configuration files?

borine, 2024-11-03:
closing due to lack of activity

closed 2024-11-03 by borine

rogerxxxx, 2024-11-03:
I never could get bluetooth audio working as desired, nor have I had any free time during Summer.  Most of my free time working with computers tends to be during the colder ruts of Winter.

Regardless, as with anything being wireless, really lost interest with the working functionality of bluetooth.