arkq/bluez-alsa #713: LE Audio support in bluez-alsa

teddyxiong53, 2024-05-27:
### Feature description

> Now I try add LE audio function in my buildroot system. I now use the pipewire, but I think it is too hard to use. Can bluealsa support LE Audio? Just like how it support the A2DP. Thank you for your great work.

arkq, 2024-05-27:
> Can bluealsa support LE Audio? Just like how it support the A2DP.

As for now, BlueALSA does not support LE Audio. I've been thinking about that, but there are other things to do and LE Audio is very low priority for me right now (it would be another big feature which will require maintenance, at least at the very early stage of integration with BlueALSA). But please stay tuned, one day it will for sure be added to this project.

Asikbct, 2024-10-25:
@teddyxiong53 , May I know if you have any analysis on LE Audio development? 
One general question, To make LE audio work, Do we need to update the kernel version?
Is there any kernel/driver requirement here?