arkq/bluez-alsa #714: Invalid argument changing the ALSA playback volume in bluealsa-aplay

Vinchethescript, 2024-06-02:
### Problem
Yesterday, I reinstalled bluealsa (as a sink) after a fresh Raspberry Pi OS Bookworm install on my Raspberry Pi 4, but the volume control from my phone wasn't working, even though I could see the volume changes from `bluetoothctl`.

I noticed, though, that it does change the volume of the `bluealsa` capture device, even though I can't hear the difference: when the volume is set to any value except 0, you always hear it at the maximum volume; when it's 0, the audio is just muted.

### Reproduction steps
Simply run bluealsa and bluealsa-aplay. I used these parameters: `-p midi --midi-advertisement -p a2dp-source -p a2dp-sink --a2dp-volume -c sbc -c aac -c aptx -c aptx-hd -c ldac --ldac-abr --ldac-quality=high`. The volume will not change.

### Setup
- Raspberry Pi OS Bookworm (64-bit) 
- The latest version at the time of writing (`v4.2.0`)
- BlueZ `5.66`
- ALSA `1.2.8`
- Built from branch `master` and release tag `v4.2.0` (`261f184 (HEAD, tag: v4.2.0) Minor version bump (release 4.2.0)`).
- Configure options: `../configure --enable-manpages --enable-aac --enable-ldac --with-libopenaptx --enable-mp3lame --enable-mpg123 --enable-aptx --enable-aptx-hd --enable-msbc --enable-rfcomm --enable-hcitop --enable-systemd --enable-cli --enable-faststream --enable-midi --enable-ofono --enable-upower --enable-a2dpconf --with-systemdbluealsaargs='-S -p midi --midi-advertisement -p a2dp-source -p a2dp-sink --a2dp-volume -c sbc -c aac -c aptx -c aptx-hd -c ldac --ldac-abr --ldac-quality=high'`

### Additional context
Checking the logs (even with the `--enable-debug` flag on the `./configure` command and recompiling), the only actually useful info I could see was:
```
bluealsa-aplay: [29403] E: ../../../utils/aplay/aplay.c:349: Couldn't get ALSA mixer playback dB level: Invalid argument
bluealsa-aplay: [29339] E: ../../../utils/aplay/aplay.c:433: Couldn't set ALSA mixer playback dB level: Invalid argument
```
on the logs of `bluealsa-aplay`.

borine, 2024-06-02:
In general it is not a good idea to run bluealsa and pipewire together unless you really know what you are doing.

So, first I assume you have disabled the `pipewire` bluetooth support (you cannot have both `pipewire` and `bluealsa` fighting over A2DP and expect to have any kind of stable Bluetooth audio from either of them).

Second, `pipewire` does not like applications accessing the sound card(s) directly. You will find problems if an application sends audio to pipewire at the same time as `bluealsa-aplay` is sending audio to `plughw`

Third, assuming `pipewire` has installed itselt as `default` ALSA PCM, it appears from the logs that the `pipewire` ALSA pcm plugin `Master` control does not offer dB level settings as required by `bluealsa-aplay`. You can check this by running
`amixer scontents`

Vinchethescript, 2024-06-02:
The audio only passes through PipeWire when running as user, as root it will directly go to ALSA. As soon as I noticed that PipeWire was installed, I instantly disabled and stopped it so it wouldn't give problems, as I don't need it since I only use the audio for BlueALSA. I also tried this with PulseAudio to see if PipeWire was the problem, and it persists, even after setting up the PulseAudio integration as of the wiki page (the bluepulse script/command from the wiki even fails to initialize the module-alsa-sink).

borine, 2024-06-02:
> Opening ALSA mixer: name=default elem=Master index=0
> Couldn't open ALSA mixer: Mixer element not found

I think that error message is very clear: your `default` mixer does not have a control called "Master". As I said in my first comment, use `amixer scontents` to see the available scontrols on the default mixer, and to check if they have dB scale support; and see the [bluealsa-aplay manual page](https://github.com/arkq/bluez-alsa/blob/v4.2.0/doc/bluealsa-aplay.1.rst) for how to use a different control name.

Vinchethescript, 2024-06-02:
Okay, it turned out that the device name was "PCM", so I had to specify it on bluealsa-aplay. There's one issue now: if I set the volume from `alsamixer` to anything higher to 0dB, bluealsa-aplay will crash because of an assertion error:
```

bluealsa-aplay: [79994] D: ../../../utils/aplay/aplay.c:692: Opening ALSA playback PCM: name=default channels=2 rate=96000
bluealsa-aplay: [79994] D: ../../../utils/aplay/aplay.c:708: Opening ALSA mixer: name=default elem=PCM index=0
bluealsa-aplay: [79994] D: ../../../utils/aplay/aplay.c:467: Setting up ALSA mixer volume synchronization
bluealsa-aplay: ../../../utils/aplay/aplay.c:369: io_worker_mixer_volume_sync_ba_pcm: Assertion `volume_db_sum <= 0LL' failed.
[1]    79993 abort      bluealsa-aplay --volume=mixer --mixer-name=PCM --mixer-index=0
```

The Raspberry Pi's maximum volume is at 4 dB, and 0 dB is 96% of the volume. Is it not possible to make it go up to 100%?

borine, 2024-06-02:
A little additional information. I've just found this: https://github.com/raspberrypi/linux/issues/2982

So it seems this apparent anomaly is deliberately coded into the RPi audio driver, although the reasoning that it gives "a bit of extra boost" seems rather odd. The maximum volume is the same no matter if they choose to label it as 0dB or 4dB. How can a label make any difference? Anyway it is what it is. I wonder if this driver is unique in having this seemingly bizarre dB scale?

Vinchethescript, 2024-06-03:
I don't know if only the Raspberry Pi has this 4dB, but at this point I think anyone could change this by just patching the kernel. I don't know if it can be any helpful, but in the `amixer scontents` command I noticed this line:
```
 Limits: Playback -10239 - 400
```
which are the minimum and maximum volume values.

arkq, 2024-06-03:
@Vinchethescript, please try this branch: https://github.com/arkq/bluez-alsa/tree/dbrange

paul-1, 2024-06-03:
Its been that way on Raspi since the early days.  The 4dB is a software(or firmware) boost.   I realize that it makes no difference for bluetooth, but at 0dB it produced bit perfect output on HDMI.

Vinchethescript, 2024-06-03:
> @Vinchethescript, please try this branch: https://github.com/arkq/bluez-alsa/tree/dbrange

This patch didn't change much, as it still reaches up to 0dB. This time, though, there's no assertion error anymore, but when I try to reach a higher value than 0dB through alsamixer or `amixer sset PCM 100%`, it gets back to 0dB. If I try to raise the volume using the arrow buttons, it instead lowers somehow (and I checked if it also happens when bluealsa-aplay isn't running and when audio is not playing - it doesn't).

arkq, 2024-06-03:
> This time, though, there's no assertion error anymore, but when I try to reach a higher value than 0dB through alsamixer or amixer sset PCM 100%, it gets back to 0dB.

Makes sense. I will try to test it locally then. But the approach will be to map 100% to max dB for selected mixer control.

>  I realize that it makes no difference for bluetooth, but at 0dB it produced bit perfect output on HDMI.

Would it make any sense to limit dB level to 0dB and do not allow this boost? My current approach would be to rescale 0-100% to "some very low arbitrary dB level" - "max dB of used mixer control". Such rescaling will allow this 4dB boost for 100% Bluetooth volume, but then it will be rather impossible to set 0dB when linked with Bluetooth controls.

paul-1, 2024-06-03:
For what it's worth, setting 0dB on a raspi was almost impossible to do with the graphical mixer.  The only way we could do it reliably was to use amixer to set the dB to zero.  So I don't see a big deal just scaling based on min and max reported from alsa.

arkq, 2024-06-04:
@Vinchethescript I've just updated the [dbrange](https://github.com/arkq/bluez-alsa/tree/dbrange) branch. Please, try it again.

Vinchethescript, 2024-06-05:
It now works great, and volume can go all the way up both when managed from my phone and from alsamixer. While testing, though, I got into another issue: if I stop the music, then change the volume from my phone after it gets marked as inactive and then play it again, the volume will get back to the previous value. If I change it through alsamixer, instead, it will set it to the value from alsamixer as soon as audio is played.

Vinchethescript, 2024-06-05:
Update: it just crashed because of an assertion error while the device was inactive:
```
bluealsa-aplay: [91798] D: ../../../utils/aplay/aplay.c:735: Opening ALSA playback PCM: name=default channels=2 rate=96000
bluealsa-aplay: [91798] D: ../../../utils/aplay/aplay.c:342: Opening ALSA mixer: name=default elem=PCM index=0
bluealsa-aplay: [91798] D: ../../../utils/aplay/aplay.c:510: Setting up ALSA mixer volume synchronization
bluealsa-aplay: [91798] D: ../../../utils/aplay/aplay.c:646: BT device marked as inactive: F0:39:65:CA:B4:4A
bluealsa-aplay: simple.c:270: snd_mixer_selem_has_playback_channel: Assertion `(elem)->type == SND_MIXER_ELEM_SIMPLE' failed.
[1]    91797 abort      bluealsa-aplay --volume=mixer --mixer-name=PCM --mixer-index=0
```

I don't know what caused this as I wasn't looking at the logs when it happened.

arkq, 2024-06-05:
> Update: it also just crashed because of an assertion error while the device was inactive:

Could you run bluealsa-aplay via gdb and in case of crash send here the backtrace?

Vinchethescript, 2024-06-05:
> Could you run bluealsa-aplay via gdb and in case of crash send here the backtrace?
For some reason, I can't seem to reproduce this error anymore. It happened randomly when I was not playing anything from the phone. Also, in the source code there's no file called `simple.c`, so I don't know where this assertion is located. 

> So when the playback mixer is released, it is not linked with Bluetooth mixer. Then, when playback starts, the mixers are linked and volume is synchronized again.
I see. This also means it can't set the volume as soon as the playback starts, right?

arkq, 2024-06-05:
> This also means it can't set the volume as soon as the playback starts, right?

The volume is synchronized just before the playback starts (from playback PCM to Bluetooth volume - speaker's volume is a master in that case). So, if the playback mixer has -60 dB and phone 100%, phone volume indicator should be turned down.

Vinchethescript, 2024-06-05:
I see. Since the issue has been resolved, I think this can be closed. Thank you!

commit 726581e3 (2024-06-04): Account for playback volume dB max value

Fixes #714
commit 5d8c71cf (2024-06-04): Account for playback volume dB max value

Fixes #714
closed 2024-06-05 by Vinchethescript