arkq/bluez-alsa #717: test-alsa-ctl fails with bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol

pacho2, 2024-06-05:
### Problem

With bluez-alsa-4.2.0, test-alsa-ctl fails on Gentoo (maybe because of it running on a sandboxed environment):
```
FAIL: test-alsa-ctl
===================

EXECV PRELOAD: .libs/aloader.so
WARNING: Couldn't forward ENV variables
Running suite(s): /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c

TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:73: test_controls
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/tmp/dbus-Zrq1L60e7i,guid=b0c3ba8c02e5372d6d7adda766607344
bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol
bluealsa-mock: W: Couldn't get HCI version: Address family not supported by protocol
BLUEALSA_DBUS_SERVICE_NAME=org.bluealsa
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:SBC
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:SBC
bluealsa-mock: E: Couldn't open SCO socket: Address family not supported by protocol
BLUEALSA_READY=SCO:12:34:56:78:9A:BC:(null)

TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:110: test_controls_battery
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/tmp/dbus-pwuKl3Iu8C,guid=3688056ba540ae5c565dad5666607344
bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol
bluealsa-mock: W: Couldn't get HCI version: Address family not supported by protocol
BLUEALSA_DBUS_SERVICE_NAME=org.bluealsa
bluealsa-mock: E: Couldn't open SCO socket: Address family not supported by protocol
BLUEALSA_READY=SCO:23:45:67:89:AB:CD:CVSD

TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:149: test_controls_extended
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/tmp/dbus-nAmJLNISwe,guid=a456854e3dc3828b53c3ba2266607344
bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol
bluealsa-mock: W: Couldn't get HCI version: Address family not supported by protocol
BLUEALSA_DBUS_SERVICE_NAME=org.bluealsa
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:SBC
bluealsa-mock: E: Couldn't open SCO socket: Address family not supported by protocol
BLUEALSA_READY=SCO:12:34:56:78:9A:BC:(null)
ALSA lib /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/src/asound/bluealsa-ctl.c:275:(bluealsa_dev_fetch_battery) Couldn't get device battery status: Object does not exist at path “/org/bluealsa/hci0/dev_23_45_67_89_AB_CD/rfcomm”

TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:229: test_bidirectional_a2dp
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/tmp/dbus-FWpS45XHhk,guid=ef70910b3f170e0e8233fda266607344
bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol
bluealsa-mock: W: Couldn't get HCI version: Address family not supported by protocol
BLUEALSA_DBUS_SERVICE_NAME=org.bluealsa
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:FastStream
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:SBC

TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:265: test_device_name_duplicates
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/tmp/dbus-KYEKXZYwdb,guid=73b235c15ef53406e8ab518466607344
bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol
bluealsa-mock: W: Couldn't get HCI version: Address family not supported by protocol
BLUEALSA_DBUS_SERVICE_NAME=org.bluealsa
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:SBC
TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:359: test_single_device
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/tmp/dbus-TVj5Z7Z0L3,guid=a9af3996f4874fc98c7608ed66607344
bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol
bluealsa-mock: W: Couldn't get HCI version: Address family not supported by protocol
BLUEALSA_DBUS_SERVICE_NAME=org.bluealsa.test
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:SBC
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:SBC

TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:397: test_single_device_not_connected
[... 47 lines cut ...]
BLUEALSA_READY=A2DP:12:34:56:78:9A:BC:SBC
BLUEALSA_READY=A2DP:23:45:67:89:AB:CD:SBC
bluealsa-mock: E: Couldn't open SCO socket: Address family not supported by protocol
BLUEALSA_READY=SCO:12:34:56:78:9A:BC:(null)
92%: Checks: 13, Failures: 1, Errors: 0
/var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:216:F:/var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-ctl.c:test_controls_extended:0: Assertion 'snd_ctl_elem_value_get_enumerated(elem, 0) == sco_codec_enum_items > 1 ? 1 : 0' failed: snd_ctl_elem_value_get_enumerated(elem, 0) == 0, sco_codec_enum_items > 1 ? 1 : 0 == 1
FAIL test-alsa-ctl (exit status: 1)
```

### Setup

> - the OS distribution and version

Gentoo Linux
> - the version of BlueALSA (`bluealsa --version`)

4.2.0

> - the version of BlueZ (`bluetoothd --version`)

5.75

> - the version of ALSA (`aplay --version`)

1.2.11

arkq, 2024-06-09:
It seems that the problem is indeed with `libsandbox.so` being preloaded.

Maybe you are more familiar with running application in Gentoo's sandbox. When running tests, in bluez-alsa, I'm preloading custom library to catch calls to ALSA library in order not to use system-wide ALSA configuration. This works fine outside of the sandbox, but when running `ebuild bluez-alsa-9999.ebuild test` it fails to work. The flow is as follows:

1. Run test application.
2. The `getenv(LD_PRELOAD)` shows: `libsandbox.so`
3. Customize LD_PRELOAD (prepend `.libs/aloader.so` to it)
4. Call `execve()` on self with modified env.
5. The `getenv(LD_PRELOAD)` shows: `.libs/aloader.so:libsandbox.so`
6. The `aloader.so` is detected in LD_PRELOAD, so test application proceeds with tests
7. Run `ldd` on self (i.e.: `sprintf(command, "ldd %s", argv[0])`), which shows:
    ```
    linux-vdso.so.1 (0x00007ffdcb7e8000)
    libsandbox.so => /usr/lib64/libsandbox.so (0x00007f7ee4320000)
    libasound.so.2 => /usr/lib64/libasound.so.2 (0x00007f7ee4237000)
    libcheck.so.0 => /usr/lib64/libcheck.so.0 (0x00007f7ee4229000)
    libc.so.6 => /usr/lib64/libc.so.6 (0x00007f7ee404a000)
    libm.so.6 => /usr/lib64/libm.so.6 (0x00007f7ee3f69000)
    /lib64/ld-linux-x86-64.so.2 (0x00007f7ee435a000)
    ```

The custom library is not preloaded.... So, the test will fail if the bluez-alsa is already installed on the system.

@pacho2, have you even seen similar issues when installing packages from portage?

closed 2024-06-10 by arkq

pacho2, 2024-06-16:
Thanks a lot, it works fine now

commit 6ec69160 (2024-06-10): Fix LD_PRELOAD environment variable overwrite

Fixes #717
commit a51e0603 (2024-06-10): Fix LD_PRELOAD environment variable overwrite

Fixes #717