arkq/bluez-alsa #718: test-alsa-midi fails due to permission errors

pacho2, 2024-06-05:
### Problem

With bluez-alsa-4.2.0, test-alsa-midi fails on Gentoo (maybe because of it running on a sandboxed environment):
```
FAIL: test-alsa-midi
====================

Running suite(s): /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-midi.c

TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-midi.c:59: test_port
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/tmp/dbus-0y8ZErsc7T,guid=52a2a6a5f1bea5560bd7208066607344
bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol
bluealsa-mock: W: Couldn't get HCI version: Address family not supported by protocol
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.11/work/alsa-lib-1.2.11/src/seq/seq_hw.c:528:(snd_seq_hw_open) open /dev/snd/seq failed: Permission denied
bluealsa-mock: E: Couldn't open ALSA sequencer: Permission denied
bluealsa-mock: E: Couldn't acquire local MIDI transport: Permission denied
BLUEALSA_DBUS_SERVICE_NAME=org.bluealsa
cleaning up pid 1665

TEST: /var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-midi.c:100: test_sequencer
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/tmp/dbus-53cGwTxng9,guid=c155554cc43c7529d1bee42766607348
bluealsa-mock: W: Couldn't get HCI device info: Address family not supported by protocol
bluealsa-mock: W: Couldn't get HCI version: Address family not supported by protocol
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.11/work/alsa-lib-1.2.11/src/seq/seq_hw.c:528:(snd_seq_hw_open) open /dev/snd/seq failed: Permission denied
bluealsa-mock: E: Couldn't open ALSA sequencer: Permission denied
bluealsa-mock: E: Couldn't acquire local MIDI transport: Permission denied
BLUEALSA_DBUS_SERVICE_NAME=org.bluealsa
cleaning up pid 2743
0%: Checks: 2, Failures: 0, Errors: 2
/var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-midi.c:59:E:/var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-midi.c:test_port:0: (after this point) Test timeout expired
/var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-midi.c:100:E:/var/tmp/portage/media-sound/bluez-alsa-4.2.0/work/bluez-alsa-4.2.0/test/test-alsa-midi.c:test_sequencer:0: (after this point) Test timeout expired
FAIL test-alsa-midi (exit status: 1)

```

### Setup

> - the OS distribution and version

Gentoo Linux
> - the version of BlueALSA (`bluealsa --version`)

4.2.0

> - the version of BlueZ (`bluetoothd --version`)

5.75

> - the version of ALSA (`aplay --version`)

1.2.11

arkq, 2024-06-05:
> maybe because of it running on a sandboxed environment

It fails when you are trying to emerge bluez-alsa with test USE flag?

closed 2024-06-05 by arkq

pacho2, 2024-06-16:
Thanks a lot for the fix and the work

Regards

commit 4b0ec2e0 (2024-06-05): Check for sequencer read/write access before test

Fixes #718