arkq/bluez-alsa #720: hfp-ag is auto-selected instead of hfp-hf, leads to distorted microphone recording

lowlyocean, 2024-06-11:
### Setup
There are two bluetooth devices:

1. Bluetooth headset (with plain mic)
2. Bluetooth speakerphone (with echo-cancelling mic)

That I've been pairing to two machines:

1. Raspberry Pi OS 64 bit running BlueAlsa (SCO profile for both capture and playback) 
a. I tried both built-in BT chip, and a USB adapter (both show identical behavior)
2. Desktop PC running Ubuntu Jammy and PulseAudio

### Problem
I have an issue when I pair Device 2 (speakerphone) with Machine 1 (Blue Alsa): there is distortion _and_ pitch-shift during when recording from the microphone. It sounds demonic. From what I can tell, I suspect the issue would be related to "hfp-ag" profile being used for the recording source instead of "hfp-hf". I wasn't able to identify any way to explicitly use "hfp-hf " profile for the recording source -- can that be specified explicitly anyhow?

I can confirm that playback of `hcidump -A output.raw` contains the distortion, so I suspect the speakerphone's echo cancellation somehow isn't cooperating with the profile being set to "AG" (because the same device works paired on the PulseAudio machine that appears to be using HFP-HP profile under the hood.. and I can pair some other device to this machine without distortion)

### Additional context

Configuration of Raspberry Pi with BlueAlsa:

`bluealsa -p a2dp-source -p a2dp-sink -p hfp-hf -p hsp-hs -p hfp-ag -p hsp-ag`

```
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (fe010000-1234-5678-abcd-00805f9b34fb)
```

> `pcm.!default {
>     type asym
>     playback.pcm {
>         type plug
>         slave.pcm {
>           type bluealsa
>             device "12:34:56:78:AB:CD"
>             profile "sco"
>         }
>     }
>     capture.pcm {
>         type plug
>         slave.pcm {
>             type bluealsa
>             device "12:34:56:78:AB:CD"
>             profile "sco"
>         }
>     }
> }
> 
> ctl.!default {
>     type bluealsa
> }

Ubuntu machine with PulseAudio:

[image]

[image]

[image]

borine, 2024-06-12:
Please can you state:

* the version of bluealsa that you are using, and if self-built from source the options used for configure.
* The codec and sample rate used by both the devices when used with bluealsa and when used with pulseaudio
* the application used for recording with bluealsa (with settings used for sample format, sample rate and channels)

I don't have any echo-cancelling devices to try this for myself, but if I am given enough information to reproduce the rest of your set-up I may be able to track down some possible causes.

borine, 2024-06-12:
> arecord -vvv ...

The output from that should show the setup of each ALSA plugin used. Please could you post that section of the output so that we can see which plugins are used, and what the setup of each is? It is possible that the pitch-shift may be caused by a plugin; but it is also possible that dropped SCO frames may result in a raised pitch, combined with frequent underruns. (In this case playing from the hcidump file would also result in the raised pitch, but without the underruns).

borine, 2024-09-29:
@lowlyocean is this issue still occuring? It would be good to have more information, especially debug logs from bluealsa, so that we can begin to look for the underlying cause. I think it is not possible to make any more progress on this without those logs.

borine, 2024-11-03:
closing due to lack of activity

closed 2024-11-03 by borine