arkq/bluez-alsa #721: RPI 4: Arecord the audio stream from the phone through Bluezalsa

Arika-Lei, 2024-06-17:
### Problem

My test board is Raspberry 4, l try to use Bluezalsa to arecord the audio stream from the phone, but something errors happened:

root@raspberrypi:/home/pi# arecord -D bluealsa:DEV=A4:CC:B3:7C:24:27 -f S16_le -c 2 -r 44100 test.wav
[2152] D: ../../../src/asound/bluealsa-pcm.c:1473: Getting BlueALSA PCM: CAPTURE A4:CC:B3:7C:24:27 a2dp
[2152] D: ../../../src/asound/bluealsa-pcm.c:1222: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Setting constraints
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
[2152] D: ../../../src/asound/bluealsa-pcm.c:561: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Initializing HW
[2152] D: ../../../src/asound/bluealsa-pcm.c:581: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Couldn't open PCM: Device or resource busy
arecord: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: 125000
PERIOD_SIZE: (5512 5513)
PERIOD_BYTES: 22050
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 22050
BUFFER_BYTES: 88200
TICK_TIME: 0
[2152] D: ../../../src/asound/bluealsa-pcm.c:471: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Closing

### Reproduction steps

l installed 'bluez-alsa-utils' from Debian.
l reboot RPI 4, and then connect the bluetooth from my phone (The bluetooth audio codec l choose SBC). Then, l try 'arecord‘ command to capture the audio stream from my phone, but not success.
Then l try 'bluealsa-aplay -l' to confirm the device and my phone has been recognized.

root@raspberrypi:/home/pi# bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
**** List of CAPTURE Bluetooth Devices ****
hci0: A4:CC:B3:7C:24:27 [Redmi], trusted phone
  A2DP (SBC): S16_LE 2 channels 44100 Hz

### Setup

> - bluez-alsa-utils is already the newest version (4.0.0-2).
> - the version of BlueALSA (`bluealsa --version`): v4.0.0
> - the version of BlueZ (`bluetoothd --version`): 5.66
> - the version of ALSA (`aplay --version`): version 1.2.8

### Additional context

l didn't setup my local .asoundrc file and it is empty in my asound.conf, is there something wrong with that?

Arika-Lei, 2024-06-17:
l stop the bluealsa-aplay service, capture is OK!
But when l replay the recorded audio file, it sounds like...fast forward?2x speed

borine, 2024-06-17:
> l didn't setup my local .asoundrc file and it is empty in my asound.conf

Those files are not necessary, and it is usually easier to debug without them, so that is OK.

>But when l replay the recorded audio file, it sounds like...fast forward?

If you have the ffmpeg package installed, please post the output of
```
ffprobe -hide_banner test.wav
```
or if not, then the output of
```
file test.wav
ls -l test.wav
```
where `test.wav` is the file created by

`arecord -D bluealsa:DEV=A4:CC:B3:7C:24:27 -f S16_le -c 2 -r 44100 test.wav`

Arika-Lei, 2024-06-18:
It's very nice of you to reply so fast.

**that is the result of the command:**
root@raspberrypi:/home/pi# file test.wav
test.wav: RIFF (little-endian) data, WAVE audio, Microsoft PCM, 16 bit, stereo 44100 Hz
root@raspberrypi:/home/pi# ls -l test.wav
-rw-r--r-- 1 root root 5644332 Jun 18 09:39 test.wav

**and that is the process:**
root@raspberrypi:/home/pi# arecord -D bluealsa:DEV=A4:CC:B3:7C:24:27 -f S16_le -c 2 -r 44100 test.wav
[2381] D: ../../../src/asound/bluealsa-pcm.c:1473: Getting BlueALSA PCM: CAPTURE A4:CC:B3:7C:24:27 a2dp
[2381] D: ../../../src/asound/bluealsa-pcm.c:1222: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Setting constraints
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
[2381] D: ../../../src/asound/bluealsa-pcm.c:561: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Initializing HW
[2381] D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: FIFO buffer size: 16384 frames
[2381] D: ../../../src/asound/bluealsa-pcm.c:605: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Selected HW buffer: 4 periods x 22048 bytes <= 88200 bytes
[2381] D: ../../../src/asound/bluealsa-pcm.c:636: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Initializing SW
[2381] D: ../../../src/asound/bluealsa-pcm.c:636: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Initializing SW
[2381] D: ../../../src/asound/bluealsa-pcm.c:677: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Prepared
[2381] D: ../../../src/asound/bluealsa-pcm.c:636: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Initializing SW
[2381] D: ../../../src/asound/bluealsa-pcm.c:388: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Starting
[2383] D: ../../../src/asound/bluealsa-pcm.c:231: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Starting IO loop: 7
^CAborted by signal Interrupt...
arecord: pcm_read:2221: read error: Interrupted system call
[2381] D: ../../../src/asound/bluealsa-pcm.c:424: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Stopping
[2383] D: ../../../src/asound/bluealsa-pcm.c:167: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: IO thread cleanup
[2381] D: ../../../src/asound/bluealsa-pcm.c:615: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Freeing HW
[2381] D: ../../../src/asound/bluealsa-pcm.c:471: /org/bluealsa/hci0/dev_A4_CC_B3_7C_24_27/a2dpsnk/source: Closing

borine, 2024-06-18:
`arecord` always writes a WAV file header of 44 bytes (i.e. no content metadata), followed by the PCM audio frames. So your file contains (5644332 - 44) bytes of audio frames.

The frame size for 2ch s16_le is 4 bytes, so your file contains ((5644332 - 44) / 4) PCM frames.

So, sampled at 44100 frames per second, your file contains 32 seconds of audio.

I can only assume that your playback application is playing it at the wrong speed if it is only taking 2 seconds to play it!

Arika-Lei, 2024-06-18:
l'm sorry for that l had a new record today. The playback application also displays almost 32 seconds, but actually l have played maybe 1m10s on my phone. l have played the 32 seconds audio file, and the speed looks like fast->normal->fast...
l cannot say the exact playtime now because l have broken the environment...

Today l tried to install bluez-alsa from Github, so l remove the 'bluez-alsa-utils' from Debian.
After l installed FDK-AAC from Github, l tried the following steps:

# download the current code
git clone https://github.com/Arkq/bluez-alsa.git
cd bluez-alsa

# configure build environment
# only after all dependencies are done
autoreconf --install --force
mkdir build
cd build

# based on existing Moode build scripts with aac enabled
../configure --disable-hcitop --with-alsaplugindir=/usr/lib/arm-linux-gnueabihf/alsa-lib --enable-aac

# compile and install
make
sudo make install

# reboot and test
sudo reboot

**Problem**
root@raspberrypi:/home/pi# sudo systemctl enable bluealsa.service
Failed to enable unit: Unit file bluealsa.service does not exist.
root@raspberrypi:/home/pi# sudo systemctl enable bluealsa-aplay.service
Failed to enable unit: Unit file bluealsa-aplay.service does not exist.
root@raspberrypi:/home/pi# sudo systemctl start bluealsa.service
Failed to start bluealsa.service: Unit bluealsa.service not found.

l cannot start the bluealsa.service and bluealsa-aplay.service :(

**Additional context**
**First command l tried:**
root@raspberrypi:/home/pi# ls  /usr/lib/arm-linux-gnueabihf/alsa-lib
libasound_module_ctl_bluealsa.a   libasound_module_pcm_bluealsa.a
libasound_module_ctl_bluealsa.la  libasound_module_pcm_bluealsa.la
libasound_module_ctl_bluealsa.so  libasound_module_pcm_bluealsa.so

**Second command l tried:**
root@raspberrypi:/home/pi# sudo bluealsa -i hci0 -p a2dp-source -p a2dp-sink
bluealsa: W: UUID already registered in BlueZ [hci0]: 0000110A-0000-1000-8000-00805F9B34FB
bluealsa: W: UUID already registered in BlueZ [hci0]: 0000110B-0000-1000-8000-00805F9B34FB

**Third command l tried:**
root@raspberrypi:/home/pi# sudo systemctl daemon-reload

**Forth command l tried:**
root@raspberrypi:/home/pi# ls /usr/lib/systemd/system/bluealsa.service
ls: cannot access '/usr/lib/systemd/system/bluealsa.service': No such file or directory
root@raspberrypi:/home/pi# ls /etc/systemd/system/bluealsa.service
ls: cannot access '/etc/systemd/system/bluealsa.service': No such file or directory

**Fifth command l tried:**
root@raspberrypi:/home/pi# service --status-all
 [ - ]  alsa-utils
 [ - ]  apparmor
 [ + ]  bluetooth
 [ - ]  console-setup.sh
 [ + ]  cron
 [ + ]  cups
 [ + ]  cups-browsed
 [ + ]  dbus
 [ + ]  dphys-swapfile
 [ + ]  fake-hwclock
 [ - ]  fio
 [ - ]  hwclock.sh
 [ - ]  keyboard-setup.sh
 [ + ]  kmod
 [ + ]  lightdm
 [ + ]  networking
 [ - ]  nfs-common
 [ - ]  plymouth
 [ + ]  plymouth-log
 [ + ]  procps
 [ + ]  raspi-config
 [ - ]  rpcbind
 [ - ]  rsync
 [ - ]  saned
 [ + ]  ssh
 [ - ]  sudo
 [ + ]  triggerhappy
 [ + ]  udev
 [ - ]  x11-common

Setup
the version of BlueALSA (bluealsa --version): v4.2.0-16-gdd85a7b
the version of BlueZ (bluetoothd --version): 5.66
the version of ALSA (aplay --version): version 1.2.8

borine, 2024-06-18:
> based on existing Moode build scripts

You have not said what OS you are using, but if you are using Moode then please seek help from the Moode maintainers - from what you have said above it appears that Moode does not use the BlueALSA systemd unit files; I guess they must use their own, so you will have to make sure you have them installed.

If you are not using Moode, then its probably best not to use their build scripts because they are clearly specialized for the Moode environment. Instead, follow the guidance on the [wiki](https://github.com/arkq/bluez-alsa/wiki/Installation-from-source) - the configure options for using systemd are explained there. If you need AAC support I suggest you try
```
../configure --enable-systemd --enable-aac 
```

... and also, please make sure you do not have pulseaudio or pipewire running.

Arika-Lei, 2024-06-19:
The OS l am using is Raspberry Pi 4 64bit. l was not aware that l have used another environment (till today l get to know what Moode is).
Fortunately, l follow

Arika-Lei, 2024-06-19:
**root@raspberrypi:/home/pi# ../configure --enable-systemd --enable-aac --enable-aptx --with-libopenaptx --enable-aptx-hd --enable-ldac --enable-cli --with-systemdbluealsaargs="-S -p a2dp-source -p a2dp-sink -c ldac -c aptx -c aptx-hd"**

l try to configure like that, but aptx or ldac is still not effective

Arika-Lei, 2024-06-20:
l try to set my RPI 4's WIFI from 2.4GHz to 5GHz, and then everything becomes OK. l have no ideas about the relationship between the RPI 4's WIFI and the Bluetooth capture.

Arika-Lei, 2024-06-20:
l got a new phenomenon!
Audio source: bluetooth (Bluealsa)
Audio sink: headphone (ALSA hw2:0) , USB  (ALSA hw3:0)
The both sound outputs are lagging, but if l choose one output, sound is smooth.
So maybe it is because the heavy load of ALSA driver? Or the performance of my RPI4?

Arika-Lei, 2024-07-19:
As my phone is the source mode, when l select APTX or APTX-HD in it, after `bluealsa-aplay` on my RPI 4, can't recognize the capture bluetooth devices.
But when l choose SBC or AAC from my phone, my RPI 4 can show the bluetooth address of my phone.
If `bluealsa-h`, APTX and APTX-HD are supported for a2dp-source and a2dp-sink.

Arika-Lei, 2024-07-24:
Unfortunately, l still can't make it possible.
By the way, why l need to use APTX is that the sampling rate is 48000Hz.
If l use SBC or AAC, my capture device is 44100Hz, but the process needs 48000Hz.
l try the ALSA API to resample, but the effect is not so good.
Is it possible to edit `.asoundrc` to set the rate like:

pcm.bluetooth{
             type plug
             slave{
                          device "XX:XX:XX:XX:XX:XX"
                          rate 48000
             }
}

closed 2024-09-02 by Arika-Lei