arkq/bluez-alsa #727: Varying audio latency with bluealsa (Question or Bug)

maxlueerprivate, 2024-09-04:
Hello, I have the following problem OR question and hope someone can help me.

### Problem

I am noticing a varying audio latency when using bluealsa in comparison to a manual audio setup with nearly constant latency.

### Reproduction steps / Setup

> OS: Debian (Kernel 5.15)
> BlueALSA commit-tag: v4.3.0 (--disable-aac --disable-aptx --enable-cli --enable-debug)
> BlueZ 5.77
> aplay Version: 1.2.7

I have two times the same bluetooth adapter and two times the same bluetooth headphones.
I connect one bluetooth headphone to hci0 and one to hci1 where hci0 uses bluealsa (bluealsa -p a2dp-source -i hci0) and hci1 uses my manual SBC audio profile negotiation.

DBus Tree looks like this:
```
Service org.bluez:
  `-/org/bluez
    |-/org/bluez/hci0
    | `-/org/bluez/hci0/dev_5C_56_A4_96_32_93
    |   |-/org/bluez/hci0/dev_5C_56_A4_96_32_93/sep1
    |   | `-/org/bluez/hci0/dev_5C_56_A4_96_32_93/sep1/fd1
    |   `-/org/bluez/hci0/dev_5C_56_A4_96_32_93/sep2
    `-/org/bluez/hci1
      |-/org/bluez/hci1/dev_5C_56_A4_88_4D_F3
      | |-/org/bluez/hci1/dev_5C_56_A4_88_4D_F3/sep1
      | | `-/org/bluez/hci1/dev_5C_56_A4_88_4D_F3/sep1/fd0
      | `-/org/bluez/hci1/dev_5C_56_A4_88_4D_F3/sep2
      `-/org/bluez/hci1/dev_5C_56_A4_96_32_93
Service org.bluealsa:
  |-/org/bluealsa
  | `-/org/bluealsa/hci0
  |   `-/org/bluealsa/hci0/dev_5C_56_A4_96_32_93
  |     `-/org/bluealsa/hci0/dev_5C_56_A4_96_32_93/a2dpsrc
  |       `-/org/bluealsa/hci0/dev_5C_56_A4_96_32_93/a2dpsrc/sink
  `-/org/bluez
    `-/org/bluez/hci0
      `-/org/bluez/hci0/A2DP
        `-/org/bluez/hci0/A2DP/SBC
          `-/org/bluez/hci0/A2DP/SBC/source
            |-/org/bluez/hci0/A2DP/SBC/source/1
            |-/org/bluez/hci0/A2DP/SBC/source/2
            `-/org/bluez/hci0/A2DP/SBC/source/3
```

I am using GStreamer to play a tick audio signal (audiotestsrc wave=ticks) parallel to the bluealsa (audiosink - with buffer time 20ms and period time of 5ms) and manual audio setup (a2dpsink - direct to the dbus fd: /org/bluez/hci1/dev_5C_56_A4_88_4D_F3/sep1/fd0).
Thereby, I am measuring a large audio latency on the bluealsa side of up to ~150ms compared to the manual setup just after starting the pipeline. This audio latency difference is not stable. It decreases to ~50ms after a few seconds and varyies between 150ms and 50ms.
Looking onto bluealsa-cli and watching the "Delay" shows a constant latency between 150ms and 153ms. No variation of 100ms!

bluealsa-cli output:
```
root@DTL0916-002-001-SN000062:/data# bluealsa-cli info /org/bluealsa/hci0/dev_5C_56_A4_96_32_93/a2dpsrc/sink
Device: /org/bluez/hci0/dev_5C_56_A4_96_32_93
Sequence: 0
Transport: A2DP-source
Mode: sink
Running: false
Format: S16_LE
Channels: 2
Sampling: 48000 Hz
Available codecs: SBC
Selected codec: SBC
Delay: 150.6 ms
DelayAdjustment: 0.0 ms
SoftVolume: true
Volume: L: 127 R: 127
Muted: L: false R: false
```

I am wondering why there is this large variance in the latency when using bluealsa and I am unsure if this is a bluealsa or just alsa related.

Maybe someone noticed also a variation in latency or knows how to get rid of this.

Thank you in advance, 
Max

arkq, 2024-09-04:
> Thereby, I am measuring a large audio latency on the bluealsa side of up to ~150ms compared to the manual setup just after starting the pipeline.

First question: how are you measuring this latency?

> I am unsure if this is a bluealsa or just alsa related.

You can cut out ALSA from the picture by using `bluealsactl open ...` (in v4.3.0 `bluealsa-cli`). See https://github.com/arkq/bluez-alsa/blob/master/doc/bluealsactl.1.rst for reference.

maxlueerprivate, 2024-09-04:
Hi @arkq, thank you for your fast reply! I am measuring the latency 
by recording the output of both headphones with a microphone and Audacity. Here for example I have a latency of about 74ms:
[image]

Thanks for providing the command to cut out ALSA. Without ALSA I observe the same behavior.

arkq, 2024-09-04:
> This audio latency difference is not stable. It decreases to ~50ms after a few seconds and varyies between 150ms and 50ms.

The variation in latency is 100ms. So, it looks like your headset is doing something funny with the audio - playback speed is not constant. Is that right? When you play a tone, does the pitch change?

borine, 2025-02-27:
@maxlueerprivate this issue has received no updates for five months. Is it still of interest or can it be closed?

maxlueerprivate, 2025-02-27:
Issue was not bluealsa related. You can close the ticket,  thanks.

closed 2025-02-27 by borine