arkq/bluez-alsa #729: bluealsactl monitor --properties broken

borine, 2024-09-15:
`bluealsactl monitor --properties` was broken by commit 0d488c7. The volume property needs updating to be an array type. Here is a possible quick fix - it may break some existing scripts so you may wish to consider alternative presentations. I can update the manual page if you wish, once a presentation style has been chosen.
```diff
diff --git a/src/bluealsactl/cmd-monitor.c b/src/bluealsactl/cmd-monitor.c
index 1cdeb34..c9d791a 100644
--- a/src/bluealsactl/cmd-monitor.c
+++ b/src/bluealsactl/cmd-monitor.c
@@ -95,11 +95,17 @@ static dbus_bool_t monitor_dbus_message_iter_get_pcm_props_cb(const char *key,
 	}
 	else if (monitor_properties_set[PROPERTY_VOLUME].enabled &&
 			strcmp(key, monitor_properties_set[PROPERTY_VOLUME].name) == 0) {
-		if (type != (type_expected = DBUS_TYPE_UINT16))
+		if (type != (type_expected = DBUS_TYPE_ARRAY))
 			goto fail;
-		dbus_uint16_t volume;
-		dbus_message_iter_get_basic(&variant, &volume);
-		printf("PropertyChanged %s Volume 0x%.4X\n", path, volume);
+		DBusMessageIter iter;
+		uint8_t *data;
+		int len;
+		dbus_message_iter_recurse(&variant, &iter);
+		dbus_message_iter_get_fixed_array(&iter, &data, &len);
+		printf("PropertyChanged %s Volume", path);
+		for (int i = 0; i < len; i++)
+			printf(" %u%s", data[i] & 0x7f, data[i] & 0x80 ? "(M)" : "");
+		printf("\n");
 	}
 
 	return TRUE;
```

arkq, 2024-09-15:
> bluealsactl monitor --properties was broken by commit 0d488c7b671

I've missed that completely... I will have to add some test coverage for that.

> it may break some existing scripts so you may wish to consider alternative presentations

I think that we are already past that. In the next major release (5.0.0) I'd like to squeeze as much refactoring as possible, so next release will not break anything. All changes will be covered in the migration guide (I have to add there changed options in few recent commits, though).

arkq, 2024-09-15:
I've pushed your patch to the https://github.com/arkq/bluez-alsa/tree/fix branch (with a NIT change `(M)` -> `[M]`) . Also, I've added a simple unit test check for volume monitoring.

>  I can update the manual page if you wish, once a presentation style has been chosen.

I'd be glad if you could push update for the `doc/bluealsactl.1.rst` to that branch  (or here as a diff, whatever is easier for you).

commit e9da8aaf (2024-09-15): Fix bluealsactl monitor after changes in 0d488c7

Closes #729
commit 31fc89ea (2024-09-24): Fix bluealsactl monitor after changes in 0d488c7

Closes #729
closed 2024-09-24 by arkq