arkq/bluez-alsa #731: LC3Plus & Opus codecs are not advertised on Android for A2DP-sink

i404788, 2024-10-08:
### Problem
LC3Plus & Opus codecs are not advertised (on Android) for A2DP-sink. All others seem to work fine (except MP3 which isn't supported on android).

I know Opus should work on my devices since I have commercial device with the same codec. I suspect LC3plus should work on at least one of my tested devices. I'm using the developer options 'Trigger Bluetooth Audio Codec Selection` to check what's available, as well as `bluealsactl codec <pcm>`. According to latter Opus should be available (but not LC3Plus), but neither are.

### Reproduction steps

Compile & install bluealsa with all freely distributed codecs (except for LDAC): 
`./configure --enable-aac --enable-lc3-swb --enable-ldac --enable-aptx --enable-aptx-hd --with-libopenaptx --enable-mp3lame --enable-opus --enable-ctl --enable-lc3plus && make -j4 && sudo make install`

Observe codes:
```
Available BT audio codecs:
  a2dp-source:	SBC, MP3, AAC, aptX, aptX-HD, LC3plus, LDAC, Opus
  a2dp-sink:	SBC, MP3, AAC, aptX, aptX-HD, LC3plus, Opus
  hfp-*:	CVSD, LC3-SWB
```

Run with all codecs enabled:
`/usr/bin/bluealsad -S --sbc-quality=xq+ --mp3-algorithm=best --mp3-vbr-quality=extreme --aac-afterburner --ldac-abr --ldac-quality=high --codec=LDAC --codec=MP3 --codec=Opus --codec=LC3plus --codec=aptX --codec=aptX-HD -p a2dp-sink`

Connect android devices; observe only SBC, AAC, aptX & aptX-HD show up as options.

### Setup

lsb_release:
```
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

kernel:
```
Linux raspberrypi 6.6.40-v8 #4 SMP PREEMPT Thu Jul 18 11:53:43 CEST 2024 aarch64 GNU/Linux
```

bluealsad: `v4.3.1-22-ge640aa4`, built from source at `e640aa4c57d35e213e14833b6525de620232f599`
bluetoothd: `5.66`
aplay: `aplay: version 1.2.8 by Jaroslav Kysela <perex@perex.cz>`

i404788, 2024-10-08:
I was able to make a quick pcap dump with [androiddump](https://www.wireshark.org/docs/man-pages/androiddump.html) for the working Opus device: https://nextcloud.devd.pw/s/cQijqiRHs7CEsyx (expires in 7 days for privacy reasons). It was pre-paired but from the capture it seems like it refreshes all capabilities upon connection.

For reference it only support SBC (44.1kHz), AAC (44.1kHz), and Opus (48kHz) and no other options.

arkq, 2024-10-09:
Yeap... it seems there is a typo in the code... please apply this patch and try again:
```patch
diff --git a/src/shared/a2dp-codecs.h b/src/shared/a2dp-codecs.h
index 86cb21b..4e09aad 100644
--- a/src/shared/a2dp-codecs.h
+++ b/src/shared/a2dp-codecs.h
@@ -706,15 +706,15 @@ typedef struct a2dp_lhdc_v5 {
 typedef struct a2dp_opus {
        a2dp_vendor_info_t info;
 #if __BYTE_ORDER == __BIG_ENDIAN
-       uint8_t rfa:2;
        uint8_t sampling_freq:1;
+       uint8_t rfa:2;
        uint8_t frame_duration:2;
        uint8_t channel_mode:3;
 #else
        uint8_t channel_mode:3;
        uint8_t frame_duration:2;
-       uint8_t sampling_freq:1;
        uint8_t rfa:2;
+       uint8_t sampling_freq:1;
 #endif
 } __attribute__ ((packed)) a2dp_opus_t;
```

Also, I'd like you to ask for a favor. Since your earbuds seem to advertise some "undocumented" configuration, I'd like to "probe" these values. The setup would involve host running `bluealsad -p a2dp-source -c opus` (A2DP source mode). Connect earbuds with the host and then it will be required to ask for custom configuration with `bluealsactl codec -f <PCM-PATH> opus:<CONFIG>`

If you are OK to do some tests, I will upload a branch with modifications required for that.

i404788, 2024-10-11:
I'll test the patch later today, also I'm happy do probe the undocumented opus features for you.

arkq, 2024-10-13:
Here is the branch for options probing: https://github.com/arkq/bluez-alsa/tree/opus-probe

The setup will look like this:
1. Start `bluealsad -p a2dp-source -c opus`
2. Connect with earbuds
3. Run `bluealsactl codec -f <PCM-PATH> opus:e0000000010092`
4. Check if A2DP is still connected, and play some audio
5. Run `bluealsactl codec -f <PCM-PATH> opus:e0000000010052`
6. Check if A2DP is still connected, and play some audio, check whether playback speed is the same as before, whether there are no underruns (audio drops)
7. Run `bluealsactl codec -f <PCM-PATH> opus:e0000000010032`
8. The same check as in 6.

In the worst case scenario, in 5 and 7, A2DP connection will be dropped, so you will have to re-connect earbuds. It's hard to guess what to expect with these extra setups, but my wild guess is that earbuds advertise these extra options by mistake, so you might experience connection drop. But, if lucky, there might be some extra sampling frequencies, most likely something like 16000 and 24000 Hz. These values are not available in Android, but earbuds might support them. So, the playback will be 4x and 2x slower than original file.

i404788, 2024-10-14:
`SBC:21150235`/`SBC:111502351`,: normal playback

`opus:e0000000010092` aka `opus`: Audio device underrun (~2x slowed it seems)

*Both `opus:e0000000010032` and `opus:e0000000010052` cause:*

on bluealsactl: `E: CMD "codec": Couldn't select BlueALSA PCM Codec: Input/output error`
on bluealsad: 
```
bluealsad: E: Invalid configuration: Invalid configuration blob: Invalid frame duration
bluealsad: E: Couldn't set A2DP configuration: GDBus.Error:org.bluez.Error.Failed: Resource temporarily unavailable
bluealsad: E: Couldn't select codec: Opus: Input/output error
```
or 
```
bluealsad: E: Invalid configuration: Invalid configuration blob: Invalid sample rate
bluealsad: E: Couldn't set A2DP configuration: GDBus.Error:org.bluez.Error.Failed: Resource temporarily unavailable
bluealsad: E: Couldn't select codec: Opus: Input/output error
```

After which the pcm is not available anymore and I need to disconnect (it doesn't drop the connect) & scan toggle/pairing-mode & connect again to get it back.

Strangely the available codecs seem to differ from your suggested (and neither worked, I/O error):
```
Available codecs: SBC:3fff0235 Opus:e000000001009b
Selected codec: SBC:21150235
```

None of the opus options I've tried gave normal playback, so potentially there is still an bug in the `source` driver?
----

I've made a capture using hcidump (started on `opus` then went to `sbc`, then to `opus:e000000001009b` which gave `Input/output error` and `Invalid frame duration`) but it doesn't seem quite as interesting as the android dump, maybe I'm missing some setting here: https://nextcloud.devd.pw/s/SLAaQRG3mtBP53s

If you need any more probes let me know.

arkq, 2024-10-14:
> Invalid configuration: Invalid configuration blob: Invalid frame duration

Yeap, it's not that easy to hack the system... I've messed up the `opus-probe` branch :D I'll update it (and test it on my side) and let you know when it's ready.

arkq, 2024-10-14:
I've updated `opus-probe` branch. Please, try the test again. However, before the test remove the earbuds device from the device list and pair it again.

i404788, 2024-10-14:
No more I/O errors, however still glitchy audio (assuming sample-rate mismatch but I'm not sure how to set the correct sr). 
```
Device: /org/bluez/hci0/dev_XX...
Sequence: 1
Transport: A2DP-source
Mode: sink
Running: false
Format: S16_LE
Channels: 2
ChannelMap: FL FR
Rate: 16000 Hz
Available codecs: SBC Opus
Selected codec: Opus
Delay: 300.0 ms
ClientDelay: 0.0 ms
SoftVolume: false
Volume: 127 127
Mute: off off
```

It's a bit hard to tell because of the glitchy audio, but it could be that these features turn on 'spatial audio' (head-tracking audio channels) which would be pretty cool.

arkq, 2024-10-14:
> I was wrong about the sample-rate mismatch and feature; but still glitchy audio and Audio device underrun detected.

The glitchy part I would put aside for now, because I've experienced the same thing on RPi3 recently (with SBC, AAC and other codecs...). It might be due to some conflict with other components on the system. I will look at that as well.

The question that I have is about the audio speed. The changes in the `opus-probe` branch assume that these not-used-by-Android bits are for 24 kHz and 16 kHz sampling rates. So, BlueALSA exposes PCM as 24 kHz and 16 kHz. If earbuds indeed expects 16 kHz sample rate, played audio should be at a normal speed. Otherwise, you would hear >2x speed up in playback speed. So, was the playback speed correct for all these configurations (regardless of glitches)? To test this theory you could make one more test with patch like this:

```patch
diff --git a/src/a2dp-opus.c b/src/a2dp-opus.c
index 030f1d6..c6bb4fb 100644
--- a/src/a2dp-opus.c
+++ b/src/a2dp-opus.c
@@ -52,8 +52,8 @@ static const struct a2dp_bit_mapping a2dp_opus_channels[] = {
 };
 
 static const struct a2dp_bit_mapping a2dp_opus_rates[] = {
-       { OPUS_SAMPLING_FREQ_16000, { 16000 } },
-       { OPUS_SAMPLING_FREQ_24000, { 24000 } },
+       { OPUS_SAMPLING_FREQ_16000, { 48000 } },
+       { OPUS_SAMPLING_FREQ_24000, { 48000 } },
        { OPUS_SAMPLING_FREQ_48000, { 48000 } },
        { 0 }
 };
```

This patch will change the encoder to use 48 kHz regardless of chosen configuration. So, now you should hear playback speed slow down for 24 and 16 kHz (if previously the speed was correct).

i404788, 2024-10-14:
> The changes in the opus-probe branch assume that these not-used-by-Android bits are for 24 kHz and 16 kHz sampling rates ...
> So, was the playback speed correct for all these configurations (regardless of glitches)?

Ah I see then I think it does still correspond with sample-rate then: 
All configurations had the same speed, all need 2x speed-up to sound vaguely correct (ignoring stutter/glitches). I'll try your patch to be sure.

i404788, 2024-10-14:
Hmm the plot thickens. If I apply your patch only 0x92 works (others just don't seem to play anything); but with the original opus-probe it seems like everything plays at the same speed. I'm using `mpv --audio-device=alsa/bluealsa` in case that can be an issue.

To me that seems like it is the (correct) lower sample rates but it's just with glitchy playback due to underrun.

arkq, 2024-10-14:
> others just don't seem to play anything; but with the original opus-probe it seems like everything plays at the same speed

So, these options are for other sampling rates. Many tanks, for testing! I will look into glitches issue now.

arkq, 2024-10-14:
Which version of RPi you are using? I've just tested Opus playback on RPi4 and RPi3 with on-board Bluetooth, and in both cases there are not underruns... However, I'm not streaming to earbuds (of course I do not have earbuds with Opus), but to my laptop. On both RPis I'm running "Debian GNU/Linux 12 (bookworm)" aarch64. I've tested playback with `mpv` just like you. RPis are running in headless mode with WiFi connection.

i404788, 2024-10-16:
I've been testing the earbuds on my laptop (Void Linux rolling with Intel BT chipset), I wanted to test on the RPI as well but haven't found the time yet, I originally assumed BT itself was working OK since SBC doesn't cause underrun. 

I'll try to test on the RPI soon which runs the same as your setup (Debian 12 headless).

arkq, 2024-11-17:
I'm going to close this issue as the original bug was solved. If you will check playback on RPi and you will still have underruns please open a separate issue so the bug could be tracked properly.

closed 2024-11-17 by arkq