arkq/bluez-alsa #733: Volume not updating with bluealsa plugin

Asikbct, 2024-10-24:
### Problem
When our DUT acts as A2DP source and connects then streams audio to external BT speaker using bluealsa plugin. With some model of speakers like realme buds headsets and JBL speakers, when we start to steam audio we could not control volume at initial. later like after 15 to 20 seconds we could able to control the volume of external BT speaker through MediaTransport volume property.

I have tried with the latest master branch when we try to get bluealsa PCM it throws error "**ALSA lib bluealsa-pcm.c:1638:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: Get PCM: Get properties: Incorrect variant for 'Volume': q != a**"

### Reproduction steps
1. connect DUT with external BT speaker.
2. Start audio stream with bluealsa plugin.
3. Audio will start play from external BT speaker.
4. If you increase/decrease the volume from DUT, it will not reflect on the external BT speaker.
5. After some time, Volume can be updated.

### Setup

OS ver: Linux username 5.4.210 #2 SMP PREEMPT Thu Oct 17 10:18:00 IST 2024 aarch64 GNU/Linux
bluealsa ver: v4.1.1
bluetoothd ver: v5.72
aplay: version 1.2.1 by Jaroslav Kysela <perex@perex.cz>

### Additional context

whenever we get volume update issue once started the streaming, that time we get this error print from bluez

> D/bluetoothd[885]: profiles/audio/avctp.c:req_timeout() transaction 4 retry false
> D/bluetoothd: bluetoothd[885]: profiles/audio/avctp.c:req_timeout() transaction 4 retry false

arkq, 2024-10-25:
> ALSA lib bluealsa-pcm.c:1638:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: Get PCM: Get properties: Incorrect variant for 'Volume': q != a

It seems that you are trying to use `bluealsad` and bluealsa ALSA plugin from different builds. There was a non-backward compatible change in 0d488c7b67.

> profiles/audio/avctp.c:req_timeout() transaction 4 retry false

It looks like something is not right with the sink device (speaker/earbuds) then. Alternatively try different versions of BlueZ (maybe there is some bug in BlueZ). If nothing helps, in order to get better user experience with such devices, you will have to switch to software volume. Add `SOFTVOL` option to the ALSA device string, e.g. `bluealsa:DEV=01:23:45:67:89:AB,PROFILE=a2dp,SOFTVOL=yes`

Asikbct, 2024-10-25:
@arkq Thanks for the response. 
`bluealsa:DEV=01:23:45:67:89:AB,PROFILE=a2dp,SOFTVOL=yes`
if I give a PCM device name like the above, Will it work with Absolute volume support or only software also volume handling only? If it does not support abs volume support, how to change the sink speaker's system volume instead of audio volume?

arkq, 2024-10-30:
> if I give a PCM device name like the above, Will it work with Absolute volume support or only software Alsa volume handling?

This will use software volume scaling only (on BlueALSA side).

> How to change the sink speaker's system volume instead of audio volume?

If you speaker has some buttons, then you can have (probably not desired, due to double scaling) possibility to scale volume on the speaker side too. Also, you might try adjusting the D-Bus Volume property of acquired media transport, e.g.: `gdbus call --system -d org.bluez -o /org/bluez/hci0/dev_1C_48_F9_9D_81_5C/fd1 -m org.freedesktop.DBus.Properties.Set org.bluez.MediaTransport1 Volume '<uint16 30>'` when that is possible after that initial 15-20 seconds. But the root cause why it's not possible to control volume has to be either in BlueZ or in your speaker.

Asikbct, 2024-11-12:
I tested with BlueZ version **5.72**, which is already quite recent. Is there still a need to update BlueZ?

I also verified the same behavior using other third-party Bluetooth audio sources. They are able to control the volume immediately upon connecting to the same Realme Buds and JBL speakers.

I'm not certain which audio server they are using to stream audio to the external speaker.

mrfixit2001, 2025-01-28:
This non-backwards compatible change breaks things beyond just bluez. In my case, I'm using the latest bluez 5.79 and there's still issues. Specifically, for me, the latest alsa-utils no longer works.

The relevant breaking change in my case is in dbus-client-pcm.c. For my needs I've just patched in a case to allow the volume property to be queried the "old way" with a type of UINT16 and return the first channel's volume instead of failing. The "new way" is to return an array containing the volume for both channels, which alsa-utils doesn't respect.

Someone here needs to communicate back to the alsa-utils maintainers to update things on their end, or be friendly to those non-developers and make some things backwards compatible ;)

arkq, 2025-01-28:
> Specifically, for me, the latest alsa-utils no longer works.

You are taking about utils from ALSA project, like `amixer`? Or you are referring to bluez-alsa-utils package on Debian? By the way I don't know why they have packaged bluez-alsa project as bluez-alsa-utils, for me it does not make sense at all :D

If you are taking about bluez-alsa (this project) then backward compatible change should not affect you as long as you are using ALL components from the same build. If you are using Debian package, just stick with it. If you want to test latest master, UNINSTALL Debian package and install bluez-alsa from source. There should be no need for any patching.... The only possibility that something is not right is that some 3rd party app is talking with BlueALSA D-Bus system service. If so, could you post here what you are running to see that compatibility issue?

mrfixit2001, 2025-01-28:
The incompatible alsa-util that I ran across is alsaloop, it throws the error referenced in this issue:
`ALSA lib bluealsa-pcm.c:1542:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: Get PCM: Get properties: Incorrect variant for 'Volume': a != q`

> > Specifically, for me, the latest alsa-utils no longer works.
> 
> You are taking about utils from ALSA project, like `amixer`? Or you are referring to bluez-alsa-utils package on Debian? By the way I don't know why they have packaged bluez-alsa project as bluez-alsa-utils, for me it does not make sense at all :D
> 
> If you are taking about bluez-alsa (this project) then backward incompatible change should not affect you as long as you are using ALL components from the same build. If you are using Debian package, just stick with it. If you want to test latest master, UNINSTALL Debian package and install bluez-alsa from source. There should be no need for any patching.... The only possibility that something is not right is that some 3rd party app is talking with BlueALSA D-Bus system service. If so, could you post here what you are running to see that compatibility issue?

arkq, 2025-01-28:
> The incompatible alsa-util that I ran across is alsaloop

OK, so the problem is with bluez-alsa. You are using alsaloop, but together with bluez-alsa, and the issue is related to PCM which alsaloop uses. If you are using bluealsa PCM, which is part of this project, it seems that you have either not uinstalled distribution provided package, or something went wrong with upgrade... I'm not by the PC right now so it's hard for me to provide you with exact commands how to check what is wrong. But the problem seems to be with new bluez-alsa ALSA plug-ins not being installed, or being installed alongside with OLD plug-ins and for some reason ALSA uses the old ones. I'll try to post some checking commands tomorrow.

mrfixit2001, 2025-01-28:
> OK, so the problem is with bluez-alsa. You are using alsaloop, but together with bluez-alsa, and the issue is related to PCM which alsaloop uses. If you are using bluealsa PCM, which is part of this [...]

Appreciate the response! To add clarity - I'm using a custom Linux os built from source via the buildroot toolchain. It does not have a package manager and I'm building and patching all libraries from source.

bluealsad from bluez-alsa us used to allow bluetooth audio devices to properly connect and create the PCMs.
Then alsaloop is used to route the bluetooth audio to the loopback audio adapter. Unless there is a blue-alsa tool available for this purpose, then I will need to use both tools to meet the configuration need of the OS.

In any event - I've patched around the issue as explained earlier. The purpose of my post was to raise the awareness of an  incompatibility between these tools. But if there's a better way to do this, I welcome any ideas! Thanks!

Asikbct, 2025-01-29:
> when that is possible after that initial 15-20 seconds. But the root cause why it's not possible to control volume has to be either in BlueZ or in your speaker.

I found the root cause—it was due to unnecessary handling in our BlueZ stack. I had added some custom code to meet our requirements, which inadvertently caused issues during connection establishment. This modification increased the connection timeout for some profiles, preventing the volume from being set for 15–20 seconds.

After adjusting the logic, everything is now working correctly. Thanks!

closed 2025-01-29 by Asikbct

arkq, 2025-01-30:
@mrfixit2001 

> I'm using a custom Linux os built from source via the buildroot toolchain

Hmm... so it's rather strange because it seems that you are using components from different bluez-alsa commits. The log `ALSA lib bluealsa-pcm.c:1542:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM` shows that you are using ALSA plugin from `v4.3.1` but `bluealsad` component is from master (`bluealsa` was renamed to `bluealsad`). You have to fix your buildroot configuration to use the same SHA for all bluez-alsa components. Otherwise, you are doomed to failure...