arkq/bluez-alsa #735: Bluetooth input (source) incorrectly setting alsa output volume

janui, 2024-11-01:
### Problem

Bluetooth input (source) incorrectly setting alsa output volume

### Reproduction steps

> - bluealsa systemd unit command:
/usr/bin/bluealsad -p a2dp-sink -p a2dp-source -c MP3
> - bluealsa-aplay systemd unit command:
/usr/bin/bluealsa-aplay --pcm=hw:0,0 --mixer-device=hw:0 --mixer-name=Headphone 00:00:00:00:00:00
> - alsa settings for hw:0,0
Simple mixer control '**Headphone**',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 174
  Mono:
  Front Left: Playback 147 [84%] [-10.13dB] [on]
  Front Right: Playback 147 [84%] [-10.13dB] [on]
> - To reproduce
Connect input device via bluetooth and play music.
Check that bluealsa is playing:
`bluealsactl info <pcm>`
Output:
```
Device: <pcm>
Sequence: 1
Transport: A2DP-sink
Mode: source
Running: true
Format: S16_LE
Channels: 2
ChannelMap: FL FR
Rate: 44100 Hz
Available codecs: SBC AAC
Selected codec: AAC
Delay: 53.2 ms
ClientDelay: 0.0 ms
SoftVolume: false
Volume: 127 127
Mute: off off
```
Set bluealsa volume to zero:
`bluealsactl volume <pcm> 0`
Get the alsa volume:
`amixer -c 0 sget Headphone`
The alsa volume should now be set to 0, something like this:
```
Simple mixer control 'Headphone',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 174
  Mono:
  Front Left: Playback 0 [0%] [-65.25dB] [on]
  Front Right: Playback 0 [0%] [-65.25dB] [on]
```
Now set the bluealsa volume to 100%:
`bluealsactl volume <pcm> 127`
Check it:
`bluealsactl volume <pcm>`
Output:
`Volume: 127 127`
Now check the alsa volume:
`amixer -c 0 sget Headphone`
Output:
```
Simple mixer control 'Headphone',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 174
  Mono:
  Front Left: Playback 147 [84%] [-10.13dB] [on]
  Front Right: Playback 147 [84%] [-10.13dB] [on]
```
You can now see that the alsa volume control is set to 84%, whereas it should be 100%. The limits of the alsa volume control are 0 to 174, those of the bluealsa volume control are 0 to 127. It is not translating the bluealsa volume level correctly to the reflect the limits of the alsa device. According to my calculations there does seem to be some sort of volume level translation taking place, but it is not correct.

### Setup

> - the OS distribution and version
Linux 6.6.51+rpt-rpi-2712 RPiOS-bookworm aarch64 (64bit) running on a Raspberry Pi 5B
> - the version of Bluealsa
Version: v4.3.1-24-gf93a6e8
> - extra Bluealsa information:
Adapters: hci0
Profiles:
  A2DP-source : SBC MP3 AAC
  A2DP-sink   : SBC MP3 AAC
Available BT audio codecs:
  a2dp-source:  SBC, MP3, AAC, aptX, aptX-HD, FastStream, LDAC
  a2dp-sink:    SBC, MP3, AAC, aptX, aptX-HD, FastStream
  hfp-*:        CVSD
> - the version of BlueZ
5.66
> - the version of ALSA
aplay: version 1.2.8 by Jaroslav Kysela <perex@perex.cz>
> - if self-built from source, please state the branch and commit
unknown, can be provided if required
### Additional context
Also tried using the software name of the output hardware device with identical results.:
> - bluealsa-aplay systemd unit command:
/usr/bin/bluealsa-aplay --pcm=**hw:CARD=Audio,DEV=0** --mixer-device=**hw:CARD=Audio** --mixer-name=Headphone 00:00:00:00:00:00

borine, 2024-11-02:
I see something similar with an intel PCH card.

@arkq I think this is caused by a typo in commit 0d488c7b
utils/aplay/aplay.c, line 458:

`for (size_t i = 1; i < ba_pcm->channels; i++) {` 

should be

`for (size_t i = 0; i < ba_pcm->channels; i++) {`

arkq, 2024-11-02:
Something is not right with the volume indeed, but I had no time to investigate that yet. My case is that bluealsa-aplay sets the volume lower an lower every playback stop/start. I'm not sure whether that is the same case. 

This start on 1 instead on 0 definitely looks like a typo! I'm off the computer right now, but I will look at it today evening or tomorrow.

borine, 2024-11-02:
There's another problem with volume control, but it appears to be bluez related. With bluez 5.79 (just released yesterday) AVDTP absolute volume no longer works; it is OK with bluez 5.78 and earlier. I will try to determine which commit introduced this failure, but not until much later next week.

@janui you should be OK with bluez 5.66. If you are building bluez-alsa from source, could you try the one-line change in the comment above to see if it fixes your issue?

arkq, 2024-11-02:
@janui I've just pushed borine's patch to master, please try it and let us know whether your issue is fixed by that.

> There's another problem with volume control, but it appears to be bluez related.

Yes, I've observed that too, but also had no time to bisect bluez to find exact commit. The problem with absolute volume might be cause by https://github.com/bluez/bluez/commit/923379409010602254233103689c8ed2ff2a6cc6 and/or https://github.com/bluez/bluez/commit/8baabd2f5c131f902d79e20829b117edffd46d1a

arkq, 2024-11-17:
I'm closing this issues as it seems that the bug was fixed. If during the retest you will still see volume setting problems, please reopen this issue.

Also thanks to borine the bug in BlueZ was fixed too (will be available in the next release): https://github.com/bluez/bluez/commit/de1b3a5ac3991f5cfb9278c2887c0be2b71dc58b

closed 2024-11-17 by arkq