arkq/bluez-alsa #737: Distorted sound after commit 45af1a7

Galicarnax, 2024-11-10:
### Problem

After building the latest version, the sound in BT earbuds is distorted. Reverting to the version I used previously restores the sound quality. I have tried building intermediate commits, and found the one after which the issue arises: 45af1a7.

Looking into `journalctl -u bluealsa.service`, I see nothing suspicious. The systemd service is the default one provided in your project: `ExecStart=/usr/bin/bluealsad -S -p a2dp-source -p a2dp-sink`. The output of `bluealsactl info <pcm>` also shows no difference between the versions:
```
Device: /org/bluez/hci0/dev_B0_4A_6A_0F_87_94
Sequence: 0
Transport: A2DP-source
Mode: sink
Running: true
Format: S16_LE
Channels: 2
ChannelMap: FL FR
Rate: 44100 Hz
Available codecs: SBC AAC
Selected codec: AAC
Delay: 396.9 ms
ClientDelay: 0.0 ms
SoftVolume: true
Volume: 121 121
Mute: off off
```
I can provide any further information on your request.

### Setup
- the OS distribution and version: Arch Linux ARM on Raspberry Pi 4B.
- the version of BlueALSA: v4.3.1-26-g45af1a7
- the version of BlueZ: 5.79-1
- the version of ALSA: 1.2.12
- if self-built from source, please state the branch and commit, and the used configure options: `master` branch, commit `45af1a7`. Configure options: `./configure --enable-cli --enable-aac --enable-systemd`

arkq, 2024-11-10:
Are you using 32 or 64 bit mode?

Galicarnax, 2024-11-10:
```
> uname -m
aarch64
```

arkq, 2024-11-10:
Many thanks for reporting that issues!

closed 2024-11-10 by arkq

Galicarnax, 2024-11-10:
Confirm: no issue with the latest commit. Thanks!

commit 59c953a2 (2024-11-10): Fix audio scaling bug introduced in 45af1a7

Fixes #737