arkq/bluez-alsa #741: bluealsa stops when bluealsa-play is launched (A2DP Sink)

mwetterw, 2024-12-06:
### Problem

bluealsa quits immediately as soon as bluealsa-play is launched.

### Reproduction steps

1. Launch `btattach -P bcm -B /dev/ttyAMA0`
2. Launch `bluetoothd -n -p a2dp`
3. Launch `bluealsa -p a2dp-sink`
4. Pair, Connect, Trust with the Phone to the Raspberry Pi
5. Launch `bluealsa-aplay -v -Dplughw:CARD=Headphones --profile-a2dp`

bluealsa stops unexpectedly.

### Setup

:warning: Note that I'm not sure at all there is a bug. My issue may come from my very specific setup.
I may miss a kernel option, a compilation flag or a configuration file.

- Completely custom embedded Linux distribution (not even Buildroot: built with my own custom tool from sources).
- Run on the Raspberry 3B (64bit build)
- With the Raspberry Pi full-featured "normal" GPU firmware (start.elf and fixup.elf and not the cutdown version)
- With the **upstream** Linux kernel compiled from the sources (and not the Raspberry Pi fork of it)
- With the BCM43438A1 blob firmware loaded into the BT chip
- My kernel build does not activate the kernel modules facility. When a feature of the kernel is enabled, it is built-in into the kernel.
- On my custom distribution, there is no dynamic loader at all, so every single lib and binary is **statically built**.
- I didn't install the bluealsa pcm and ctl libraries in alsa-lib folder. (Am I correct that it's not necessary for a simple A2DP sink scenario?)

#### Versions

Cross-compiler **aarch64-linux-musl-gcc** 11.4.0
Libc **musl** 1.2.5

linux 6.6.63 (aarch64)

pcre2 10.44
libffi 3.4.6
sbc 2.0
fdk-aac 2.0.3
alsa-lib 1.2.13
alsa-utils 1.2.13
ncurses 6.4
readline 8.2
expat 2.6.4
dbus 1.14.10
glib 2.82.2
bluez 5.79
zlib 1.3.1
bluez-alsa 4.3.1

### Additional context

```
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
[... 346 lines cut ...]
CONFIG_CRYPTO_ECHAINIV=y
CONFIG_CRYPTO_ESSIV=y
CONFIG_CRYPTO_USER_API_HASH=y
CONFIG_CRYPTO_USER_API_SKCIPHER=y
CONFIG_CRYPTO_USER_API_AEAD=y
CONFIG_CRYPTO_AES_ARM64_NEON_BLK=y
CONFIG_CRYPTO_AES_ARM64_BS=y
CONFIG_CRYPTO_AES_ARM64_CE=y
CONFIG_CRYPTO_AES_ARM64_CE_BLK=y
CONFIG_CRYPTO_AES_ARM64=y
CONFIG_CRYPTO_AES_ARM64_CE_CCM=y
CONFIG_SYSVIPC=y
CONFIG_SERIAL_DEV_BUS=y
CONFIG_BT=y
CONFIG_BT_HCIUART=y
CONFIG_BT_HCIUART_BCM=y
CONFIG_TTY=y
CONFIG_SERIAL_AMBA_PL011=y
CONFIG_SERIAL_AMBA_PL011_CONSOLE=y
CONFIG_LOCALVERSION="-test"
CONFIG_BLK_DEV_INITRD=y
CONFIG_INITRAMFS_SOURCE="root.list.sort"
CONFIG_CMDLINE="console=ttyS1,115200 ip=dhcp"
CONFIG_CMDLINE_FORCE=y
```

borine, 2024-12-06:
> bluetoothd -p a2dp

Why are you using that option? Try without the `-p a2dp` then you may have more luck.

mwetterw, 2024-12-06:
> > bluetoothd -p a2dp
> 
> Why are you using that option? Try without the `-p a2dp` then you may have more luck. 

At first I did many attempts without restricting the Bluetooth profiles loaded by bluetoothd.
But this makes the situation less clear and creates problems:
`bluetoothd` said "I support many things" to the Bluetooth world. For example, it advertised AVRCP, so my Phone tried to talk AVRCP with bluetoothd.
It triggered issues such has "/dev/input" missing.
That's why I ended up activating `CONFIG_INPUT_UINPUT` in my kernel config to solve this issue.

So to sum up, I restricted the profiles to reduce the scope of the problem in order to see clearer and tackle one problem at a time.
For now, the only goal is to hear the music being played, not be able to use remote control.
Once that issue is solved, I will turn on more profiles, more codecs, etc.

borine, 2024-12-06:
The problem line, for me, in your logs is the bluealsa log entry:
```
bluealsa: [178] E: main.c:141: Lost BlueALSA D-Bus name: org.bluealsa
```
which is occurring when bluealsa tries to open the AVDTP transport stream endpoint. That is odd, why would D-Bus unregister the service name at that point?
I've never seen that before, so my guess is that some other bluetooth function is at play. Try adding other likely bluetooth plugins, one at a time, to see if any make a difference. I would start with the policy plugin:
```
bluetoothd -n -p a2dp,policy
```
Beyond that, I don't see any other clue in the logs as to what is happening here.

arkq, 2024-12-06:
> The problem line, for me, in your logs is the bluealsa log entry:
> `bluealsa: [178] E: main.c:141: Lost BlueALSA D-Bus name: org.bluealsa`

Yes, this seems to be the issue here (at least the very last stage of the issue). To investigate it better please collect logs from `dbus-monitor --system` as root. The important part should be just before the name is lost (maybe few messages before). I've seen things like that and mostly they were related to some issues with D-Bus message processing within a code which uses glib for handling D-Bus connection (message was corrupted which cases connection to be dropped). However, it might be something else like some "required" plugin not being loaded by BlueZ.

mwetterw, 2024-12-06:
Thanks for your help!

You'll find the whole dbus conversation [here](https://github.com/user-attachments/files/18041622/dbus.txt).
I've analyzed the whole dbus session.
My own decoding of the list of "dbus actors" in play:
```
1.0:            bluez
1.5, 1.7:       bluetoothctl
1.6:            bluealsa
1.8, 1.9, 1.10: bluealsa-aplay
```

The whole dbus log seems logical. Except here:

> method call time=1733440786.944519 sender=:1.10 -> destination=org.bluealsa serial=5 path=/org/bluealsa; interface=org.freedesktop.DBus.ObjectManager; member=GetManagedObjects
> method return time=1733440786.946730 sender=:1.6 -> destination=:1.10 serial=22 reply_serial=5
>    array [
>       dict entry(
>          object path "/org/bluealsa/hci0/dev_60_00_01_02_03_04/a2dpsnk/source"
>          array [
>             dict entry(
>                string "org.bluealsa.PCM1"
>                array [
>                   dict entry(
>                      string "Device"
>                      variant                         object path "/org/bluez/hci0/dev_60_00_01_02_03_04"
>                   )
>                   dict entry(
>                      string "Sequence"
>                      variant                         uint32 0
>                   )
>                   dict entry(
>                      string "Transport"
>                      variant                         string "A2DP-sink"
>                   )
> [...]
> **method call time=1733440786.949010 sender=:1.10 -> destination=org.bluealsa serial=6 path=/org/bluealsa/hci0/dev_60_00_01_02_03_04/a2dpsnk/source; interface=org.bluealsa.PCM1; member=Open**
> signal time=1733440786.956571 sender=org.freedesktop.DBus -> destination=:1.6 serial=11 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=**NameLost**
>    string "org.bluealsa"
> signal time=1733440786.956693 sender=org.freedesktop.DBus -> destination=(null destination) serial=14 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=**NameOwnerChanged**
>    string "org.bluealsa"
>    string ":1.6"
>    string ""
> signal time=1733440786.959041 sender=org.freedesktop.DBus -> destination=:1.6 serial=12 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=**NameLost**
>    string ":1.6"
> signal time=1733440786.959138 sender=org.freedesktop.DBus -> destination=(null destination) serial=20 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=**NameOwnerChanged**
>    string ":1.6"
>    string ":1.6"
>    string ""
> **error time=1733440786.959229 sender=org.freedesktop.DBus -> destination=:1.10 error_name=org.freedesktop.DBus.Error.NoReply reply_serial=6
>    string "Message recipient disconnected from message bus without replying"**

It appears that `bluealsa` crashes just after being asked by `bluealsa-aplay` to `Open` the `org.bluealsa.PCM1` `org/bluealsa/hci0/dev_60_00_01_02_03_04/a2dpsnk/source`.

It is not yet clear for me if the loss of the DBus name is a cause or a consequence.
Isn't it just a consequence? Maybe the name is lost because `bluealsa` crashed when trying to do the actions of opening the PCM.
I think we should zoom in in the `bluealsa` code portion opening the PCM and add debug messages.

arkq, 2024-12-06:
Ok, so now, if you can, try to run bluealsa under gdb. You can add breakpoint in the function which logs name lost and when stopped print back trace. Maybe this will show something. If this will not show anything useful then it might be required to add some additional logs or breakpoints to help track the root cause...

borine, 2024-12-06:
> Maybe the name is lost because bluealsa crashed when trying to do the actions of opening the PCM

If bluealsa crashed *before* the name lost event, then there cannot be a "Name Lost" error message in the bluealsa log - but there is such a message in the bluealsa log.

also at the end of the bluealsa log:
```
bluealsa: [178] D: ba-adapter.c:144: Freeing adapter: hci0
```
indicates that the bluealsa daemon does not crash, but shuts down cleanly.

mwetterw, 2024-12-06:
> Ok, so now, if you can, try to run bluealsa under gdb. You can add breakpoint in the function which logs name lost and when stopped print back trace. Maybe this will show something. If this will not show anything useful then it might be required to add some additional logs or breakpoints to help track the root cause...
> 
> EDIT: Before running `bluealsa` with `gdb` please install debug symbols for `libc` and `glib`. Otherwise we will see nothing useful.

I know a bit more. I found the `if` in `bluealsa`'s code making it shutdown.

[bluealsa_pcm_open()](https://github.com/arkq/bluez-alsa/blob/v4.3.1/src/bluealsa-dbus.c#L426) is the handler that is called when `bluealsa-aplay` asks `bluez` via D-Bus to open the PCM.
It fails when calling [ba_transport_pcm_signal_send()](https://github.com/arkq/bluez-alsa/blob/v4.3.1/src/bluealsa-dbus.c#L514).
`ba_transport_pcm_signal_send()` fails because of [this if](https://github.com/arkq/bluez-alsa/blob/v4.3.1/src/ba-transport-pcm.c#L546) and so the function returns `-1`.
So if I understand correctly, we are trying to send a signal to a transport PCM whose state is not running.

Note that I try to launch `bluealsa-aplay` once the Phone is connected, but before the Phone plays any sound.
Is it correct? Or should I first play sound with the Phone, and then start `bluealsa-aplay`?
If I first play sound with the phone before launching `bluealsa-aplay`, I get another error:

```
bluealsa: [251] D: bluez.c:1480: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: [251] E: ba-transport.c:327: Couldn't acquire transport: GDBus.Error:org.bluez.Error.Failed: I/O error
bluealsa: [251] D: bluez.c:1480: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: [251] D: bluez.c:1480: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: [251] E: ba-transport.c:327: Couldn't acquire transport: GDBus.Error:org.bluez.Error.NotAuthorized: Operation Not Authorized
```

I hope this will help us getting closer and closer to my issue. :)

borine, 2024-12-07:
> should I first play sound with the phone, and then start bluealsa-aplay?

you can start bluealsa-aplay whenever you want. I always start it at boot and just leave it running until shutdown.

> It fails when calling ba_transport_pcm_signal_send().

No, that is not a failure. An A2DP sink cannot start an audio stream, it has to wait for the A2DP source device (ie the phone) to do that. The point of sending that signal is that the PCM I/O thread may already be running (e.g. because of the keep-alive option after the previous client connection); but in this case there is no existing audio stream from the phone, so there can be no I/O thread running and so nowhere to deliver that signal.

> bluealsa: [251] E: ba-transport.c:327: Couldn't acquire transport: GDBus.Error:org.bluez.Error.NotAuthorized: Operation Not Authorized

So now we have a genuine error. Bluez is not permitting bluealsa to acquire the transport. That sounds very much like an error in the system config. The Bluez documentation is not very helpful:
https://github.com/bluez/bluez/blob/master/doc/org.bluez.MediaTransport.rst#fd-uint16-uint16-acquire - it just states that it can fail with org.bluez.Error.NotAuthorized without giving any reason for possible causes. So I can't give any hints on where to start looking, but I am still convinced that this issue is related to policy management either in Bluez or DBus, or both.

mwetterw, 2024-12-07:
I finally found the root-cause of my issue!
It was very hard to find. I'm actually lucky to find this now.

I was missing `#define HAVE_UNIX_FD_PASSING 1` in `config.h` when compiling the **D-Bus**.

Thank you again for your help.

closed 2024-12-07 by mwetterw