arkq/bluez-alsa #744: Installing bluez-alsa from last volumio realease on PC won't meet bluez version requirement

ronerone, 2024-12-19:
Hi, trying to compile bluez-alsa from a volumio distribution won't meet the requirement.

volumio@volumio:~/bluez-alsa/build$ sudo ../configure --enable-aac --disable-payloadcheck --enable-aptx --enable-aptx-hd --enable-ldac --enable-msbc --enable-lc3plus --with-bluealsaduser=bluez
...
...
checking for BLUEZ... no
configure: error: Package requirements (bluez >= 5.51) were not met:
Requested 'bluez >= 5.51' but version of BlueZ is 5.50

No OS upgrade available. No libbluetooth-dev newer version.

volumio@volumio:~/bluez-alsa/build$ uname -a
Linux volumio 6.6.32-volumio #2 SMP PREEMPT_DYNAMIC Wed May 29 23:04:42 CEST 2024 x86_64 GNU/Linux

rc  bluez                              5.55-3.1+deb11u1                    amd64        Bluetooth tools and daemons
ii  bluez-obexd                        5.50-1.2~deb10u2                    amd64        bluez obex daemon
ii  bluez-tools                        2.0~20170911.0.7cb788c-2            amd64        Set of tools to manage Bluetooth devices for linux

aplay: version 1.1.8 by Jaroslav Kysela <perex@perex.cz>

building from the git master branch.

Is there any way to workaround this issue?

(my main objective is simply to be able to play music from an old laptop through its embedded bluetooth interface to a standalone bluetooth receiver attached on a hifi system).

I thank you in advance!

Ron.

arkq, 2024-12-19:
> Requested 'bluez >= 5.51' but version of BlueZ is 5.50

If your distribution does not have bluez >= 5.51 the latest version of bluez-alsa which you can use is v4.1.1

ronerone, 2024-12-19:
Thank you!
I'm not really familiar to git tree organization. i'm gonna try to find it!
Ron.

ronerone, 2024-12-19:
i found it there: https://github.com/arkq/bluez-alsa/releases/tag/v4.1.1
is it still git clonable or do I simply download the archive?

ronerone, 2024-12-20:
Requested 'sbc >= 1.5' but version of SBC is 1.4

How can i know which bluez-alsa version will meet my poor environment? by trying them one after the other?
Thank you for your help!

ronerone, 2024-12-20:
I kept digging the issue.
I could find the codec archives, i suppose LDAC_DEC won't be used in my case so maybe no issue with the lack.
and what about libbsd...

checking for ALSA... yes
checking for BLUEZ... yes
checking for DBUS1... yes
checking for GIO2... yes
checking for GLIB2... yes
checking for SBC... yes
checking for LIBBSD... no
checking for AAC... yes
checking for APTX... yes
checking for APTX_HD... yes
checking for LDAC_ABR... yes
checking for LDAC_DEC... no
checking for LDAC_ENC... yes
checking for SPANDSP... yes

So now i'm stuck with the make error relative to undefined reference to `sbc_reinit_a2dp'.

/usr/bin/ld: a2dp-sbc.o: in function `a2dp_sbc_enc_thread':
/home/volumio/bluez-alsa-4.1.0/build/src/../../src/a2dp-sbc.c:229: undefined reference to `sbc_reinit_a2dp'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:675: bluealsa] Error 1
make[3]: Leaving directory '/home/volumio/bluez-alsa-4.1.0/build/src'
make[2]: *** [Makefile:792: all-recursive] Error 1
make[2]: Leaving directory '/home/volumio/bluez-alsa-4.1.0/build/src'
make[1]: *** [Makefile:460: all-recursive] Error 1
make[1]: Leaving directory '/home/volumio/bluez-alsa-4.1.0/build'
make: *** [Makefile:392: all] Error 2

arkq, 2024-12-20:
> undefined reference to sbc_reinit_a2dp'

There was a bug in libsbc 1.4, so that's why the requirement was  bumped to 1.5. You can compile sbc from the latest source, though.  Get it from here: https://kernel.googlesource.com/pub/scm/bluetooth/sbc/

ronerone, 2024-12-20:
thanks a lot, i'm gonna try that!

ronerone, 2024-12-20:
I'm probably not taught enough about compilation... as there is no configure script inside the directory, I don't know how to proceed :/
https://kernel.googlesource.com/pub/scm/bluetooth/sbc/+/b2c20a3ecf5ed00f63c94077a9762b505935ede9

arkq, 2024-12-20:
> there is no configure script inside the directory

```sh
./bootstrap
./configure
make install
```

ronerone, 2024-12-20:
meanwhile, i could get the precompiled package and its depencies, currently trying with it.
Thanks for the info regarding bootstrap.

ronerone, 2024-12-20:
I could run make and make install correctly, rebooting the system.
May i directly find the bluetooth interface available?

ronerone, 2024-12-20:
my last question was... n00b's one.
As i compiled an old bluez-alsa, i suppose things are not supposed to be the same as described here:
https://github.com/arkq/bluez-alsa?tab=readme-ov-file
currently checking init.d

ronerone, 2024-12-20:
i finally installed all the bluez packages, which allowed me to get the blutooth services.

-bash-5.0# dpkg -l bluez*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name               Version                  Architecture Description
+++-==================-========================-============-===================
ii  bluez              5.50-1.2~deb10u2         amd64        Bluetooth tools and
un  bluez-alsa         <none>                   <none>       (no description ava
un  bluez-audio        <none>                   <none>       (no description ava
ii  bluez-cups         5.50-1.2~deb10u2         amd64        Bluetooth printer d
ii  bluez-firmware     1.2-4                    all          Firmware for Blueto
ii  bluez-hcidump      5.50-1.2~deb10u2         amd64        Analyses Bluetooth
un  bluez-input        <none>                   <none>       (no description ava
un  bluez-network      <none>                   <none>       (no description ava
ii  bluez-obexd        5.50-1.2~deb10u2         amd64        bluez obex daemon
un  bluez-pan          <none>                   <none>       (no description ava
un  bluez-serial       <none>                   <none>       (no description ava
ii  bluez-test-scripts 5.50-1.2~deb10u2         all          test scripts of blu
ii  bluez-test-tools   5.50-1.2~deb10u2         amd64        test tools of bluez
ii  bluez-tools        2.0~20170911.0.7cb788c-2 amd64        Set of tools to man
un  bluez-utils        <none>                   <none>       (no description ava

A lsmod shows there is no use of the bluetooth module:
-bash-5.0# lsmod | grep blue
bluetooth             733184  0
ecdh_generic           12288  1 bluetooth
libaes                 12288  2 bluetooth,aes_generic
crc16                  12288  2 bluetooth,ext4
-bash-5.0#

Anyway, according to some posts i could find, my dell studio 1555 may require a bluez-hid2hci package which doesn't appear here.
This becomes out of my skills :/

ronerone, 2024-12-20:
How can i check if i really have a BT adapter in that laptop? (i'm starting to wonder about it)
-bash-5.0# bluetoothctl -v
bluetoothctl: 5.50
-bash-5.0# dmesg | grep -i blue
[ 2329.850164] Bluetooth: Core ver 2.22
[ 2329.850218] NET: Registered PF_BLUETOOTH protocol family
[ 2329.850220] Bluetooth: HCI device and connection manager initialized
[ 2329.850227] Bluetooth: HCI socket layer initialized
[ 2329.850231] Bluetooth: L2CAP socket layer initialized
[ 2329.850237] Bluetooth: SCO socket layer initialized
-bash-5.0# ~

arkq, 2024-12-20:
> Will it support a usb dongle?

It all depends what software you have installed. For start, you need a kernel which supports Bluetooth USB dongles. Most distribution provided kernels do, but if you have a custom one, then you have to enable Bluetooth stuff - this might give you some hint: https://wiki.gentoo.org/wiki/Bluetooth#Kernel If your kernel has BT support and you have BT adapter attached, then you should see it under `/sys/class/bluetooth`. Typically there is going to be `hci0`. But the first thing to check is actually `bluetoothctl show`. If it shows controller which is `Powered: yes`, then you are good.

ronerone, 2024-12-20:
I really appreciate your help.
I ordered an ASUS BT-500. Let's wait until it's delivered... in about a week.
Have a good night, or day.

ronerone, 2024-12-27:
Hi! I received the dongle today.
I gave a try to solve it by myself.
At the moment, it will require pulseaudio to be running in order to connect to the bluetooth receiver, but no sound.
I followed few information from this tutorial https://community.volumio.com/t/guide-volumio-bluetooth-broadcaster-requires-pulseaudio/8857 but currently encountered two issues:
1- pulseaudio seems not to start as a service and i have to run it manually, which allows BT connection to the receiver (otherwise, after pairing, it remains unconnectable)
2- there is still no sound output from the receiver, which I suppose is due to improper config on my side. 
The fact I followed some of the above tutorial instructions is probably a bad thing as I was seduced by the automatic BT connection and audio routing it described, but that bay be absolutly no more compatible with 

But anyway, this is far better, as now I do have a BT interface :)

Now if you have few minutes to offer me, advising some crucial points to verify, i'll enjoy. I'm not far from reinstalling a windows with media player ^^

ronerone, 2024-12-27:
oh, another weird behavior is the fact that the bluetooth controller (usb dongle) isn't directly available after booting. Each time I have to unplug and replug it. I suppose  it's not related to bluez-alsa.

ronerone, 2024-12-27:
[B03Pro+]# version
Version 5.50
[B03Pro+]# exit
-bash-5.0# bluetoothctl show
Controller MYDONGLEMAC (public)
        Name: volumio
        Alias: volumio
        Class: 0x000c010c
        Powered: yes
        Discoverable: yes
        Pairable: yes
<...>
        Discovering: no

-bash-5.0# bluetoothctl paired-devices
Device MYRECEIVERMAC B03Pro+
-bash-5.0# bluetoothctl info MYRECEIVERMAC
Device MYRECEIVERMAC (public)
        Name: B03Pro+
        Alias: B03Pro+
        Class: 0x00240418
        Icon: audio-card
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
<...>
-bash-5.0#

ronerone, 2024-12-31:
@arkq , I gave up and installed a very light win10, it worked immediatly.
Thanks for your advices anyway.
Best regards,
Ron.

closed 2025-01-01 by arkq