arkq/bluez-alsa #746: BLE-MIDI crashes with SEGV on SYSEX messages

omersiar, 2024-12-24:
### Problem

BLE-MIDI as [server](https://github.com/arkq/bluez-alsa/wiki/Using-BlueALSA-as-BLE-MIDI-server#running-bluealsa-as-a-ble-midi-server) is working just fine for MIDI CC and NOTE messages

It crashes when SYSEX midi messages sent from a mobile phone to server in success.  For example after 4th or 5th send.

### Reproduction steps

Compile and install with --enable-midi option (as of master commit e909d9a1d2b193b4fb8fc12cc1e00f9b2dd6940b)
Run [server](https://github.com/arkq/bluez-alsa/wiki/Using-BlueALSA-as-BLE-MIDI-server#running-bluealsa-as-a-ble-midi-server) as systemd service

```
[Unit]
Description=BLE-MIDI
After=jackd.service
BindsTo=jackd.service
Requires=bluetooth.service

[Service]
User=root
WorkingDirectory=/root
ExecStart=/usr/bin/bluealsad --midi-advertisement -p midi --xapl-resp-name=VDX7-MIDI
Restart=always
RestartSec=3

[Install]
WantedBy=multi-user.target
```
Send SYSEX messages (patches.fm)

### Setup

Raspberry Pi 3A+ aarch64 Linux pi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
v4.3.1-35-ge909d9a
aplay: version 1.2.8 by Jaroslav Kysela <perex@perex.cz>
master e909d9a1d2
e909d9a (HEAD -> master, origin/master, origin/HEAD) Command line option to enable all available codecs
--enable-midi

### Additional context
Started ble-midi.service - BLE-MIDI.
pi systemd[1]: ble-midi.service: Main process exited, code=killed, status=11/SEGV
pi systemd[1]: ble-midi.service: Failed with result 'signal'.

omersiar, 2024-12-24:
[image]

arkq, 2024-12-30:
> It crashes when SYSEX midi messages sent from a mobile phone to server in success.

How you are sending MIDI from your phone? Are you using some app? I'd like to reproduce the setup on my side.

Alternatively apply patch like this: 
```patch
diff --git a/src/midi.c b/src/midi.c
index 0548d0e..76fb772 100644
--- a/src/midi.c
+++ b/src/midi.c
@@ -104,6 +104,8 @@ static gboolean midi_watch_read_ble_midi(GIOChannel *ch,
                return FALSE;
        }

+       hexdump("MIDI", data, len);
+
        snd_seq_event_t ev = { 0 };
        snd_seq_ev_set_source(&ev, t->midi.seq_port);
        snd_seq_ev_set_subs(&ev);
```

configure bluez-alsa with `--enable-debug` and post logs from before the crash (just MIDI dumps)

omersiar, 2025-01-01:
Happy new year.  Thank you for your interest, since I have enabled core-dump for systemd, it never crashed again, I made lots of related and unrelated configuration changes so can not able to find a proper reproducer for this. 

I will let you know if I able to gather more information.

omersiar, 2025-01-02:
https://play.google.com/store/apps/details?id=com.awisman.midiconnectorfree

then use that virtual port with patches.fm to send sysex to VDX7

arkq, 2025-01-02:
Thanks for the info. I've done some tests, but `bluealsad` did not crash. I suspect that the crash might have something to do with BLE packet drops (maybe it would be nice to do some fuzzing test to cover that)... or there is something wrong with this patches.fm site or midiconnector app... On the first run I had a lot of "No message of desired type" warnings which are caused by "incorrect" BLE-MIDI packets. However, such issue goes away after some time of patches.fm usage. Also, I've discovered that it's not possible to send patch bank (due to SysEx message size limit on BlueALSA side). I'll try to improve SysEx support to allow unlimited size.

arkq, 2025-01-03:
I've just pushed the improvement for the SysEx support into the master. Now, it should be possible to transfer entire patch bank via BLE-MIDI.

omersiar, 2025-01-03:
Great 🚀 . Thanks, I remember now, I have clicked on sending entire bank.

arkq, 2025-01-07:
> Thanks, I remember now, I have clicked on sending entire bank.

OK. I will close this issue for now then. If you will encounter a crash again reopen this issue and remember to attach some logs... preferably the BLE-MIDI stream which causes the segfault.

closed 2025-01-07 by arkq