arkq/bluez-alsa #747: Issue with volume control

itsdeyan, 2025-01-02:
Hi,
I'm running BlueZ-ALSA on a headless Arch Linux machine, configured to function as a Bluetooth speaker. Audio routing works correctly with the following command:
`bluealsa-aplay -S --pcm=loudness --mixer-device=hw:A300Pro --mixer-name=Attenuation --mixer-index=0 --single-audio --volume=mixer`
Audio streams from my phone without issues, and I can adjust the volume. However, the `Attenuation` mixer specified in the command isn't being used for volume changes. In `alsamixer`, `Attenuation` and all other mixers/channels remain at 100% regardless of volume adjustments.
Any ideas why the specified mixer is being ignored or how to ensure it's used for volume control? Thanks!

borine, 2025-01-03:
Without any information at all about your system or debug logs from bluealsa or bluez it is impossible for anyone to be certain about the cause of this issue.

I think the most likely cause is Bluez 5.79 which has broken AVRCP remote volume control. See https://github.com/bluez/bluez/issues/1007

If not that, then perhaps the 'phone has bluetooth AbsoluteVolume disabled.

itsdeyan, 2025-01-03:
> I think the most likely cause is Bluez 5.79 which has broken AVRCP remote volume control. See [bluez/bluez#1007](https://github.com/bluez/bluez/issues/1007)

You nailed it. I downgraded Bluez to 5.74 and it works like a charm. Thanks!

closed 2025-01-03 by itsdeyan