arkq/bluez-alsa #748: MPV/MPG123 Buffer Stuttering Audio

rogerxxxx, 2025-01-14:
Have been having long time problems with stuttering bluetooth audio after the bluetooth device/client has been power cycled on/off.  Upon the bluetooth (automatically) reconnecting to the bluetooth server and playing audio using mpg123/mpv, audio then becomes stuttered/pops.

$ xbps-query -s bluez

[*] bluez-5.78_1            Bluetooth tools and daemons
[*] bluez-alsa-3.1.0_1      Bluetooth Audio ALSA Backend
[*] bluez-deprecated-5.78_1 Bluetooth tools and daemons - deprecated tools
[*] libbluetooth-5.78_1     Library to use the Bluez Linux Bluetooth Stack

Having some time for debugging finally after many years and happening upon some debug incantations whereas seemingly little easy debugging instructions (eg. aplay -D bluealsa:SRV=org.bluealsa,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp some.wav, and with mpg123/mpv using "-a"), I found the likely reason(s) for the playback stuttering when using mpv -v:

[ao/alsa] buffersize: 4410 samples
[ao/alsa] period size: 1102 samples
[ao/alsa] device buffer: 4410 samples.
[ao/alsa] using soft-buffer of 22050 samples.

And happened upon a working instance fluke when mpv happened to playback fine or without stuttering:

[ao/alsa] buffersize: 4800 samples
[ao/alsa] period size: 1200 samples
[ao/alsa] device buffer: 4800 samples.
[ao/alsa] using soft-buffer of 24000 samples.

Now setting mpv "--alsa-buffer-time=0 -v" mpv plays with no stutter too!

ao/alsa] buffersize: 16384 samples
[ao/alsa] period size: 4096 samples
[ao/alsa] device buffer: 16384 samples.
[ao/alsa] using soft-buffer of 22050 samples.

Mpg123 was also affect by playback stuttering and popping, and "--devbuffer 30" or "--devbuffer 60" significantly improves playback without stuttering however for intermittent popping and seemingly "--devbuffer 120" seems about right with no apparent popping noises.

So, long story short, is this a known bug fixed within later versions of ALSA Bluez, or likely not fixed?  I have already tried compiling a recent version of bluez-alsa Void Linux package, so far 4.0 compiled but encountered subsequent dbus bugs, and likely as to why Void Linux developers elected not upgrading, likely due to incompatibility issues with the current Void operating system and/or possibly fluctuating compile/install routine.

Surprised nobody has written any really good debugging/troubleshooting webpages for Linux bluetooth.

borine, 2025-01-17:
> is this a known bug fixed within later versions of ALSA Bluez,

Very likely it is; although blueez-alsa v3.1.0 is now 4 years old and I do not remember clearly that far back. If you really need an answer to this question then you must search through the issues for `mpv` or `mpg123` to see if this issue has been reported previously.

Given that you are using a distribution package, your first course of action should be to raise an issue with that distribution so that the package managers are made aware. They can then decide whether to upgrade to a more recent commit, or to produce their own patch to the old code.

rogerxxxx, 2025-01-17:
You're on the ball too, and I already did search for mpv/mpg123 issues within the github.com bluez-alsa closed/open/etc issues.  I pretty much found nothing, along with searching google.com.  Regardless, this bug is surely now well documented! :-)

And, already did make a post on the non-official Void Linux reddit.com site, posting bluez-alsa version was significantly out-dated.  (Didn't get much feedback, if any at all.)  However, like I stated, I'm guessing those packaging ran into packaging problems, as I could almost drop-in (compile via Void Linux xbps-src packaging) bluez-alsa-4.0 version (not being the latest version), but ran into further dbus problems possibly not easily/readily resolved.  When things get finicky rather than simplified (eg. dbus complexity), people readily loose interest!

Go ahead close this bug, and I'll post/follow-up once either I install a recent version, or more likely Void Linux pushes a newer version... if I'm still alive.  Guessing, most using bluetooth will likely just use pulseaudio, while those not depending on bluetooth devices will just use ALSA.

Usually when packages get stable, or the build scripts become stable, nice just having drop-in and compile/install packages/build scripts.  Also makes debugging or falling back to earlier versions from sources extremely easier.

arkq, 2025-01-18:
If you have bluez-5.78 then definitively you should go for latest bluez-alsa (master) or at least 4.3.1.

> but ran into further dbus problems possibly not easily/readily resolved

If you have some problems with D-Bus, it's most likely due to incorrect configuration file or config file being not installed at all. The easiest way to go forward is to uninstall distribution version of bluez-alsa, get fresh copy of master, configure it (without any additional options, for start) and install it (just like in the https://github.com/arkq/bluez-alsa/blob/master/INSTALL.md guide). If that will succeed, then you can try to enable some additional features.

After installation you should have something like:
```console
$ cat /usr/share/dbus-1/system.d/org.bluealsa.conf
<!-- This configuration file specifies the required security policies
     for BlueALSA core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy group="audio">
    <allow send_destination="org.bluealsa"/>
  </policy>

</busconfig>
```

Then make sure that your user is added to the `audio` group. Also, try to search for any other D-Bus system configuration files which might define something for `org.bluealsa` (use ripgrep or recursive grep in directories `/etc/dbus-1` and `/usr/share/dbus-1`). If you will find some, remove it (or move it somewhere else if you want to keep it).

I'm not sure what hardware you are using, but just in case double check that Bluetooth on your system (source, sink, or both) does not perform any scans:
```console
$ bluetoothctl show |grep Discovering
        Discovering: no
```

rogerxxxx, 2025-01-18:
I have most of the previously mentioned already.

Void Linux seems to be installing the dbus file into /etc/dbus-1/system.d/bluealsa.conf, where all the other /etc/dbus-1/ files are.  (The file is mostly identical except for user root, likely an older bluez-alsa included version.)  I just have a really tough time overall understand the entire dbus system, with udev already taxing my thinking skills.

Discovery is showing no, user already within the audio group.

Think we're getting side-tracked with this stuttering bug, posting software packaging bug(s) atop of the stuttering bug.  Granted, good info trying to display how easily upgrading is, just no free time here for playing, why I moved from Gentoo (source-based) Linux distribution to a binary (Void Linux) distribution.

When I get a little free time and remember to, I'll retry rolling/compiling another (current stable version) bluez-alsa Void Linux package; for experimenting where the hang-up/delay is with the Void Linux package upgrading of bluez-alsa package, aside from click-n-play users being distracted by preferring pulse-audio, etc.

arkq, 2025-01-19:
> Think we're getting side-tracked with this stuttering bug, posting software packaging bug(s) atop of the stuttering bug.

Not at all. The case is that since you are using old version of bluez-alsa, I'm not goin to investigate the issue unless I'm sure that this bug still appears in the current master branch. This project is privately funded (by my and other active developers, which is @borine, spare time - and time is money) with warranty "rules" defined by MIT license: the software is provided "as is", without warranty of any kind. Anyway, as spare time allows, we are trying to answer questions and fix bugs which are discovered by others. Sometimes, we are even developing new features which are asked by others!

> aside from click-n-play users being distracted by preferring pulse-audio

And to be honest that's what we are encouraging regular users to do - use PulseAudio (or PipeWire nowadays). These projects have much more funds, so community around them is a better choice to solve Linux general Bluetooth issues/questions. However, bluez-alsa should work for most cases and for uses who have some ALSA knowledge and know what they want - UNIX-like Bluetooth integration with ALSA without sound server (this part is crucial) - bluez-alsa might be a perfect choice.

So, saying all this, I'm really encouraging you to try bluez-alsa current master. Prior-knowlege of D-Bus is really not required here. If master branch does not work, simply post logs from `bluealsad` (configure project with --enable-debug) or `mpv`/`mpg123`. Without actual logs, I'm afraid that there is nothing more that we can do.

rogerxxxx, 2025-01-19:
The "configure project with --enable-debug" was another item I needed for proper debugging!

I usually only briefly scan "configure --help", probably like most other packagers!

rogerxxxx, 2025-01-20:
Keep getting an error:

$ aplay -D bluealsa:DEV=F8:5C:7E:2F:CA:CA,PROFILE=a2dp,SRV=org.bluealsa /tmp/test.wav
ALSA lib bluealsa-pcm.c:1057:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: No such method “GetPCMs”
aplay: main:850: audio open error: No such device

$ ~/src/bluez-alsa/bluez-alsa-4.3.1.mine/utils/aplay/bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: F8:5C:7E:2F:CA:CA [JBL Charge 5], trusted audio-card
  A2DP (SBC): S16_LE 2 channels 44100 Hz
**** List of CAPTURE Bluetooth Devices ****

$ ~/src/bluez-alsa/bluez-alsa-4.3.1.mine/utils/aplay/bluealsa-aplay -L
bluealsa:DEV=F8:5C:7E:2F:CA:CA,PROFILE=a2dp,SRV=org.bluealsa
    JBL Charge 5, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 44100 Hz

arkq, 2025-01-21:
> ALSA lib pcm.c:2722:(snd_pcm_open_noupdate) Unknown PCM bluealsa:DEV=FF:XX:XX:XX:XX:XX,PROFILE=a2dp

It seems that the ALSA configuration was not installed properly. Or it was... Check whether you have directory `/etc/alsa/conf.d` and what is inside that directory. If you have (mostly) symlinks inside, and there is no `20-bluealsa.conf` in it, add a symlink there to `/usr/share/alsa/alsa.conf.d/20-bluealsa.conf` (if such file exists). So you will have:

```console
$ ls -l /etc/alsa/conf.d
...
lrwxrwxrwx 1 root root  44 Jan 1 2020 20-bluealsa.conf -> /usr/share/alsa/alsa.conf.d/20-bluealsa.conf
```

rogerxxxx, 2025-01-21:
Now that I likely had the tarball manually built and possibly working, decided returning to making the packaging script for Void Linux.

Now get the following error:

bluealsa -p a2dp-sink
bluealsa: E: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa

/usr/share/dbus-1/system.d/
/usr/share/dbus-1/system.d/bluealsa.conf

Tried restarting just dbus service, then restarted...

Looking at prior bug/issues, recompiled with "--user" configure options:

configure_args="--enable-aac --enable-aptx --with-libopenaptx
 --enable-manpages --disable-debug --disable-hcitop
--with-bluealsauser=bluealsa --with-bluealsaaplayuser=bluealsa"
hostmakedepends="pkg-config automake libtool python3-docutils"

Also presuming the "-p a2dp-sink" executable option is correct, with not needing the "-p a2dp-source" executable option.

rogerxxxx, 2025-01-21:
Just tried the following without using "--devbuffer 120" after powering cycling the JBL bluetooth speakers, subsequently automatically reconnecting to the bluetooth server, and I hear no stuttering after the speakers are reconnected.  Or so far with initial testing.

mpg123 -a bluealsa:SRV=org.bluealsa,PROFILE=a2dp https://live.ideastream.org/wksu3.128.mp3

So this stuttering bug is likely isolated to older versions such as bluez-alsa-3.1.0_1, the version listed within the current Void Linux distribution package repositories.

borine, 2025-03-25:
It appears that this issue is specific to old, unsupported, versions of the bluez-alsa code and no longer applies to the last release or current master branch. Closing.

closed 2025-03-25 by borine

rogerxxxx, 2025-03-26:
Maybe I forgot to mention, or mentioned else where without updating here (eg. Email); but I compiled an updated current bluez-alsa-4.3.1 on Void Linux, and posted the changes to the Void Linux build template on their reddit forum.

Updated bluez-alsa-4.3.1 (or a subsequent version) fixes this stuttering bug.

Void Linux template changes.

1. File template: Package version increased to 4.3.1

2. File template: The bluealsauser/bluealsaaplayuser needs to be specified within configure.

3. File template: Make depends now requires glib-devel for /usr/bin/gdbus-codegen for configure.

4. File template: No longer requires /etc/dbus-1/system.d/bluealsa.conf, file moved to /usr/share/dbus-1/system.d and included within the main make install.

5. File run: The bluealsa daemon now requires specifying an (initial?) profile, (eg. -p a2dp-source)

Patch for Void Linux bluez-alsa-4.3.1 follows.  Likely will require a reboot, Void Linux runit alsa/bluetooth/bluez-alsa services do not seem to restart cleanly with the updated package.  I've been stable here using bluez-alsa-4.3.1 for the past few months, subsequent bluetooth speaker reconnects without having to rediscover/re-authorize bluetooth connections.

```
--- srcpkgs/bluez-alsa/template.orig 2025-01-13 20:17:10.259619264 -0500

+++ srcpkgs/bluez-alsa/template 2025-01-23 20:46:10.406416425 -0500

 @@ -1,20 +1,21 @@

# Template file for 'bluez-alsa'

pkgname=bluez-alsa

-version=3.1.0

+version=4.3.1

revision=1

build_style=gnu-configure

configure_args="--enable-aac --enable-aptx --with-libopenaptx

- --enable-manpages --disable-debug --disable-hcitop"

+ --enable-manpages --disable-debug --disable-hcitop

+ --with-bluealsauser=_bluez_alsa --with-bluealsaaplayuser=_bluez_alsa"

hostmakedepends="pkg-config automake libtool python3-docutils"

-makedepends="alsa-lib-devel fdk-aac-devel libbluetooth-devel

- libglib-devel libopenaptx-devel sbc-devel"

+makedepends="alsa-lib-devel fdk-aac-devel glib-devel

+ libbluetooth-devel libglib-devel libopenaptx-devel sbc-devel"

short_desc="Bluetooth Audio ALSA Backend"

maintainer="Toyam Cox <Vaelatern@voidlinux.org>"

license="MIT"

homepage="https://github.com/Arkq/bluez-alsa"

changelog="https://raw.githubusercontent.com/Arkq/bluez-alsa/master/NEWS"

distfiles="https://github.com/Arkq/bluez-alsa/archive/v${version}.tar.gz"

-checksum=d9a878d9fd7f18842c9c8ed83bc96cab684753c90f2751d575a3cdd776ec78a5

+checksum=933fe898dfac21fdfeb5f4ffa685c2aa2db9c064d639170ac2652f156e956a2a

system_accounts="_bluez_alsa"

_bluez_alsa_groups="audio"

@@ -28,9 +29,6 @@

}

post_install() {

- vsed -i ${DESTDIR}/etc/dbus-1/system.d/bluealsa.conf \

- -e 's:user="root":user="_bluez_alsa":'

-

vlicense LICENSE

vsv bluez-alsa

}

--- srcpkgs/bluez-alsa/files/bluez-alsa/run.orig 2025-01-21 20:23:20.315742586 -0500

+++ srcpkgs/bluez-alsa/files/bluez-alsa/run 2025-01-21 18:47:23.171329508 -0500

@@ -2,6 +2,8 @@

exec 2>&1

[ -r ./conf ] && . ./conf

+OPTS=${OPTS:- -p a2dp-source}

+

install -d -m0755 -o _bluez_alsa -g audio /run/bluealsa

exec chpst -u _bluez_alsa:audio bluealsa $OPTS
```