arkq/bluez-alsa #752: Stuttering audio, especially when changing volume

s1lviu, 2025-02-08:
### Problem

Audio stream stuttering, especially when changing volume - it starts to stutter. And that stutter starts to remediate when the audio streaming is stopped for some seconds. 

### Reproduction steps

I use the following configuration to link the svxlink software to to my raspberry pi.

```
pi@svxlink:~ $ cat /etc/asound.conf
defaults.bluealsa {
    interface "hci0"
    device "D9:35:75:33:7F:XX"
    profile "a2dp"
    service "org.bluealsa"
}
```

This is how I pass the audio device to the svxlink software:

```
AUDIO_DEV="alsa:bluealsa"
AUDIO_CHANNEL=0
```

### Setup

I'm using a Raspberry Pi 3B+ and a Debian 12 bookworm
```
pi@svxlink:~ $ uname -a
Linux svxlink 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
pi@svxlink:~ $ bluealsad --version
v4.3.1-42-g4a661e8
pi@svxlink:~ $ bluetoothd --version
5.66
pi@svxlink:~ $ aplay --version
aplay: version 1.2.8 by Jaroslav Kysela <perex@perex.cz>
pi@svxlink:~ $ ls
bluez-alsa  bt-auto-reconnect.sh  mmdvm_display  saycharlie  shutdown_button.py  svxlink
pi@svxlink:~/bluez-alsa $ git log -1 --oneline
4a661e8 (HEAD -> master, origin/master, origin/HEAD) Stub code for LHDC codec with other cap versions
pi@svxlink:~/bluez-alsa $
```

### Additional context

A PoC after I change the volume: https://streamable.com/r38e2v

```
08.02.2025 19:03:59: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:03:59: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:03:59: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:03:59: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:03:59: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:03:59: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:03:59: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:03:59: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:03:59: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:03:59: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:03:59: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:03:59: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:03:59: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:03:59: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:03:59: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:03:59: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:04:00: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:04:00: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:04:00: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:04:00: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:04:00: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
[... 35 lines cut ...]
08.02.2025 19:04:00: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:04:00: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:04:00: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:04:00: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:04:00: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:04:00: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:04:00: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:04:00: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:04:01: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:04:01: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:04:01: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:04:01: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:04:01: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:04:01: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:04:01: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:04:01: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:04:01: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:04:01: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:04:01: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:04:01: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
08.02.2025 19:04:01: [2842] D: ../../../src/asound/bluealsa-pcm.c:244: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Pausing IO thread
08.02.2025 19:04:01: [544] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
08.02.2025 19:04:01: [544] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
08.02.2025 19:04:01: [2842] D: ../../../src/asound/bluealsa-pcm.c:258: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread resumed
```

borine, 2025-02-08:
From the logs it looks like the ALSA device is experiencing constant underruns. Either the application is not writing audio frames fast enough, or else the start threshold is too low.

s1lviu, 2025-02-10:
> From the logs it looks like the ALSA device is experiencing constant underruns. Either the application is not writing audio frames fast enough, or else the start threshold is too low.

Thanks, @borine 

Do you have any suggestions, at least to make this problem less disturbing?

borine, 2025-02-12:
It is not clear what you mean by "change the volume". Do you:
 - press buttons on the bluetooth speaker ?
 - press some buttons connected to the RPi ?
 - use some internal software volume control of the application?
 - use alsamixer ? (In which case, which ALSA device and control name do you use?)

I've never used svxlink and know nothing about it. The logs you give contain no error messages, the bluealsa PCM plugin is behaving normally. You need to find out why the application is unable to provide audio frames fast enough. I have never seen a report that operating any kind of volume control has slowed down the throughput of any ALSA device, so that is a complete mystery to me.

s1lviu, 2025-02-13:
> It is not clear what you mean by "change the volume". Do you:
>
> * press buttons on the bluetooth speaker ?
> * press some buttons connected to the RPi ?
> * use some internal software volume control of [...]

Hi,

I have two cases:
— I press the volume buttons on the Bluetooth speaker
— I change the volume programmatically

```
MIXER_CONTROL = "Tellur Callisto A2DP"
BLUEALSA_CARD = "bluealsa"  

... 

 subprocess.run(
        ["amixer", "-D", BLUEALSA_CARD, "sset", MIXER_CONTROL, f"{volume}%"],
        check=True,
    )
```

I mention that even some frames are lost, on wired speakers, the problem never ever occurs. 

Maybe some insights that can help:

```
Fresh log to see the svxlink (audio) config:

10.02.2025 21:55:44: ------ Opus encoder parameters ------
10.02.2025 21:55:44: Frame size           = 320
10.02.2025 21:55:44: Complexity           = 9
10.02.2025 21:55:44: Bitrate              = 20000
10.02.2025 21:55:44: VBR                  = YES
10.02.2025 21:55:44: Constrained VBR      = YES
10.02.2025 21:55:44: Maximum audio bw     = MEDIUMBAND
10.02.2025 21:55:44: Audio bw             = FULLBAND
10.02.2025 21:55:44: Signal type          = VOICE
10.02.2025 21:55:44: Application type     = AUDIO
10.02.2025 21:55:44: Inband FEC           = NO
10.02.2025 21:55:44: Expected Packet Loss = 0%
10.02.2025 21:55:44: DTX                  = NO
10.02.2025 21:55:44: LSB depth            = 16
10.02.2025 21:55:44: --------------------------------------
10.02.2025 21:55:44: ------ Opus decoder parameters ------
10.02.2025 21:55:44: Gain       = 0dB
10.02.2025 21:55:44: --------------------------------------
10.02.2025 21:55:44: ReflectorLogic: Using audio codec "OPUS"
10.02.2025 21:55:44: ReflectorLogic: Talker stop on TG #226: YO2LCF-M
13.02.2025 14:45:00: SimplexLogic: Sending short identification...
13.02.2025 14:45:00: Playing short voice ID
13.02.2025 14:45:00: MultiTx: Turning the transmitter ON
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:1634: Getting BlueALSA PCM: PLAYBACK D9:96:75:33:7F:AB a2dp
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:1389: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Setting constraints
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:566: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Initializing HW
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:648: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: FIFO buffer size: 2048 frames
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:653: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Selected HW buffer: 4 periods x 2048 bytes == 8192 bytes
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:684: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Initializing SW
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:684: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Initializing SW
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:684: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Initializing SW
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
13.02.2025 14:45:00: [546] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
13.02.2025 14:45:00: [644] D: ../../../src/asound/bluealsa-pcm.c:235: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting IO loop: 19
13.02.2025 14:45:03: [546] D: ../../../src/asound/bluealsa-pcm.c:428: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Stopping
13.02.2025 14:45:03: [644] D: ../../../src/asound/bluealsa-pcm.c:171: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: IO thread cleanup
13.02.2025 14:45:03: [546] D: ../../../src/asound/bluealsa-pcm.c:663: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Freeing HW
13.02.2025 14:45:03: [546] D: ../../../src/asound/bluealsa-pcm.c:475: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Closing
[... 106 lines cut ...]
10.02.2025 21:54:12: [552] D: ../../../src/asound/bluealsa-pcm.c:684: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Initializing SW
10.02.2025 21:54:12: [552] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
10.02.2025 21:54:12: [552] D: ../../../src/asound/bluealsa-pcm.c:684: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Initializing SW
10.02.2025 21:54:12: [552] D: ../../../src/asound/bluealsa-pcm.c:734: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Prepared
10.02.2025 21:54:12: [552] D: ../../../src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting
10.02.2025 21:54:12: [1347] D: ../../../src/asound/bluealsa-pcm.c:235: /org/bluealsa/hci0/dev_D9_96_75_33_7F_AB/a2dpsrc/sink: Starting IO loop: 18
10.02.2025 21:54:12: ReflectorLogic: Talker start on TG #226: YO2MPC-P
10.02.2025 21:54:14: ReflectorLogic: UDP frame(s) lost. Expected seq=1649 but received 1651. Resetting next expected sequence number to 1652
10.02.2025 21:54:23: ReflectorLogic: UDP frame(s) lost. Expected seq=2103 but received 2104. Resetting next expected sequence number to 2105
[... cut ...]
10.02.2025 21:55:12: ReflectorLogic: UDP frame(s) lost. Expected seq=4554 but received 4555. Resetting next expected sequence number to 4556
10.02.2025 21:55:14: ReflectorLogic: UDP frame(s) lost. Expected seq=4653 but received 4655. Resetting next expected sequence number to 4656
10.02.2025 21:55:17: ReflectorLogic: UDP frame(s) lost. Expected seq=4803 but received 4805. Resetting next expected sequence number to 4806
10.02.2025 21:55:18: ReflectorLogic: UDP frame(s) lost. Expected seq=4853 but received 4854. Resetting next expected sequence number to 4855
```

borine, 2025-04-20:
@s1lviu has there been any progress with this issue? Have you been able to try with larger ALSA device block sizes for A2DP, or to try the default block size with HFP or HSP?

s1lviu, 2025-04-20:
Hi @borine 
Sorry for the late response. I managed to solve with a €2 tx/rx bluetooth dongle a day after. Hope this thread will help others to fix the issue if needed. Thanks again for your patience and understanding.
All the best!

borine, 2025-04-20:
> I managed to solve with a €2 tx/rx bluetooth dongle a day after

That's good news! Thanks for the reponse, I'm closing this now as completed.

closed 2025-04-20 by borine