arkq/bluez-alsa #753: Bluealsa failing to dowloand and install on pi 5

seanolan2002, 2025-02-12:
> Please read the [troubleshooting guide](../blob/master/TROUBLESHOOTING.md)
> before raising a new issue.

### Problem

I am trying to installl the bluealsa repos on my raspberry Pi 5 OS, If I try to use the sudo apt install command I get the an erroe 'E: Package 'bluealsa' has no installation candidate'. If I dowload it through the source using the steps porvided on  
'https://github.com/arkq/bluez-alsa/wiki/Installation-from-source' its seems to work okay and I can see the file location in the file application but if i search for the address in the terminal it cannot be found and if I used the command 'bluealsa --version' I get 'bash: bluealsa: command not found' in return. 

### Reproduction steps

All 3rd party libaries where successfully installed.
The source code was cloned and built.
The directory was built with a few issues with some libaries but these were not consideredvto be essential.

### Setup

Raspberry Pi OS Debian v12 64 bit full bookworm
git clone https://github.com/arkq/bluez-alsa.git

### Additional context

I have had this issue for over 3 week now and have found no solution or issue rasied liekt his before, any help would be greatly appreciated as it is needed for roughly 30% of my thesis and is really holding me back. Thank you

borine, 2025-02-12:
First of all, you are using the RPiOS Full image which uses PulseAudio for all audio (including Bluetooth). So why do you need BlueALSA? If you are certain that you really want to use BlueALSA then you would first need to disable Bluetooth in PulseAudio, and a likely consequence of that is many of the GUI applications will lose the ability to use Bluetooth audio. Please confirm that you really do have a genuine need for BlueALSA on RPiOS full image, because if so then you have many more issues to solve before even starting the bluealsad daemon.

seanolan2002, 2025-02-13:
I am trying to display metadata from a bluetooth streaming source for a car UI. I wish to display the artist name, song title and time remaining in a song from my research I cannot display this information us pulseaudio as it doesnt get access from the bluetooth communication.

arkq, 2025-02-13:
> I am trying to display metadata from a bluetooth streaming source for a car UI

This project will also not do this for you. Showing player information is not in the scope of BlueALSA service. In order to get that you will need an app which will get it for you from this BlueZ D-Bus API: https://github.com/bluez/bluez/blob/master/doc/org.bluez.MediaPlayer.rst#dict-track-readonly

borine, 2025-02-13:
If you prefer to use an existing application to display the player information rather than writing your own DBus application, you can use the bluez mpris-proxy utility. This should be available from the RPiOS repository (maybe it is even already installed on the full image). This presents the bluetooth media data using the MPRIS protocol which is understood by many linux audio applications. It works with both BlueALSA (bluealsa-aplay) and Pipewire (and PulseAudio).

borine, 2025-02-14:
Are you running the mpris-proxy service? `playerctl` requires an MPRIS player, so to convert a Bluetooth AVRCP player to an MPRIS one you must run `mpris-proxy`

seanolan2002, 2025-02-14:
No, i cannot install it using apt, is there a source that i can clone to install it 

~ $ sudo apt-get install mpris-proxy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package mpris-proxy

borine, 2025-02-14:
I think on RPiOS it is actually part of the bluez package, so it is probably already installed. You will most likely need to write your own systemd service unit file to have it run automatically, but for testing just start it in a terminal:
```console
$ mpris-proxy
```

seanolan2002, 2025-02-14:
Yes it seems to be installed

$ mpris-proxy
org.bluez appeared
Bluetooth Adapter /org/bluez/hci0 found

borine, 2025-02-14:
OK that's great. So now when you connect the bluetooth player, mpris should find that too. For example when I connect my iphone to my bluealsa sink, mpris-proxy outputs:
```console
Bluetooth Player /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/player0 found
Player org.mpris.MediaPlayer2.iPhone created
Bluetooth Transport /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/fd7 found
```
and `playerctl -l` gives
```console
$ playerctl -l
iPhone
```

seanolan2002, 2025-02-14:
This command is working for me too in the terminal but when I run the appplication it fails, I added a section of code to start mpris-proxy but it doesnt seem to make a differnace as I get an error log "Failed to start mpris-proxy". I ahve added this line of code below

    def ensure_mpris_proxy(self):
        try:
            # Attempt to start the mpris-proxy service (this command may vary depending on your system)
            print("Ensuring mpris-proxy is enabled...")

            # Attempt to start mpris-proxy if it's not running
            result = subprocess.run(
                ["dbus-launch", "--exit-with-session", "playerctld", "start"],
                stdout=subprocess.PIPE, stderr=subprocess.PIPE, text=True
            )
            if result.returncode == 0:
                print("mpris-proxy started successfully.")
            else:
                print(f"Failed to start mpris-proxy: {result.stderr}")
        except Exception as e:
            print(f"Error starting mpris-proxy: {e}")

borine, 2025-02-14:
??? That code snippet has nothing to do with mpris-proxy. I think you are now well off-topic for this project, you need more general help with running services, and that has nothing at all to do with bluealsa.

seanolan2002, 2025-02-14:
I agree, thank you for all your help and pointing me in the right direction

closed 2025-02-14 by seanolan2002