arkq/bluez-alsa #754: pkg-config is not fully able to complete configuration

sergey-suloev, 2025-02-14:
> Please read the [troubleshooting guide](../blob/master/TROUBLESHOOTING.md)
> before raising a new issue.

### Problem

> A clear and concise description of what the bug is.
> If possible, please check if the bug still exists in the master branch (or
> the latest release if you are not using it already).

### Reproduction steps

> Provide a minimal example of how to reproduce the problem. State `bluealsad`
> command line arguments and the content of .asoundrc file (if PCM alias with
> "type bluealsa" was added to that file).

### Setup

> - the OS distribution and version
> - the version of BlueALSA (`bluealsad --version`)
> - the version of BlueZ (`bluetoothd --version`)
> - the version of ALSA (`aplay --version`)
> - if self-built from source, please state the branch and commit
> (`git log -1 --oneline`), and the used configure options.

### Additional context

> Add any other context about the problem here, e.g. log messages printed by
> `bluealsad` and/or client application.
>
> Please delete instructions prefixed with '>' to prove you have read them.

sergey-suloev, 2025-02-14:
#!/bin/sh
R=/path/to/sysroot
export PKG_CONFIG_SYSROOT_DIR="${R}"
export PKG_CONFIG_LIBDIR="${R}/lib/pkgconfig:${R}/usr/lib/pkgconfig:${R}/usr/share/pkgconfig"
export PKG_CONFIG_PATH=
exec /usr/bin/pkg-config "$@"

sergey-suloev, 2025-02-14:
user@host: ls -l /path/to/sysroot/usr/lib/pkgconfig/
total 216
-rw-r--r-- 1 root root 271 фев 14 13:19 alsa.pc
-rw-r--r-- 1 root root 152 фев 14 13:19 alsa-topology.pc
-rw-r--r-- 1 root root 191 фев 14 13:15 blkid.pc
-rw-r--r-- 1 root root 214 фев 14 14:49 bluez.pc
-rw-r--r-- 1 root root 262 фев 14 13:16 com_err.pc
-rw-r--r-- 1 root root 616 фев 14 13:10 dbus-1.pc
-rw-r--r-- 1 root root 250 фев 14 13:16 e2p.pc
-rw-r--r-- 1 root root 277 фев 14 13:09 expat.pc
-rw-r--r-- 1 root root 247 фев 14 13:16 ext2fs.pc
-rw-r--r-- 1 root root 419 фев 14 14:18 form.pc
-rw-r--r-- 1 root root 692 фев 14 13:09 gio-2.0.pc
-rw-r--r-- 1 root root 211 фев 14 13:09 gio-unix-2.0.pc
-rw-r--r-- 1 root root 530 фев 14 13:09 girepository-2.0.pc
-rw-r--r-- 1 root root 505 фев 14 13:09 glib-2.0.pc
-rw-r--r-- 1 root root 254 фев 14 13:09 gmodule-2.0.pc
-rw-r--r-- 1 root root 254 фев 14 13:09 gmodule-export-2.0.pc
-rw-r--r-- 1 root root 254 фев 14 13:09 gmodule-no-export-2.0.pc
-rw-r--r-- 1 root root 254 фев 14 13:09 gobject-2.0.pc
-rw-r--r-- 1 root root 223 фев 14 13:09 gthread-2.0.pc
-rw-r--r-- 1 root root 297 фев 14 14:20 history.pc
-rw-r--r-- 1 root root 327 фев 14 15:11 libbsd-ctor.pc
-rw-r--r-- 1 root root 379 фев 14 15:11 libbsd-overlay.pc
-rw-r--r-- 1 root root 268 фев 14 15:11 libbsd.pc
-rw-r--r-- 1 root root 316 фев 14 13:12 libcrypto.pc
-rw-r--r-- 1 root root 199 фев 14 13:09 libffi.pc
-rw-r--r-- 1 root root 260 фев 14 14:19 libical.pc
-rw-r--r-- 1 root root 264 фев 14 14:20 libmd.pc
-rw-r--r-- 1 root root 419 фев 14 14:23 libncurses.pc
-rw-r--r-- 1 root root 252 фев 14 13:13 libnl-3.0.pc
-rw-r--r-- 1 root root 303 фев 14 13:13 libnl-cli-3.0.pc
-rw-r--r-- 1 root root 241 фев 14 13:13 libnl-genl-3.0.pc
-rw-r--r-- 1 root root 253 фев 14 13:13 libnl-idiag-3.0.pc
-rw-r--r-- 1 root root 245 фев 14 13:13 libnl-nf-3.0.pc
-rw-r--r-- 1 root root 250 фев 14 13:13 libnl-route-3.0.pc
-rw-r--r-- 1 root root 248 фев 14 13:13 libnl-xfrm-3.0.pc
-rw-r--r-- 1 root root 274 фев 14 13:09 libpcre2-16.pc
-rw-r--r-- 1 root root 274 фев 14 13:09 libpcre2-32.pc
-rw-r--r-- 1 root root 271 фев 14 13:09 libpcre2-8.pc
-rw-r--r-- 1 root root 264 фев 14 13:12 libssl.pc
-rw-r--r-- 1 root root 266 фев 14 13:10 libsyslog.pc
-rw-r--r-- 1 root root 419 фев 14 14:18 menu.pc
-rw-r--r-- 1 root root 642 фев 14 13:15 mount.pc
-rw-r--r-- 1 root root 448 фев 14 14:18 ncurses++.pc
-rw-r--r-- 1 root root 419 фев 14 14:18 ncurses.pc
-rw-r--r-- 1 root root 218 фев 14 13:12 openssl.pc
-rw-r--r-- 1 root root 421 фев 14 14:18 panel.pc
-rw-r--r-- 1 root root 311 фев 14 14:20 readline.pc
-rw-r--r-- 1 root root 186 фев 14 14:20 sbc.pc
-rw-r--r-- 1 root root 204 фев 14 13:15 smartcols.pc
-rw-r--r-- 1 root root 275 фев 14 13:20 sndfile.pc
-rw-r--r-- 1 root root 273 фев 14 13:16 ss.pc
-rw-r--r-- 1 root root 426 фев 14 14:18 tinfo.pc
-rw-r--r-- 1 root root 241 фев 14 13:16 uuid.pc
-rw-r--r-- 1 root root 252 фев 14 13:08 zlib.pc

borine, 2025-02-14:
Are you sure you have included all the relevant .pc file directories? I know every build environment is different, but for my cross-compiler I have
```
PKG_CONFIG_LIBDIR=$sysroot/usr/local/lib/arm-linux-gnueabihf/pkgconfig:$sysroot/usr/local/lib/pkgconfig:$sysroot/usr/local/share/pkgconfig:$sysroot/usr/lib/pkgconfig:$sysroot/usr/share/pkgconfig:$sysroot/usr/lib/arm-linux-gnueabihf/pkgconfig
```

sergey-suloev, 2025-02-14:
> Hmm, everything works just fine as it is in my cross-build environment (build: x86_64-linux-gnu, host: arm-linux-gnueabihf)
> 
> $ arm-linux-gnueabihf-ldd ./rpi-bookworm/Build/bluealsa/utils/rfcomm/bluealsa-rfcomm 
>         libdbus-1.so.3 => /lib/arm-linux-gnueabihf/libdbus-1.so.3 (0x8badf00d)
>         libsystemd.so.0 => /lib/arm-linux-gnueabihf/libsystemd.so.0 (0x8badf00d)
>         libcap.so.2 => /lib/arm-linux-gnueabihf/libcap.so.2 (0x8badf00d)
>         libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x8badf00d)
>         ld-linux-armhf.so.3 => /lib/ld-linux-armhf.so.3 (0x8badf00d)
>         libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0x8badf00d)
>         libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0x8badf00d)
>         liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x8badf00d)
>         libzstd.so.1 => /lib/arm-linux-gnueabihf/libzstd.so.1 (0x8badf00d)
>         liblz4.so.1 => /lib/arm-linux-gnueabihf/liblz4.so.1 (0x8badf00d)
>         libreadline.so.8 => /lib/arm-linux-gnueabihf/libreadline.so.8 (0x8badf00d)
>         libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0x8badf00d)
>         libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x8badf00d)
> As you can see it has picked up libreadine and libtinfo, without needing me to define any environment variables for configure. If you are needing to define BLUEZ_LIBS and GIO2_LIBS then surely this suggests that you build environment is incomplete?

No, everything is complete here. But you should not rely on the assumption that the binaries are built with  rpath because this is not always the case

arkq, 2025-02-14:
> I wonder, can you remember why pkgconfig was not used for libreadline?

I think that there was no pkg-config support for readline when it was added as a project dependency. If the pkg-config for readline nowadays is widely supported among Linux distributions we should add it.

arkq, 2025-03-18:
The libreadline pkg-config suport was added to master, closing.

closed 2025-03-18 by arkq