arkq/bluez-alsa #761: No audiostream when connecting via bluetoothctl

ChristXXY, 2025-05-02:
Hey there,

I use this project with a raspi 4B (with bluetooth dongle) and HifiBerry DAC Pro 2 to make my audio amplifier "smart". As part of my smarthome project, I want to tell my raspberrypi to connect to my devices (for example phone, tablet, pc). When i establish the bluetooth connection from my devices to the raspberry everything works great. 
Output in `bluetoothctl`:
```
Transport /org/bluez/hci0/dev_B0_54_76_ED_24_5B/fd4
```

But when I connect vice versa via `bluetootctl `using` connect {MAC_OF_PHONE}` it results in delay and poor audio quality. 

Output of `bluetoothctl`:
```
Transport /org/bluez/hci0/dev_B0_54_76_ED_24_5B/sep1/fd6
```
Which means its not using the same endpoint as above and no active Audio-Stream is established right?

Any idea how to fix that?

Thanks!!

borine, 2025-05-03:
Your transport object paths are as expected and correct; that is just the way that bluez implements its D-Bus interface for AVDTP. Whatever the problem is , it is not the transport object paths.

If you want help you need to give some basic information about your bluealsa setup. If we cannot reproduce the problem then we cannot investigate it. Had you used the bug report template instead of the blank template when raising this issue then that would have prompted you for the information required to at least make a start on problem analysis. Please have a look at that template and provide the information requested.

ChristXXY, 2025-05-13:
ah ok I am new to this world... But here the infos:
OS distribution and version: Debian GNU/Linux 12 (bookworm)
BlueALSA: v4.0.0
BlueZ 5.66
ALSA: 1.2.8

I could already fix the problem by running these commands together:
```
sudo pkill bluealsa-aplay
bluetoothctl connect {device_adress}
bluealsa-aplay {device_adress}
```

Because whenever I start bluealsa-aplay I get this:
```
bluealsa-aplay: D: aplay.c:690: Creating PCM worker C8:6C:3D:F6:0C:1B
bluealsa-aplay: D: aplay.c:1018: Starting main loop
bluealsa-aplay: E: aplay.c:448: Couldn't open PCM: Device or resource busy
bluealsa-aplay: D: aplay.c:422: Exiting PCM worker C8:6C:3D:F6:0C:1B
```

borine, 2025-05-13:
> BlueALSA: v4.0.0

The first thing to say is that version is 3 years old now. There have been many improvements since then, and I would recommend you update to release v4.3.1 (or to the latest source code if possible).  If you installed from the Debian repository then perhaps you should raise an issue with debian so that their package maintainer might be prompted to upgrade.

> bluealsa-aplay: E: aplay.c:448: Couldn't open PCM: Device or resource busy

That message tells me that some other process (or thread) has already opened the ALSA device. So the first thing to determine is what application is using the device. As you are using Debian, my first suspicion is `pipewire`. Do you have `pipewire` installed on this machine? If so, you should **not** be using BlueALSA - although it is possible to use both together, you need some fairly detailed knowledge of pipewire, BlueALSA, Bluez and ALSA to do it successfully.

If the problem is not pipewire, then perhaps `bluealsa-aplay` has already opened the ALSA device for some other bluetooth device or profile connection, and that's why restarting `bluealsa-aplay` fixes it.

The third possiblity is that there is some other application which is grabbing the ALSA device, but in that case it is hard to see how restarting `bluealsa-aplay` can have fixed the problem.

ChristXXY, 2025-05-13:
Thanks for the response. I will try to install the latest source code and if it doenst work I will use pipewire.

closed 2025-05-13 by ChristXXY