arkq/bluez-alsa #770: issue with volume control

lpmasser, 2025-06-16:
### Problem

Hi, I made a Bluetooth speaker using a Linux development board, and when I connect it with an **Android phone**, all functions work normally.
However, when I connect it with an **iPhone**, the volume control fails and it always outputs at maximum volume.

### Reproduction steps

build options:
`--enable-aac --enable-aptx --with-libopenaptx --enable-a2dpconf --enable-systemd`

bluealsa options: 
`--loglevel=debug -p a2dp-sink  --all-codecs`

bluealsa-aplay options:
`--loglevel=debug -v --pcm=default:UAC2Gadget --volume=none `

### Setup

ubuntu 24.04
BlueALSA v4.3.1-68-gbc3c2b4
BlueZ 5.72
aplay: version 1.2.9

### Additional context

sudo bluealsad --loglevel=debug -p a2dp-sink  --all-codecs

```
bluealsad: [19258] D: ../../src/a2dp-aac.c:815: FDK-AAC decoder capabilities: aac=0x1a4ffff sbr=0 dmx=0x7f
bluealsad: [19258] D: ../../src/storage.c:90: Initializing persistent storage: /var/lib/bluealsa
bluealsad: [19258] D: ../../src/main.c:710: Starting main dispatching loop
bluealsad: [19258] D: ../../src/main.c:124: Acquired D-Bus service name: org.bluealsa
bluealsad: [19258] D: ../../src/bluealsa-dbus.c:407: Registering BlueALSA D-Bus manager: /org/bluealsa
bluealsad: [19258] D: ../../src/bluez.c:153: Registering media application: /org/bluez/hci0
bluealsad: [19258] D: ../../src/bluez.c:749: Exporting media endpoint object: /org/bluez/hci0/A2DP/aptX/sink/1
bluealsad: [19258] D: ../../src/bluez.c:749: Exporting media endpoint object: /org/bluez/hci0/A2DP/aptX/sink/2
bluealsad: [19258] D: ../../src/bluez.c:749: Exporting media endpoint object: /org/bluez/hci0/A2DP/AAC/sink/1
bluealsad: [19258] D: ../../src/bluez.c:749: Exporting media endpoint object: /org/bluez/hci0/A2DP/AAC/sink/2
bluealsad: [19258] D: ../../src/bluez.c:749: Exporting media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/1
bluealsad: [19258] D: ../../src/bluez.c:749: Exporting media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/2
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/sep1, ...)
bluealsad: [19258] D: ../../src/bluez.c:1368: Adding new Stream End-Point: F8:71:A6:5F:F1:EE: SRC: SBC
bluealsad: [19258] D: ../../src/bluez.c:1371: SEP capabilities blob [len=4]: 211502fa
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/sep2, ...)
bluealsad: [19258] D: ../../src/bluez.c:1368: Adding new Stream End-Point: F8:71:A6:5F:F1:EE: SRC: SBC
bluealsad: [19258] D: ../../src/bluez.c:1371: SEP capabilities blob [len=4]: 211502fa
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/sep3, ...)
bluealsad: [19258] D: ../../src/bluez.c:1368: Adding new Stream End-Point: F8:71:A6:5F:F1:EE: SRC: AAC
bluealsad: [19258] D: ../../src/bluez.c:1371: SEP capabilities blob [len=6]: 80010483e800
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/sep4, ...)
bluealsad: [19258] D: ../../src/bluez.c:1368: Adding new Stream End-Point: F8:71:A6:5F:F1:EE: SRC: AAC
bluealsad: [19258] D: ../../src/bluez.c:1371: SEP capabilities blob [len=6]: 80010483e800
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/sep5, ...)
bluealsad: [19258] D: ../../src/bluez.c:1368: Adding new Stream End-Point: F8:71:A6:5F:F1:EE: SRC: (null)
bluealsad: [19258] D: ../../src/bluez.c:1371: SEP capabilities blob [len=14]: 4c0000000180008000840084e200
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/sep6, ...)
bluealsad: [19258] D: ../../src/bluez.c:1368: Adding new Stream End-Point: F8:71:A6:5F:F1:EE: SRC: (null)
bluealsad: [19258] D: ../../src/bluez.c:1371: SEP capabilities blob [len=14]: 4c0000000180008000840084e200
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/player0, ...)
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/fd5, ...)
bluealsad: [19258] D: ../../src/dbus.c:47: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/AAC/sink/1
bluealsad: [19258] D: ../../src/storage.c:123: Loading storage: /var/lib/bluealsa/F8:71:A6:5F:F1:EE
bluealsad: [19258] D: ../../src/bluez.c:576: A2DP Sink (AAC) configured for device F8:71:A6:5F:F1:EE
bluealsad: [19258] D: ../../src/bluez.c:579: A2DP selected configuration blob [len=6]: 800104835b60
bluealsad: [19258] D: ../../src/bluez.c:581: PCM configuration: channels=2 rate=44100
bluealsad: [19258] D: ../../src/bluez.c:583: Delay reporting: unsupported
bluealsad: [19258] D: ../../src/bluez.c:749: Exporting media endpoint object: /org/bluez/hci0/A2DP/AAC/sink/3
bluealsad: [19373] D: ../../src/dbus.c:47: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_F8_71_A6_5F_F1_EE/a2dpsnk/source
bluealsad: [19258] D: ../../src/bluez.c:1296: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE, ...)
bluealsad: [19258] D: ../../src/bluez.c:1400: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved(/org/bluez/hci0/dev_F8_71_A6_5F_F1_EE/service0023/char0024/desc0026, ...)
[... 42 lines cut ...]
bluealsad: [19258] D: ../../src/bluez.c:1487: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: Volume
bluealsad: [19258] D: ../../src/bluez.c:1509: Updating A2DP volume: 23 [-24.65 dB]
```

sudo bluealsa-aplay --loglevel=debug -v --pcm=default:UAC2Gadget --volume=none

```
bluealsa-aplay: [19327] I: ../../../utils/aplay/aplay.c:1499: Selected configuration:
  BlueALSA service: org.bluealsa
  ALSA PCM device: default:UAC2Gadget
  ALSA PCM buffer time: 200000 us
  ALSA PCM period time: 50000 us
  ALSA mixer device: (not used)
  ALSA mixer element: (not used)
  Volume control type: none
  Bluetooth device(s): ANY
  Profile: A2DP
bluealsa-aplay: [19327] D: ../../../utils/aplay/aplay.c:1555: Starting main loop
bluealsa-aplay: [19327] D: ../../../utils/aplay/aplay.c:1076: Starting IO worker F8:71:A6:5F:F1:EE
bluealsa-aplay: [19372] D: ../../../utils/aplay/aplay.c:532: Setting BlueALSA source PCM volume mode: /org/bluealsa/hci0/dev_F8_71_A6_5F_F1_EE/a2dpsnk/source: pass-through
bluealsa-aplay: [19372] D: ../../../utils/aplay/aplay.c:544: Opening BlueALSA source PCM: /org/bluealsa/hci0/dev_F8_71_A6_5F_F1_EE/a2dpsnk/source
bluealsa-aplay: [19372] D: ../../../utils/aplay/aplay.c:592: Starting IO loop
bluealsa-aplay: [19372] D: ../../../utils/aplay/aplay.c:740: Opening ALSA playback PCM: name=default:UAC2Gadget channels=2 rate=44100

```

borine, 2025-06-16:
> bluealsa-aplay options:
> `--loglevel=debug -v --pcm=default:UAC2Gadget --volume=none` 

This depends on what exactly you want to achieve. The bluealsa-aplay command above tells BlueALSA not to apply any volume control requests from the remote device. So if the device sends a volume change request then it is just ignored. I guess that is what is happening with the iphone, whereas the android is not sending requests but is instead applying volume changes to the stream before it is sent to bluetooth. So do you want the Linux board to receive and apply volume control requests, or do you want the phones to operate volume control locally?

If you want to turn off remote volume control rather than just ignoring requests, then you will have to either disable AVRCP in Bluez, or disable Bluetooth AbsoluteVolume in the phones. Then both android and iphone will know to apply volume control locally. To disable AVRCP in Bluez, run the bluetoothd as
```
bluetoothd --noplugin=avrcp
```

lpmasser, 2025-06-17:
Thank you for your help. I misunderstood the volume control settings because it was my first time using an Android device, but now it works properly.

closed 2025-06-17 by lpmasser