arkq/bluez-alsa #774: Can't play any more 44.1kHz-sampled streams on a 48kHz-capable sink: Couldn't change BlueALSA PCM configuration: Input/output error

sphakka, 2025-07-30:
### Problem

I used to happily play streams sampled at different rates with bluez-alsa on a 48kHz-capable AD2P sink (WaveMaster Cube Mini). Now anything sampled differently tha  48kHz or any mono streams would cause the sink to disconnect and, of course, won't play.    

#### Symptoms

At BT sink connection (possibly unrelated)

[syslog]
```
Jul 30 14:52:44 [bluetoothd] profiles/audio/avctp.c:uinput_create() Can't open input device: No such file or directory (2)
Jul 30 14:52:44 [bluetoothd] profiles/audio/avctp.c:init_uinput() AVRCP: failed to init uinput for wm CUBE mini
```

Playing a 44.1kHz-sampled stereo wav file

```bash
$ aplay --verbose -D bluealsa test-stereo-cd.wav 
Playing WAVE 'test-stereo-cd.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
ALSA lib /var/tmp/portage/media-sound/bluez-alsa-4.3.1-r1/work/bluez-alsa-4.3.1/src/asound/bluealsa-pcm.c:590:(bluealsa_hw_params) Couldn't change BlueALSA PCM configuration: Input/output error
aplay: set_params:1456: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: 125000
PERIOD_SIZE: (5512 5513)
PERIOD_BYTES: 22050
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 22050
BUFFER_BYTES: 88200
TICK_TIME: 0 
```

[syslog]
```
ul 30 14:57:16 [bluetoothd] profiles/audio/avdtp.c:avdtp_parse_rej() OPEN request rejected: Bad State (49)
Jul 30 14:57:16 [bluealsa] Couldn't set A2DP configuration: GDBus.Error:org.bluez.Error.Failed: Input/output error
Jul 30 14:57:16 [bluealsa] Couldn't select codec: SBC: Input/output error
```

### Reproduction steps

1. Have a 44.1kHz-sampled stream. F.i.
   ```
   $ arecord -f cd -d 5 test-stereo-cd.wav
   ```
2. Play it via bluez-alsa
   ```
   $ aplay --verbose -D bluealsa test-stereo-cd.wav
   ```
3. BT sink is now unusable until reconnection...

### Setup

- GNU/Linux Gentoo, kernel v6.12.21
- BlueALSA v4.3.1
- BlueZ v5.83
- ALSA v1.2.14

### Additional context

[asound.rc]
```
pcm.bthw_WmCubeMini {
    type      bluealsa
    device    00:03:33:22:33:22
    profile   "a2dp"
    delay     10000
}
ctl.bthw_WmCubeMini {
    type      bluealsa
    device    00:03:33:22:33:22
    profile   "a2dp"
}
pcm.wmcube {
    type plug
    slave {
        pcm bthw_WmCubeMini
    }
    hint {
        show on
        description "Bluetooth WM Cube Mini speakers"
    }
}
```

sphakka, 2025-07-30:
Thanks for the workaround: it does the trick ^^

The directly related packages that got several upgrade in the last couple months are 'alsa-libs' and 'bluez'. I'll try to downgrade selectively and report.

arkq, 2025-07-30:
Could you potentially check whether this issues that you can see is not yet another manifestation of this: https://github.com/bluez/bluez/issues/1443 ?

borine, 2025-07-31:
Bisecting bluez shows that this failure starts with bluez commit [aa118e9](https://github.com/bluez/bluez/commit/aa118e965b02a6b78039c1ed55b9947698c6de59) "a2dp: Don't wait to reconfigure", which first appeared in bluez release 5.80.

Reverting that commit restores bluealsa's ability to change codec configurations.

I have not yet had time to understand the reason that this commit was introduced nor why it causes failure for bluealsa. I will continue to work on it.

arkq, 2025-07-31:
> a2dp: Don't wait to reconfigure

I guess that's because bluealsa selects configuration in a sync way... changing this to async should fix the problem, hopefully (will check that)
@borine Thanks a lot for bisicting bluez!

borine, 2025-07-31:
hmm, so the plot thickens. My original bluez sink is an rpi 2b with USB adapter. I've just built a new sink with a rpi zero W with on-board UART adapter, using the same bluez build (from the same built tar file from the bluez 5.79 source) as the pi 2B.

The pi 2B suffers this failure, the pi zero W does not. The only difference I can think of is the bluetooth protocol response times from the different hardware. So it seems that we have a problem that only manifests with certain remote devices (possibly too slow devices?). Next week I will compare bluetooth and bluealsa logs.

borine, 2025-08-01:
I found some time this afternoon to run some wireshark traces on working devices and failing devices. It is clear that in all cases the sequence of AVDTP commands and responses is correct, but some devices (the failing ones) send the accept response to a CLOSE command before they are actually ready to accept a SetConfiguration command for the same SEP. For such devices the source has to wait a short time after receiving the close accept response before sending a SetConfiguration command, otherwise it will receive a "already in use" reject response. 

So I believe that bluez commit [aa118e9](https://github.com/bluez/bluez/commit/aa118e965b02a6b78039c1ed55b9947698c6de59) is wrong, and for some devices it makes changing the codec configuration impossible.

For my own failing devices (a Broadcom usb adapter and a Realtek usb adapter) a delay of 100ms is suffiicient, whereas before that commit Bluez used a delay of 500ms. But having no delay means that sinks with such adapters cannot have their codec configurations changed after initial session setup.