arkq/bluez-alsa #778: Not building (anymore) - issue/feature request/bug

TeamB12, 2025-08-21:
> Please read the [troubleshooting guide](../blob/master/TROUBLESHOOTING.md)
> before raising a new issue.

### Problem

> A clear and concise description of what the bug is.
> If possible, please check if the bug still exists in the master branch (or
> the latest release if you are not using it already).

### Reproduction steps

> Provide a minimal example of how to reproduce the problem. State `bluealsad`
> command line arguments and the content of .asoundrc file (if PCM alias with
> "type bluealsa" was added to that file).

### Setup

> - the OS distribution and version
> - the version of BlueALSA (`bluealsad --version`)
> - the version of BlueZ (`bluetoothd --version`)
> - the version of ALSA (`aplay --version`)
> - if self-built from source, please state the branch and commit
> (`git log -1 --oneline`), and the used configure options.

### Additional context

> Add any other context about the problem here, e.g. log messages printed by
> `bluealsad` and/or client application.
>
> Please delete instructions prefixed with '>' to prove you have read them.

TeamB12, 2025-08-21:
Using and building bluealsa worked in the past here on several machines and new version will not build with error message systemd is missing.
Artix Linux here with openrc init system - no systemd installed.
Maybe this is now an "issue by design" and you want and embrace systemd while excluding linux setups with other init systems, then feel free to close this issue.

borine, 2025-08-21:
😩 so first, a rant. you ignored the text in the issue template and failed to supply the requested information. Statements like "new version" are not helpful to anyone trying to help you. I don't have Artix Linux and no no intention of installing it. To have any chance of reproducing this we need to know which commit you are building from and the command line you used to `configure`. The precise text of the error message would also be very helpful. Error messages exist to help diagnose problems, so always include them when reporting problems.

OK, so rant over.

BlueALSA is supposed to build and run without any need for systemd; so perhaps there is a mistake in the configure or make files. Please help us to locate and fix any errors by supplying relevant details about this failure so that we can investigate and hopefully prevent this happening to other users.

TeamB12, 2025-08-21:
Sorry.
Latest version was tried to build by update system of Arch Linux user package from AUR:

https://aur.archlinux.org/packages/bluealsa

Last Updated:  2025-08-19 03:40 (UTC)
pkgver:  4.3.1
Kernel: Linux 6.14.0-rt3-arch1-3-rt

The install script used the folling commands to configure and build:
autoreconf --install
	flags=( #--with-libopenaptx
		  #--enable-a2dpconf
		  --enable-aac
		  --enable-cli
		  --enable-debug
		  #--enable-hcitop
		  #--enable-ldac
		  --enable-manpages
		  #--enable-mp3lame
		  #--enable-mpg123
		  #--enable-msbc
		  --enable-ofono
		  #--enable-rfcomm
		  --enable-systemd
		  --enable-upower )
	./configure --prefix=/usr --sysconfdir=/etc "${flags[@]}"
	make

 Running an edited build script (I removed the systemd dependency, that is why I thought this would go into the direction issue by design.)

The error message
configure: error: Package requirements (systemd >= 200) were not met:

Package 'systemd' not found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables SYSTEMD_CFLAGS
and SYSTEMD_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
==> ERROR: A failure occurred in build().
    Aborting...

sorry, I do not read some things to train my good guessing, intuition and empathy.
RTFM is what I want to avoid

borine, 2025-08-21:
> The install script used the folling commands to configure and build:
autoreconf --install
flags=( #--with-libopenaptx
#--enable-a2dpconf
--enable-aac
--enable-cli
--enable-debug
#--enable-hcitop
#--enable-ldac
--enable-manpages
#--enable-mp3lame
#--enable-mpg123
#--enable-msbc
--enable-ofono
#--enable-rfcomm
--enable-systemd
--enable-upower )
./configure --prefix=/usr --sysconfdir=/etc "${flags[@]}"
make

For pity's sake. That command line **clearly** says `--enable-systemd`. Is this a wind-up? 

> my good guessing, intuition and empathy

Well in this case I see only poor guessing and intuition. And my own empathy is wearing very thin.

TeamB12, 2025-08-21:
==> Finished making: bluealsa 4.3.1-1 (Thu Aug 21 21:07:56 2025)
------------------------
Thank you, for pointing me to things I posted to you: after removing this systemd flag, the build was successful.

closed 2025-08-21 by TeamB12